olikraus/u8g2 #1502: HX1230 LCD XBM display

Jono72, 2021-05-24:
Hi All
As much as i want to use this cute display (rather than the old Nokia 5110), Its driving me mad to say the least, I know there are only a few library's out there and that display is 9 bit as opposed to the standard 8.
The Ug82 library is the only one really that seems to give me good font choices and sizes, however when I try and load my logo for the splash screen it doesn't appear as it should, I've converted my image to the correct size and orientation and using the XBM demo to test it, I have used LCD assistant and image2cpp; still I cannot get the display to show correctly, can anyone point me in the right direction here please, I have attached the BMP I'm trying to display and photo of the result i get?? Any advice would be helpful!

Thanks in advance 

John 

[image]

[Logo.zip](https://github.com/olikraus/u8g2/files/6530869/Logo.zip)

olikraus, 2021-05-24:
How does your U8g2 code look like? Please also post the XBM file.

Jono72, 2021-05-25:
Hi olikraus,

Thank you for the reply here is the code I'm using as a test, I'm using full buffer as project is using Atmel 2450, I seem to have all other aspects working, just not loading a BMP.

I'm assuming the array is out of place and pixels have shifted along, as you can sort of see the image, but its broken 

Code as follows :-

// Please UNCOMMENT one of the contructor lines below
// U8g2 Contructor List (Picture Loop Page Buffer)
// The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
// Please update the pin numbers according to your setup. Use U8X8_PIN_NONE if the reset pin is not connected

U8G2_HX1230_96X68_F_3W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 14, /* data=*/ 15, /* cs=*/ 5, /* reset=*/ 6);

// End of constructor list

#define u8g2_logo_width 96
#define u8g2_logo_height 68

//static const unsigned char u8g2_logo_97x51_bits[]PROGMEM = {
static const unsigned char u8g2_logo[] U8X8_PROGMEM = {
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0xff, 0xff, 
  0xff, 0xfe, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7f, 0xff, 0xff, 0xff, 0xfe, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 0xff, 0xff, 0xff, 0xfe, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x0f, 0xff, 0xff, 0xff, 0xfe, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0xff, 0xff, 
  0xff, 0xfe, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0xff, 0xff, 0xff, 0xfe, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0xff, 0xff, 0xfc, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0xff, 0xff, 0xff, 0xfc, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 
  0xff, 0xfc, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7f, 0xff, 0xff, 0xfc, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7f, 0xff, 0xff, 0xfc, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x3f, 0xff, 0xff, 0xfc, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3f, 0xff, 
  0xff, 0xf8, 0x00, 0x3f, 0xff, 0xff, 0xff, 0xfc, 0x00, 0x00, 0x3f, 0xff, 0xff, 0xf8, 0x00, 0x3f, 
  0xff, 0xff, 0xff, 0xff, 0xc0, 0x00, 0x3f, 0xff, 0xff, 0xf8, 0x00, 0x3f, 0xff, 0xff, 0xff, 0xff, 
  0xe0, 0x00, 0x1f, 0xff, 0xff, 0xf8, 0x00, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xf0, 0x00, 0x1f, 0xff, 
  0xff, 0xf8, 0x00, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xf0, 0x00, 0x1f, 0xff, 0xff, 0xf0, 0x00, 0x7f, 
  0xff, 0xe0, 0x01, 0xff, 0xf8, 0x00, 0x0f, 0xff, 0xff, 0xf0, 0x00, 0x7f, 0xff, 0xe0, 0x00, 0xff, 
  0xf8, 0x00, 0x0f, 0xff, 0xff, 0xf0, 0x00, 0x7f, 0xff, 0xe0, 0x01, 0xff, 0xf8, 0x00, 0x0f, 0xff, 
  0xff, 0xf0, 0x00, 0xff, 0xff, 0xe0, 0x01, 0xff, 0xfc, 0x00, 0x0f, 0xff, 0xff, 0xf0, 0x00, 0xff, 
  0xff, 0xe0, 0x01, 0xff, 0xfc, 0x00, 0x07, 0xff, 0xff, 0xf0, 0x00, 0xff, 0xff, 0xc0, 0x01, 0xff, 
  0xfc, 0x00, 0x07, 0xff, 0xff, 0xe0, 0x00, 0xff, 0xff, 0xc0, 0x01, 0xff, 0xfc, 0x00, 0x07, 0xff, 
  0xff, 0xe0, 0x00, 0xff, 0xff, 0xc0, 0x01, 0xff, 0xfe, 0x00, 0x07, 0xff, 0xff, 0xe0, 0x00, 0xff, 
  0xff, 0xc0, 0x03, 0xff, 0xfe, 0x00, 0x03, 0xff, 0xff, 0xe0, 0x01, 0xff, 0xff, 0xc0, 0x03, 0xff, 
  0xfe, 0x00, 0x03, 0xff, 0xff, 0xe0, 0x01, 0xff, 0xff, 0x80, 0x03, 0xff, 0xfe, 0x00, 0x03, 0xff, 
  0xff, 0xe0, 0x01, 0xff, 0xff, 0x80, 0x03, 0xff, 0xff, 0x00, 0x03, 0xff, 0xff, 0xc0, 0x01, 0xff, 
  0xff, 0x80, 0x03, 0xff, 0xff, 0x00, 0x01, 0xff, 0xff, 0xc0, 0x01, 0xff, 0xff, 0x80, 0x07, 0xff, 
  0xff, 0x00, 0x01, 0xff, 0xff, 0xc0, 0x03, 0xff, 0xff, 0x00, 0x07, 0xff, 0xff, 0x80, 0x01, 0xff, 
  0xff, 0xc0, 0x03, 0xff, 0xff, 0x80, 0x07, 0xff, 0xff, 0x80, 0x01, 0xff, 0xff, 0xc0, 0x03, 0xff, 
  0xff, 0x00, 0x07, 0xff, 0xff, 0x80, 0x00, 0xff, 0xff, 0x80, 0x03, 0xff, 0xff, 0x00, 0x07, 0xff, 
  0xff, 0xc0, 0x00, 0xff, 0xff, 0x80, 0x03, 0xff, 0xff, 0x00, 0x07, 0xff, 0xff, 0xc0, 0x00, 0xff, 
  0xff, 0x80, 0x03, 0xff, 0xff, 0x00, 0x0f, 0xff, 0xff, 0xc0, 0x00, 0xff, 0xff, 0x80, 0x07, 0xff, 
  0xfe, 0x00, 0x0f, 0xff, 0xff, 0xc0, 0x00, 0x7f, 0xff, 0x00, 0x07, 0xff, 0xfe, 0x00, 0x0f, 0xff, 
  0xff, 0xe0, 0x00, 0x7f, 0xff, 0x80, 0x07, 0xff, 0xfe, 0x00, 0x0f, 0xff, 0xff, 0xe0, 0x00, 0x7f, 
  0xff, 0x00, 0x07, 0xff, 0xfe, 0x00, 0x0f, 0xff, 0xff, 0xe0, 0x00, 0x7f, 0xff, 0x00, 0x07, 0xff, 
  0xfe, 0x00, 0x1f, 0xff, 0xff, 0xf0, 0x00, 0x3f, 0xff, 0x00, 0x0f, 0xff, 0xfc, 0x00, 0x1f, 0xff, 
  0xff, 0xf0, 0x00, 0x3f, 0xff, 0x00, 0x0f, 0xff, 0xfc, 0x00, 0x1f, 0xff, 0xff, 0xf0, 0x00, 0x3f, 
  0xfe, 0x00, 0x0f, 0xff, 0xfc, 0x00, 0x1f, 0xff, 0xff, 0xf0, 0x00, 0x1f, 0xfe, 0x00, 0x0f, 0xff, 
  0xfc, 0x00, 0x1f, 0xff, 0xff, 0xf8, 0x00, 0x1f, 0xfe, 0x00, 0x0f, 0xff, 0xfc, 0x00, 0x1f, 0xff, 
  0xff, 0xf8, 0x00, 0x1f, 0xfe, 0x00, 0x0f, 0xff, 0xfc, 0x00, 0x3f, 0xff, 0xff, 0xf8, 0x00, 0x1f, 
  0xfe, 0x00, 0x1f, 0xff, 0xf8, 0x00, 0x3f, 0xff, 0xff, 0xfc, 0x00, 0x0f, 0xfc, 0x00, 0x1f, 0xff, 
  0xf8, 0x00, 0x3f, 0xff, 0xff, 0xfc, 0x00, 0x0f, 0xfc, 0x00, 0x1f, 0xff, 0xf8, 0x00, 0x3f, 0xff, 
  0xff, 0xfc, 0x00, 0x0f, 0xfc, 0x00, 0x1f, 0xff, 0xf8, 0x00, 0x3f, 0xff, 0xff, 0xfc, 0x00, 0x0f, 
[... cut ...]
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 
  0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff };

void setup(void) {

  
  u8g2.begin(); 
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.drawXBMP(0,0, u8g2_logo_width, u8g2_logo_height, u8g2_logo);
  u8g2.sendBuffer();
}

void loop(void) {

}

I have already attache logo.zip

Kind Regards

John

olikraus, 2021-05-25:
I took the xbm data out of your code into gimp and created a png out of it:

[image]

So it looks like that U8g2 is correct. Maybe your conversion process to XBM is wrong. How did you convert BMP to XBM?

Oliver

Jono72, 2021-05-25:
Hi Olikraus

That's strange, I used https://javl.github.io/image2cpp/ when i paste the array into this online converter it shows me the bitmap created? What are you using to create array ?
lcd assitant does exactly the same thing

Regards

John

Jono72, 2021-05-25:
The zip file I attached logo.zip contains the picture I'm trying to create.

olikraus, 2021-05-25:
I do not know image2cpp and if this was used to generate the c code, then it is wrong. 
XBM picture format is already c code, there is no need to convert something to c or cpp.

You could follow this instructions:
https://sandhansblog.wordpress.com/2017/04/16/interfacing-displaying-a-custom-graphic-on-an-0-96-i2c-oled/

Jono72, 2021-05-25:
Hi Olikraus,

That worked perfectly, btw your library is pretty cool and very comprehensive, I am looking at the menu code to, Ive created menus before and they can get extremely complicated very quickly, jumping around all over the place, your menu function is neat, I learnt something too. Awesome job thank you :)

Regards

John

Jono72, 2021-05-25:
Hi Oliver,

I apologies in advance, last question (promise) if you don't mind, display is working great and does now pretty much everything, except invert text, I've searched around for hours, I realize you can't just change an entire library (that would be a horrendous job i suspect).

I did find that you implemented a sendF command, after searching I manged to find the command table in english for this display and realized that i can send 0x0a7, but of course this inverts the entire display.
Is there any easy way to invert selected or printed text or is the only way drawing rectangle and changing modes? I see you u have an inverse function in the U8x8,  I tried searching through source to see how it was done, Im guessing its a lot more complicated than a simple command.

This is a good display but what a pain in the butt to use compared with other standards on the market, cheap tho.

Kind Regards

John

olikraus, 2021-05-25:
> see you u have an inverse function in the U8x8

You mean this command, right?
https://github.com/olikraus/u8g2/wiki/u8x8reference#setinversefont

It is done by inverting the font itself. This is ok for u8x8 because all chars have the same size, but impossible for u8g2 because chars might have different heights (so it would look somehow ugly).

>  is the only way drawing rectangle and changing modes?

Yes: After writing a string, place a box on top with draw color 2 (XOR): https://github.com/olikraus/u8g2/wiki/u8g2reference#setdrawcolor

Of course you need to calculate the width and height of this box. The width depends on the string pixel width (https://github.com/olikraus/u8g2/wiki/u8g2reference#getstrwidth), the height is the difference (!) of ascent and descent (https://github.com/olikraus/u8g2/wiki/u8g2reference#getascent).

closed 2022-03-04 by olikraus