olikraus/u8g2 #1503: TTF to u8g2

Chroustall, 2021-05-24:
Hello,

i need this .TTF (https://fontstruct.com/fontstructions/download/1014500)  font in u8g2. I I don't know how to do it, could you please help me? Im little bit lost.

olikraus, 2021-05-24:
You font is not a vector font, so you need to figure out the point size for this font (-p for otf2bdf)
In your case
```
otf2bdf/otf2bdf  -n -p 12 -o falconded.bdf falconded.ttf
bdfconv -v -f 1 -m '32-65535' falconded.bdf -o u8g2_font_falconded.c  -n u8g2_font_falconded -d ../bdf/helvR12.bdf 
```
seems to produce good results
[image]

``` C++
/*
  Fontname: -FreeType-FalconDED-Medium-R-Normal--17-120-100-100-P-71-ISO10646-1
  Copyright: Copyright uri_ba 2014
  Glyphs: 134/134
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_falconded[1713] U8G2_FONT_SECTION("u8g2_font_falconded") = 
  "\206\0\3\4\4\4\3\4\5\11\13\0\377\11\377\11\0\1p\2\323\6\33 \5\0D\14!\7\222G"
  "\13&\2\42\10\65v\14\21*\2#\17\226E\234\240XL\202\342\305$(\26\0$\20\226E,\61"
  "\222\20\331\310D\244\204\220\11\1%\24\226E\234\60\221\20\221\240\300@\301H!\42!b!\0&\17"
  "\226E,\61\242hb$B$H#\1'\6\61v\212\1(\12\224\306+\21Ir\223\0)\13\224"
  "\306\13\61Ir\221\4\0*\17w\325\274\300\220hCGc!\21\303\0+\13fM,A!&A"
  "!\0,\11CF\33\21\22\21\0-\6&]\14\6.\6\42G\13\2/\15\226E\334@\301\210\202"
  "\21\5C\1\60\15\226E,\61\22!:\211\220\11\1\61\12\224\306\253 \221!\271 \62\15\226E\34"
  "\22$A!\22\22A\6\63\15\226E\14F\61\261Q\42\24\22\0\64\14\226E\14\21\371\205MP\12"
  "\0\65\14\226E\14\307\22C\42\24\22\0\66\14\226E\234\306\22$\42\24\22\0\67\12\226E\14Fi"
  "\62J\3\70\16\226E\34\22$\24\22!\42\24\22\0\71\16\226E\34\22$\42\224B\221\22\22\0:"
  "\7b\326\12B\2;\12s\305\32\211\42$\42\0<\11d\316+\21Ib\22=\10d\316\13t\20"
  "\4>\12d\316\13\61I\42\222\0\77\17\226E\34\22$\61A\61q\60A!\0@\15\226E,\61"
  "\22s\210&dB\0A\15\226E,\61\22$\242\207 \42\1B\15\226E\214\22\244\223\22!\242\26"
  "\0C\15\226E\34\22!\212\62\21\211\220\0D\13\226E\214\42\21\371\277\234\0E\13\226E\14\307\222"
  "\22AA\6F\13\226E\14\307\222\22A\31\1G\14\226E\34\22D)\225PH\0H\13\226E\14"
  "!J\17A\64\11I\11\226E\14&A\71\61J\13\226E\14&A\71\21\215\1K\17\226E\14!"
  "\22\224\61\301\61\42\221\42\1L\11\226E\14A\371\310\0M\13\226E\214\300\241\207 :\11N\15\226"
  "E\214\60\242\222\22\222J\324\2O\13\226E\34\22$:\241\220\0P\14\226E\214\22$\242\26A\211"
  "\0Q\13\226E\34\22$\272 \215\4R\14\226E\214\22$\242\26\231\210\4S\15\226E\34\22\304\242"
  "BA\24\22\0T\12\226E\14&A\371\4\0U\13\226E\14!\372\204B\2\0V\16\226E\14!"
  ":\211\4\5\221\11\12\1W\13\226E\14!:=\4\15\6X\21\226E\14!\42\221\240 \62\61\242"
  "(BD\2Y\16\226E\14!\42\221\240 \62A\71\1Z\11\226E\14F\61\71\62[\10\224F\14"
  "%\71!\134\16\226E\214P\321\340P\321\340P\321\0]\10\224F\14$\71)^\13D\356\33\221 "
  "\241\20\21\0_\6\26E\14\3a\16\270<\214\245\302a!\71K\222)\1b\16\270<\214\324a!"
  "\251\223\344,\252\0c\15\270<\14\305!\71\331$\323\240\2d\26\270<\214T\242\240\241\240\241\240\241"
  "\240\241\240\241\240\241 Q\5e\14\270<\214dI\246\223,\13'f\13\270<\214dI\246\223lj"
  "g\20\270<\14\305a!KaBaB\222\7\25h\13\270<\214$\371,I~Ji\14\270<\214"
  "da\42;\21\13'j\14\270<\214da\42;\21\222\61k\22\270<\214$Ia\242cEf\205"
  "DaBB\11l\12\270<\214$\373Y\70\1m\15\270<\214\24\22!\311\222\344\237\22n\24\270<"
  "\214\224!\331\202\304\202\204\302\204\302\204\344\64\222\0o\14\270<\14\305a!\371\317\203\12p\15\270<"
  "\214\324a!\71\213\16Yjq\15\270<\14\305a!\371\257d!\11r\20\270<\214\324a!\71\213"
  "\16\5\15\311\224\0s\17\270<\14\305a\221SR##\341A\5t\13\270<\214da\42\373\264\0"
  "u\13\270<\214$\371\377<\250\0v\15\270<\214$\371\237FD\10\213\26w\14\270<\214$\371\317"
  "\222\244\220$x\21\270<\214$\231FD\10\213\12ID\206dJy\16\270<\214$\231FD\10\213"
  "\354\264\0z\14\270<\214da\243:\31\13'{\15\226E\214\225\61\301P\301\261\22\3|\13\226E"
  ",Ay\62!\23\2}\15\226E\14\223\262A\321@\261\221\5\240\7\270<\214\177h\241\12\270<\214"
  "%{r\62Z\242\25\270<\214\245\302\221\240\20\221\240\302\242\20\221\240\220\301\242\5\243\22\270<\214\245"
  "B\22\221\242\302!Ia\242d!\11\244\16\270<\214\7\42#ba\42\243\7\6\246\13\270<\214%"
  "{\42\36,Z\250\15\270<\214\7\42\243\207#\243\7\6\251\17\270<\214\344\321\62\243\23\243\262Q\341"
  "\4\252\16\270<\214\7\42\243\207#\243\42#\6\253\13\270<\14EJ\366H\250\0\254\17\270<\214\224"
  "\243\23\224\243\23\224\243\223\4\256\17\270<\214d\321\261\42\223#\263\322\341\4\257\11\270<\214\77\24\77"
  "\10\260\16\270<\214\245\22\21\22\221\242\207\17\6\261\15\270<\214\207A\302\207B\302\207\5\266\15\270<"
  "\14\27#\23K&\226L\34\271\14\270<\214\207\242JF\227\36\26\272\14\270<\214\207\241KF\225\36"
  "\32\273\14\270<\14EB#;\21\22*\277\17\270<\14\305a!\241\42\243\42'\243\5\300\15\270<"
  "\214\245\302!\371\247\301\242\5\301\15\270<\14\26\243\62#\233\12\11\25\302\17\270<\14\305a!!#"
  "\302I\306\302\11\303\16\270<\214da\23\243\262#\231\7\25\304\22\270<\214\244\240\241\240\241\240\241\240"
  "a\341\42K\12\305\15\270<\214dI\246\303F\62\17*\306\16\270<\14Ua!\323a!\71\17*"
  "\307\14\270<\214da#\243\232,\61\310\17\270<\14\305a!\241A\302!\71\17*\311\16\270<\14"
  "\305a!\71\217\32\215\16*\327\22\271\274\214\207\23\223(e\344\0d%QL\36\16\367\25\270<\14"
  "\225\21\221H#\42\226LLD\206BBDF\24\0\0\0\4\377\377 \24\14\270<\214\17\313\302\17"
  "\17\3 \32\14\270<\214\37\214\214\212\214\30 \33\13\270<\14\225+\17\77\4 \36\17\270<\14\245"
  "\240\241\240\241\240\207\17\14 \42\12\270<\214\77\34\31- \71\14\270<\214\245Jv\272\264\0 :"
  "\14\270<\214\245Kv\252\264\0 \254\27\270<\14\25\21\22\221\341\21\21\22\21\22\221\341\21\21\22\21"
  "\5!\42\16\270<\214\245\302\341\207\345\301\242\5\0";
```

Chroustall, 2021-05-24:
Thanks a lot! So basically i just copy this code to u8g2_fonts? Sorry im newbie :).

Chroustall, 2021-05-24:
Thanks a lot! So basically i need only this.
[image]

Height of font is 7px like "u8g2_font_5x7_mr"

It is possible?

olikraus, 2021-05-24:
> Height of font is 7px like "u8g2_font_5x7_mr"

This is a pixel font, the height can not be influenced (well, it could, but it would look very ugly). The uppercase letters have a pixel height of 9. 

There is one more problem with the '*': The mr font will lead to a width of 9 pixel because of the '*', so I added also the hr font.

``` C++
const uint8_t u8g2_font_falconded_hr[806] U8G2_FONT_SECTION("u8g2_font_falconded_hr") = 
  "@\1\3\3\4\4\1\1\5\10\11\0\0\11\377\11\0\1\214\0\0\3\11 \7\230\343\343\77\1!\12"
  "\226\333\11\365\16\23J\0\42\13\230cId\22\231D\216\37#\22\230c%\206b\266P\60\24\14\305"
  "l\241`(\4$\22\230\343Ii\21I(\42\245JB\21I\214*\3%\23\230c\305B\221\210,"
  "\24\16\213\223b\222\244X\10\0&\20\230\343I\211\241\250\224&R\31\25'\21\0'\10\224S\305\322"
  "\61\2(\11\227_j\325Y\5\0)\11\227_\211UuT\3*\16\231g\206#\251\303\343\64\222\34"
  "\307\2+\14\230\343\243\212\205&\243X\16\4,\12\226\333\343*\224\11E\0-\11\230\343\343j\262c"
  "\4.\10\226\333\343\233P\2/\14\230c\207\305\311\342dq\30\0\60\13\230\343Ii\42\375\33U\6"
  "\61\13\227_F\205S\35i\24\0\62\15\230c\321L\312B\32Ml\262\0\63\16\230\343\230\314R\351"
  "\34 R\262\221\0\64\17\230\343\250Id\22\231Df\262\212\225\0\65\15\230\343\230L\342\232Y\244d"
  "#\1\66\15\230c\225L\342\232I'\33\11\0\67\12\230\343\230\314\252\232\25\1\70\15\230c\321LJ"
  "\66\232H\223\215\4\71\15\230c\321L:\331\312\262\32\11\0:\13\225\327\3dr\260L\16\4;\14"
  "\225\327Cer\210L$\23\1<\12\227\337#j\25\253C\0=\13\227\337#\321\350h\354\20\0>"
  "\13\227\337#\211U\205\352\60\0\77\15\230c\321L\252b\251\34A,\3@\14\230\343Iiv\254\66"
  "\252\14\0A\13\230\343Ii&\235\234t\1B\15\230\343\324LJ\265\232H\223\251\4C\13\230c\321"
  "D:k\322F\2D\23\230\343\24%\62\211L\42\223\310$\62\211L\42*\1E\15\230\343\230L\342"
  "ZM,\66Y\0F\14\230\343\230L\342ZM\254\25\0G\16\230c\321Lb\225\221d\244d#\1"
  "H\12\230\343\210\364\344\244\277\0I\13\230\343\230\214b\35M\26\0J\13\230\343\230\214b\35\211C\0"
  "K\17\230\343\210TF\265\251xJ\224\214T\0L\11\230\343\210\365\331d\1M\13\230\343\4CJN"
  "\372/\0N\21\230\343\304DJ\23\321D$\31IFz\213\0O\13\230c\321L\372O\66\22\0P"
  "\14\230\343\324L:\231jbU\0Q\14\230c\321L\372\227Qq\22\1R\16\230\343\324L:\231j"
  "\22\231H\13\0S\14\230c\321L\342bY\311F\2T\11\230\343\230\214b}\3U\12\230\343\210\364"
  "\377d#\1V\14\230\343\210\364o\241 U,\3W\14\230\343\210\364ONJ\301\10\0X\17\230\343"
  "\210\264\205\202T)\61\24\23i\1Y\14\230\343\210\264\205\202T\261n\0Z\13\230\343\230\314R\235M"
  "\26\0[\13\230c\21\211b\235\211$\0\134\24\230\343\304\1r@\34\22\7\310\1qH\34 \7D"
  "\0]\11\230c\21\311zd\2^\14\227\337\11C\261PP\216'\0_\10\230\343\343\237,\0\0\0"
  "\0\4\377\377\0";
```
``` C++
const uint8_t u8g2_font_falconded_mr[963] U8G2_FONT_SECTION("u8g2_font_falconded_mr") = 
  "@\2\4\3\4\4\1\1\5\11\11\0\0\11\377\11\0\1\272\0\0\3\246 \7\231\347\307\63\0!\13"
  "\231\347\221\323{\10\71!\0\42\15\231g\221\20\222\20\222\220\307\26\0#\24\231gI\25\22d\24\22"
  "\25\22\25\22d\24\22\25\22\3$\24\231\347\21#\212\220\210\211\20#\223\210\211\220\10\42\23\2%\23"
  "\231g\211I\21!\24\22\27'\227$H\42M\32\0&\22\231\347\21\243\12\11\23#\22\221\221\230\251"
  "\232\10\1'\10\231g\11L\217\3(\12\231g\22S\247C\65\0)\12\231g\21T\247\63U\0*"
  "\20\231g\12\213H\65t\64\25\221,\36\16\0+\14\231\347\7\223\223\262\261\222\223\6,\13\231\347\307"
  "FNLN\12\0-\11\231\347Ggc\17\16.\11\231\347\307\231\234\20\0/\15\231g\213\223K'"
  "\227N.\16\0\60\23\231\347\21#\22\221\21\221\21\221\21\221\21\21\42\23\2\61\14\231g\212\23\233\323"
  "\31\25\15\0\62\17\231g!\262\21\221\223\42\42\222\263\61\1\63\17\231\347\260\261\23\23\33\224\21\221\61"
  "\242\1\64\21\231\347P$!$!$!dc&\247\6\0\65\17\231\347\260\261\221+\262\223\21\221\61"
  "\242\1\66\20\231g\251\261\221+\262\21\221\21\221\61\242\1\67\13\231\347\260\261S\246\235*\0\70\21\231"
  "g!\262\21\221\61\42\22\221\21\221\61\242\1\71\20\231g!\262\21\221\21\221\61\252\23*\242\1:\13"
  "\231\347\225\223\7\225\223\7\4;\15\231\347\207\222\223\207\220\23\223\23\3<\13\231\347\207\23S'\250\30"
  "\0=\12\231\347\207b\17HE\14>\13\231\347\207\22T'\246\34\0\77\17\231g!\262\21\21\223\23"
  "\223\207\220\23\2@\14\231\347\21#\262G`D&\4A\21\231\347\21#\262\21\221\21\221q#\42#"
  "\42\2B\21\231\347(\262\21\221)*\22\221\21\221\261\251\1C\21\231g!\22\221\21\221\221\323DF"
  "D\210\6\0D\25\231\347\250\222\20\222\20\222\20\222\20\222\20\222\20\222\220\251\1E\16\231\347\260\261\221"
  "+*\222\223\263\61\1F\15\231\347\260\261\221+*\222\323\14\0G\20\231g!\262\221S\61#\61#"
  "\42cD\3H\24\231\347\20\221\21\221\21\221q#\42#\42#\42#\42\2I\14\231\347\260\261\222\323"
  "\225\215\11\0J\14\231\347\260\261\222\323\25\325\24\0K\22\231\347\20\221\221\230)\32\223\33\243\222\230\21"
  "\21\1L\12\231\347\220\323w\66&\0M\24\231\347\10\212\21\221q#\42#\42#\42#\42#\42\2"
  "N\26\231\347\210Q\42\63!\63!#\61#\61#\42#\42#\23\2O\23\231g!\262\21\221\21\221"
  "\21\221\21\221\21\221\61\242\1P\17\231\347(\262\21\221\21\221\261)\222S\6Q\24\231g!\262\21\221"
  "\21\221\21\221\21\221\221\230\251\232\10\1R\23\231\347(\262\21\221\21\221\261)\222\20\22\221\21\21\1S"
  "\15\231g!\262\221\253\252ScD\3T\12\231\347\260\261\222\323G\0U\25\231\347\20\221\21\221\21\221"
  "\21\221\21\221\21\221\21\221\61\242\1V\24\231\347\20\221\21\221\21\221\21\221\21\21\12\211\42\223\23\2W"
  "\24\231\347\20\221\21\221\21\221\21\221\21\221q#\42\23\24\2X\24\231\347\20\221\21\21\12\211\42\23\243"
  "\12\11\22\221\21\21\1Y\17\231\347\20\221\21\21\12\211\42\223\323\21\0Z\14\231\347\260\261\23\323\235\215"
  "\11\0[\14\231g\241\242\222\323\35\25\15\0\134\16\231\347\10\24\214\14\24\214\14\24\14\1]\12\231g"
  "\241\242\323+\66\0^\15\231\347\21\13\211\12\11\223\307\10\0_\10\231\347\307\265\11\0\0\0\0\4\377"
  "\377\0";
```

Chroustall, 2021-05-26:
Hmm, showing the "arrows" instead of "@" dosent work still :(
[image]
[image]

Chroustall, 2021-05-26:
#include "U8glib.h"
#include <U8g2lib.h>
#define DCSBIOS_IRQ_SERIAL
#include "DcsBios.h"
#include <Wire.h>
#define D0 13
#define D1 11
#define DC 9
#define CS 8
#define RES 10

U8G2_SSD1309_128X64_NONAME0_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 8, /* dc=*/ 9, /* reset=*/ 10);  
const uint8_t u8g2_font_falconded[1713] U8G2_FONT_SECTION("u8g2_font_falconded") = 
  "\206\0\3\4\4\4\3\4\5\11\13\0\377\11\377\11\0\1p\2\323\6\33 \5\0D\14!\7\222G"
  "\13&\2\42\10\65v\14\21*\2#\17\226E\234\240XL\202\342\305$(\26\0$\20\226E,\61"
  "\222\20\331\310D\244\204\220\11\1%\24\226E\234\60\221\20\221\240\300@\301H!\42!b!\0&\17"
  "\226E,\61\242hb$B$H#\1'\6\61v\212\1(\12\224\306+\21Ir\223\0)\13\224"
  "\306\13\61Ir\221\4\0*\17w\325\274\300\220hCGc!\21\303\0+\13fM,A!&A"
  "!\0,\11CF\33\21\22\21\0-\6&]\14\6.\6\42G\13\2/\15\226E\334@\301\210\202"
  "\21\5C\1\60\15\226E,\61\22!:\211\220\11\1\61\12\224\306\253 \221!\271 \62\15\226E\34"
  "\22$A!\22\22A\6\63\15\226E\14F\61\261Q\42\24\22\0\64\14\226E\14\21\371\205MP\12"
  "\0\65\14\226E\14\307\22C\42\24\22\0\66\14\226E\234\306\22$\42\24\22\0\67\12\226E\14Fi"
  "\62J\3\70\16\226E\34\22$\24\22!\42\24\22\0\71\16\226E\34\22$\42\224B\221\22\22\0:"
  "\7b\326\12B\2;\12s\305\32\211\42$\42\0<\11d\316+\21Ib\22=\10d\316\13t\20"
  "\4>\12d\316\13\61I\42\222\0\77\17\226E\34\22$\61A\61q\60A!\0@\15\226E,\61"
  "\22s\210&dB\0A\15\226E,\61\22$\242\207 \42\1B\15\226E\214\22\244\223\22!\242\26"
  "\0C\15\226E\34\22!\212\62\21\211\220\0D\13\226E\214\42\21\371\277\234\0E\13\226E\14\307\222"
  "\22AA\6F\13\226E\14\307\222\22A\31\1G\14\226E\34\22D)\225PH\0H\13\226E\14"
  "!J\17A\64\11I\11\226E\14&A\71\61J\13\226E\14&A\71\21\215\1K\17\226E\14!"
  "\22\224\61\301\61\42\221\42\1L\11\226E\14A\371\310\0M\13\226E\214\300\241\207 :\11N\15\226"
  "E\214\60\242\222\22\222J\324\2O\13\226E\34\22$:\241\220\0P\14\226E\214\22$\242\26A\211"
  "\0Q\13\226E\34\22$\272 \215\4R\14\226E\214\22$\242\26\231\210\4S\15\226E\34\22\304\242"
  "BA\24\22\0T\12\226E\14&A\371\4\0U\13\226E\14!\372\204B\2\0V\16\226E\14!"
  ":\211\4\5\221\11\12\1W\13\226E\14!:=\4\15\6X\21\226E\14!\42\221\240 \62\61\242"
  "(BD\2Y\16\226E\14!\42\221\240 \62A\71\1Z\11\226E\14F\61\71\62[\10\224F\14"
  "%\71!\134\16\226E\214P\321\340P\321\340P\321\0]\10\224F\14$\71)^\13D\356\33\221 "
  "\241\20\21\0_\6\26E\14\3a\16\270<\214\245\302a!\71K\222)\1b\16\270<\214\324a!"
  "\251\223\344,\252\0c\15\270<\14\305!\71\331$\323\240\2d\26\270<\214T\242\240\241\240\241\240\241"
  "\240\241\240\241\240\241 Q\5e\14\270<\214dI\246\223,\13'f\13\270<\214dI\246\223lj"
  "g\20\270<\14\305a!KaBaB\222\7\25h\13\270<\214$\371,I~Ji\14\270<\214"
  "da\42;\21\13'j\14\270<\214da\42;\21\222\61k\22\270<\214$Ia\242cEf\205"
  "DaBB\11l\12\270<\214$\373Y\70\1m\15\270<\214\24\22!\311\222\344\237\22n\24\270<"
  "\214\224!\331\202\304\202\204\302\204\302\204\344\64\222\0o\14\270<\14\305a!\371\317\203\12p\15\270<"
  "\214\324a!\71\213\16Yjq\15\270<\14\305a!\371\257d!\11r\20\270<\214\324a!\71\213"
  "\16\5\15\311\224\0s\17\270<\14\305a\221SR##\341A\5t\13\270<\214da\42\373\264\0"
  "u\13\270<\214$\371\377<\250\0v\15\270<\214$\371\237FD\10\213\26w\14\270<\214$\371\317"
  "\222\244\220$x\21\270<\214$\231FD\10\213\12ID\206dJy\16\270<\214$\231FD\10\213"
  "\354\264\0z\14\270<\214da\243:\31\13'{\15\226E\214\225\61\301P\301\261\22\3|\13\226E"
  ",Ay\62!\23\2}\15\226E\14\223\262A\321@\261\221\5\240\7\270<\214\177h\241\12\270<\214"
  "%{r\62Z\242\25\270<\214\245\302\221\240\20\221\240\302\242\20\221\240\220\301\242\5\243\22\270<\214\245"
  "B\22\221\242\302!Ia\242d!\11\244\16\270<\214\7\42#ba\42\243\7\6\246\13\270<\214%"
  "{\42\36,Z\250\15\270<\214\7\42\243\207#\243\7\6\251\17\270<\214\344\321\62\243\23\243\262Q\341"
  "\4\252\16\270<\214\7\42\243\207#\243\42#\6\253\13\270<\14EJ\366H\250\0\254\17\270<\214\224"
  "\243\23\224\243\23\224\243\223\4\256\17\270<\214d\321\261\42\223#\263\322\341\4\257\11\270<\214\77\24\77"
  "\10\260\16\270<\214\245\22\21\22\221\242\207\17\6\261\15\270<\214\207A\302\207B\302\207\5\266\15\270<"
  "\14\27#\23K&\226L\34\271\14\270<\214\207\242JF\227\36\26\272\14\270<\214\207\241KF\225\36"
  "\32\273\14\270<\14EB#;\21\22*\277\17\270<\14\305a!\241\42\243\42'\243\5\300\15\270<"
  "\214\245\302!\371\247\301\242\5\301\15\270<\14\26\243\62#\233\12\11\25\302\17\270<\14\305a!!#"
  "\302I\306\302\11\303\16\270<\214da\23\243\262#\231\7\25\304\22\270<\214\244\240\241\240\241\240\241\240"
  "a\341\42K\12\305\15\270<\214dI\246\303F\62\17*\306\16\270<\14Ua!\323a!\71\17*"
  "\307\14\270<\214da#\243\232,\61\310\17\270<\14\305a!\241A\302!\71\17*\311\16\270<\14"
  "\305a!\71\217\32\215\16*\327\22\271\274\214\207\23\223(e\344\0d%QL\36\16\367\25\270<\14"
  "\225\21\221H#\42\226LLD\206BBDF\24\0\0\0\4\377\377 \24\14\270<\214\17\313\302\17"
  "\17\3 \32\14\270<\214\37\214\214\212\214\30 \33\13\270<\14\225+\17\77\4 \36\17\270<\14\245"
  "\240\241\240\241\240\207\17\14 \42\12\270<\214\77\34\31- \71\14\270<\214\245Jv\272\264\0 :"
  "\14\270<\214\245Kv\252\264\0 \254\27\270<\14\25\21\22\221\341\21\21\22\21\22\221\341\21\21\22\21"
  "\5!\42\16\270<\214\245\302\341\207\345\301\242\5\0";

void onDedLine1Change(char* newValue)
{
[... 32 lines cut ...]
  u8g2.sendBuffer();
}
DcsBios::StringBuffer<25> dedLine4Buffer(0x454a, onDedLine4Change);

void onDedLine5Change(char* newValue) {
 
  u8g2.setFont(u8g2_font_falconded);
  u8g2.setCursor(0, 60);
  u8g2.print(newValue);
  u8g2.sendBuffer();
}
DcsBios::StringBuffer<25> dedLine5Buffer(0x4564, onDedLine5Change);

void setup() {
  u8g2.begin();

DcsBios::setup();
}
// fonts https://github.com/olikraus/u8g2/wiki/fntlistall#4-pixel-height

void loop()
{
  
  DcsBios::loop();
}

olikraus, 2021-05-26:
Not sure whether it is wise to call "U8glib.h", but independently from this, the string which you pass to onDedLine1Change() is still not visible.

Chroustall, 2021-05-26:
Still same issue with or without U8glib.h :/
#include <U8g2lib.h>
#define DCSBIOS_IRQ_SERIAL
#include "DcsBios.h"
#include <Wire.h>
#define D0 13
#define D1 11
#define DC 9
#define CS 8
#define RES 10

U8G2_SSD1309_128X64_NONAME0_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 8, /* dc=*/ 9, /* reset=*/ 10);  
const uint8_t u8g2_font_falconded_mr[963] U8G2_FONT_SECTION("u8g2_font_falconded_mr") = 
  "@\2\4\3\4\4\1\1\5\11\11\0\0\11\377\11\0\1\272\0\0\3\246 \7\231\347\307\63\0!\13"
  "\231\347\221\323{\10\71!\0\42\15\231g\221\20\222\20\222\220\307\26\0#\24\231gI\25\22d\24\22"
  "\25\22\25\22d\24\22\25\22\3$\24\231\347\21#\212\220\210\211\20#\223\210\211\220\10\42\23\2%\23"
  "\231g\211I\21!\24\22\27'\227$H\42M\32\0&\22\231\347\21\243\12\11\23#\22\221\221\230\251"
  "\232\10\1'\10\231g\11L\217\3(\12\231g\22S\247C\65\0)\12\231g\21T\247\63U\0*"
  "\20\231g\12\213H\65t\64\25\221,\36\16\0+\14\231\347\7\223\223\262\261\222\223\6,\13\231\347\307"
  "FNLN\12\0-\11\231\347Ggc\17\16.\11\231\347\307\231\234\20\0/\15\231g\213\223K'"
  "\227N.\16\0\60\23\231\347\21#\22\221\21\221\21\221\21\221\21\21\42\23\2\61\14\231g\212\23\233\323"
  "\31\25\15\0\62\17\231g!\262\21\221\223\42\42\222\263\61\1\63\17\231\347\260\261\23\23\33\224\21\221\61"
  "\242\1\64\21\231\347P$!$!$!dc&\247\6\0\65\17\231\347\260\261\221+\262\223\21\221\61"
  "\242\1\66\20\231g\251\261\221+\262\21\221\21\221\61\242\1\67\13\231\347\260\261S\246\235*\0\70\21\231"
  "g!\262\21\221\61\42\22\221\21\221\61\242\1\71\20\231g!\262\21\221\21\221\61\252\23*\242\1:\13"
  "\231\347\225\223\7\225\223\7\4;\15\231\347\207\222\223\207\220\23\223\23\3<\13\231\347\207\23S'\250\30"
  "\0=\12\231\347\207b\17HE\14>\13\231\347\207\22T'\246\34\0\77\17\231g!\262\21\21\223\23"
  "\223\207\220\23\2@\14\231\347\21#\262G`D&\4A\21\231\347\21#\262\21\221\21\221q#\42#"
  "\42\2B\21\231\347(\262\21\221)*\22\221\21\221\261\251\1C\21\231g!\22\221\21\221\221\323DF"
  "D\210\6\0D\25\231\347\250\222\20\222\20\222\20\222\20\222\20\222\20\222\220\251\1E\16\231\347\260\261\221"
  "+*\222\223\263\61\1F\15\231\347\260\261\221+*\222\323\14\0G\20\231g!\262\221S\61#\61#"
  "\42cD\3H\24\231\347\20\221\21\221\21\221q#\42#\42#\42#\42\2I\14\231\347\260\261\222\323"
  "\225\215\11\0J\14\231\347\260\261\222\323\25\325\24\0K\22\231\347\20\221\221\230)\32\223\33\243\222\230\21"
  "\21\1L\12\231\347\220\323w\66&\0M\24\231\347\10\212\21\221q#\42#\42#\42#\42#\42\2"
  "N\26\231\347\210Q\42\63!\63!#\61#\61#\42#\42#\23\2O\23\231g!\262\21\221\21\221"
  "\21\221\21\221\21\221\61\242\1P\17\231\347(\262\21\221\21\221\261)\222S\6Q\24\231g!\262\21\221"
  "\21\221\21\221\21\221\221\230\251\232\10\1R\23\231\347(\262\21\221\21\221\261)\222\20\22\221\21\21\1S"
  "\15\231g!\262\221\253\252ScD\3T\12\231\347\260\261\222\323G\0U\25\231\347\20\221\21\221\21\221"
  "\21\221\21\221\21\221\21\221\61\242\1V\24\231\347\20\221\21\221\21\221\21\221\21\21\12\211\42\223\23\2W"
  "\24\231\347\20\221\21\221\21\221\21\221\21\221q#\42\23\24\2X\24\231\347\20\221\21\21\12\211\42\23\243"
  "\12\11\22\221\21\21\1Y\17\231\347\20\221\21\21\12\211\42\223\323\21\0Z\14\231\347\260\261\23\323\235\215"
  "\11\0[\14\231g\241\242\222\323\35\25\15\0\134\16\231\347\10\24\214\14\24\214\14\24\14\1]\12\231g"
  "\241\242\323+\66\0^\15\231\347\21\13\211\12\11\223\307\10\0_\10\231\347\307\265\11\0\0\0\0\4\377"
  "\377\0";

void onDedLine1Change(char* newValue)
{
 
  u8g2.setFont(u8g2_font_falconded_mr);
  u8g2.setCursor(0, 20);
  u8g2.print(newValue);
  u8g2.sendBuffer();

  }
DcsBios::StringBuffer<25> dedLine1Buffer(0x44fc, onDedLine1Change);

void onDedLine2Change(char* newValue) {
  
  u8g2.setFont(u8g2_font_falconded_mr);
  u8g2.setCursor(0, 30);
  u8g2.print(newValue);
  u8g2.sendBuffer();
}
DcsBios::StringBuffer<25> dedLine2Buffer(0x4516, onDedLine2Change);

void onDedLine3Change(char* newValue) {
  
  u8g2.setFont(u8g2_font_falconded_mr);
  u8g2.setCursor(0, 40);
  u8g2.print(newValue);
  u8g2.sendBuffer();
}
DcsBios::StringBuffer<25> dedLine3Buffer(0x4530, onDedLine3Change);

void onDedLine4Change(char* newValue) {
  
  u8g2.setFont(u8g2_font_falconded_mr);
  u8g2.setCursor(0, 50);
  u8g2.print(newValue);
  u8g2.sendBuffer();
}
DcsBios::StringBuffer<25> dedLine4Buffer(0x454a, onDedLine4Change);

void onDedLine5Change(char* newValue) {
 
  u8g2.setFont(u8g2_font_falconded_mr);
  u8g2.setCursor(0, 60);
  u8g2.print(newValue);
  u8g2.sendBuffer();
}
DcsBios::StringBuffer<25> dedLine5Buffer(0x4564, onDedLine5Change);

void setup() {
  u8g2.begin();

DcsBios::setup();
}
// fonts https://github.com/olikraus/u8g2/wiki/fntlistall#4-pixel-height

void loop()
{
  
  DcsBios::loop();
}

Chroustall, 2021-05-26:
Hm, ok I´m little bit lost. If I deleted "U8glib.h", got same issues. It is possible to force in some way "u8g2_font_5x7_mr" or "falconded" to display "arrows" instead of "@"

olikraus, 2021-05-26:
> Ok I´m little bit lost

Same here. How can I help? I can just say, that there is not U8glib.h any more:
https://github.com/olikraus/u8g2/tree/master/cppsrc

It looks like you use some unknown (at least to me) environment (DCSBIOS?). I do not know how to support further...

Chroustall, 2021-05-26:
Yeah, sorry :). I'll try to figure it out. So, many thanks for font and everything.  :)

Chroustall, 2021-05-26:
I finnaly find the problem. Can you convert for me this font please?

Chroustall, 2021-05-26:
I finnaly found the problem! :) Can you please convert this edited font one more? 

[ef_conv_zip_u6F3dzo1PDYXazLD45OJWAeF.zip](https://github.com/olikraus/u8g2/files/6549363/ef_conv_zip_u6F3dzo1PDYXazLD45OJWAeF.zip)

olikraus, 2021-05-27:
``` C++
/*
  Fontname: -FreeType-FalconDED-Medium-R-Normal--17-120-100-100-P-68-ISO10646-1
  Copyright: Copyright uri_ba 2014
  Glyphs: 64/106
  BBX Build Mode: 1
*/
const uint8_t u8g2_font_falconded_hr[806] U8G2_FONT_SECTION("u8g2_font_falconded_hr") = 
  "@\1\3\3\4\4\1\1\5\10\11\0\0\11\0\11\0\1\214\0\0\3\11 \7\230\343\343\77\1!\12"
  "\226\333\11\365\16\23J\0\42\13\230cId\22\231D\216\37#\22\230c%\206b\266P\60\24\14\305"
  "l\241`(\4$\22\230\343Ii\21I(\42\245JB\21I\214*\3%\23\230c\305B\221\210,"
  "\24\16\213\223b\222\244X\10\0&\20\230\343I\211\241\250\224&R\31\25'\21\0'\10\224S\305\322"
  "\61\2(\11\227_j\325Y\5\0)\11\227_\211UuT\3*\16\231g\206#\251\303\343\64\222\34"
  "\307\2+\14\230\343\243\212\205&\243X\16\4,\12\226\333\343*\224\11E\0-\11\230\343\343j\262c"
  "\4.\10\226\333\343\233P\2/\14\230c\207\305\311\342dq\30\0\60\13\230\343Ii\42\375\33U\6"
  "\61\13\227_F\205S\35i\24\0\62\15\230c\321L\312B\32Ml\262\0\63\16\230\343\230\314R\351"
  "\34 R\262\221\0\64\17\230\343\250Id\22\231Df\262\212\225\0\65\15\230\343\230L\342\232Y\244d"
  "#\1\66\15\230c\225L\342\232I'\33\11\0\67\12\230\343\230\314\252\232\25\1\70\15\230c\321LJ"
  "\66\232H\223\215\4\71\15\230c\321L:\331\312\262\32\11\0:\13\225\327\3dr\260L\16\4;\14"
  "\225\327Cer\210L$\23\1<\12\227\337#j\25\253C\0=\13\227\337#\321\350h\354\20\0>"
  "\13\227\337#\211U\205\352\60\0\77\15\230c\321L\252b\251\34A,\3@\14\230\343Iiv\254\66"
  "\252\14\0A\13\230\343Ii&\235\234t\1B\15\230\343\324LJ\265\232H\223\251\4C\13\230c\321"
  "D:k\322F\2D\23\230\343\24%\62\211L\42\223\310$\62\211L\42*\1E\15\230\343\230L\342"
  "ZM,\66Y\0F\14\230\343\230L\342ZM\254\25\0G\16\230c\321Lb\225\221d\244d#\1"
  "H\12\230\343\210\364\344\244\277\0I\13\230\343\230\214b\35M\26\0J\13\230\343\230\214b\35\211C\0"
  "K\17\230\343\210TF\265\251xJ\224\214T\0L\11\230\343\210\365\331d\1M\13\230\343\4CJN"
  "\372/\0N\21\230\343\304DJ\23\321D$\31IFz\213\0O\13\230c\321L\372O\66\22\0P"
  "\14\230\343\324L:\231jbU\0Q\14\230c\321L\372\227Qq\22\1R\16\230\343\324L:\231j"
  "\22\231H\13\0S\14\230c\321L\342bY\311F\2T\11\230\343\230\214b}\3U\12\230\343\210\364"
  "\377d#\1V\14\230\343\210\364o\241 U,\3W\14\230\343\210\364ONJ\301\10\0X\17\230\343"
  "\210\264\205\202T)\61\24\23i\1Y\14\230\343\210\264\205\202T\261n\0Z\13\230\343\230\314R\235M"
  "\26\0[\13\230c\21\211b\235\211$\0\134\24\230\343\304\1r@\34\22\7\310\1qH\34 \7D"
  "\0]\11\230c\21\311zd\2^\14\227\337\11C\261PP\216'\0_\10\230\343\343\237,\0\0\0"
  "\0\4\377\377\0";
```

[image]

Chroustall, 2021-05-28:
Thanks a lot!!

Chroustall, 2021-05-28:
Thanks, but still is wierd. If i use this code i got same problem as with the previous code. I tested on simply "hello world" script (like on pictures). Arrows are still bind to "@".

Test:
[image]

Result on screen:
[image]

Edited font:
[image]

Zip:
[ef_conv_zip_cTx4NqW54UWMUL7Kd5itmAeF.zip](https://github.com/olikraus/u8g2/files/6560710/ef_conv_zip_cTx4NqW54UWMUL7Kd5itmAeF.zip)

rcknr, 2021-09-20:
How do you generate font previews that you paste as images?

olikraus, 2021-09-20:
@Chroustall I think I missed your last post. The "a" might be missing, because does not appear in the preview picture.

@rcknr bdfconv (see U8g2 FAQ) can generate such preview pictures if the -d option is provided.

KPCKxd, 2021-11-26:
Hi, I may know where is the problem. Could you please convert this ttf to u8g2? I tried to do it myself but had no luck doing that
[ef_conv_zip_hi_t7s8Nupgwi0999MH7uAeF.zip](https://github.com/olikraus/u8g2/files/7607806/ef_conv_zip_hi_t7s8Nupgwi0999MH7uAeF.zip)
.

olikraus, 2021-11-26:
Did you use otf2bdf ?
It usually works without problems.

KPCKxd, 2021-12-13:
Sorry for the late response. Here should be generated bdf file of the font.

[FalconDED-9.zip](https://github.com/olikraus/u8g2/files/7705268/FalconDED-9.zip)

olikraus, 2022-03-04:
Hmm guess I missed this somehow
```
/*
  Fontname: -FreeType-FalconDED-Medium-R-Normal--17-120-100-100-P-68-ISO10646-1
  Copyright: Copyright uri_ba 2014
  Glyphs: 65/106
  BBX Build Mode: 2
*/
const uint8_t u8g2_font_falconded_mr[975] U8G2_FONT_SECTION("u8g2_font_falconded_mr") = 
  "A\2\4\3\4\4\1\1\5\11\11\0\0\11\0\11\0\1\272\3\244\3\262 \7\231\347\307\63\0!\13"
  "\231\347\221\323{\10\71!\0\42\15\231g\221\20\222\20\222\220\307\26\0#\24\231gI\25\22d\24\22"
  "\25\22\25\22d\24\22\25\22\3$\24\231\347\21#\212\220\210\211\20#\223\210\211\220\10\42\23\2%\23"
  "\231g\211I\21!\24\22\27'\227$H\42M\32\0&\22\231\347\21\243\12\11\23#\22\221\221\230\251"
  "\232\10\1'\10\231g\11L\217\3(\12\231g\22S\247C\65\0)\12\231g\21T\247\63U\0*"
  "\20\231g\12\213H\65t\64\25\221,\36\16\0+\14\231\347\7\223\223\262\261\222\223\6,\13\231\347\307"
  "FNLN\12\0-\11\231\347Ggc\17\16.\11\231\347\307\231\234\20\0/\15\231g\213\223K'"
  "\227N.\16\0\60\23\231\347\21#\22\221\21\221\21\221\21\221\21\21\42\23\2\61\14\231g\212\23\233\323"
  "\31\25\15\0\62\17\231g!\262\21\221\223\42\42\222\263\61\1\63\17\231\347\260\261\23\23\33\224\21\221\61"
  "\242\1\64\21\231\347P$!$!$!dc&\247\6\0\65\17\231\347\260\261\221+\262\223\21\221\61"
  "\242\1\66\20\231g\251\261\221+\262\21\221\21\221\61\242\1\67\13\231\347\260\261S\246\235*\0\70\21\231"
  "g!\262\21\221\61\42\22\221\21\221\61\242\1\71\20\231g!\262\21\221\21\221\61\252\23*\242\1:\13"
  "\231\347\225\223\7\225\223\7\4;\15\231\347\207\222\223\207\220\23\223\23\3<\13\231\347\207\23S'\250\30"
  "\0=\12\231\347\207b\17HE\14>\13\231\347\207\22T'\246\34\0\77\17\231g!\262\21\21\223\23"
  "\223\207\220\23\2@\14\231\347\21#\262G`D&\4A\21\231\347\21#\262\21\221\21\221q#\42#"
  "\42\2B\21\231\347(\262\21\221)*\22\221\21\221\261\251\1C\21\231g!\22\221\21\221\221\323DF"
  "D\210\6\0D\25\231\347\250\222\20\222\20\222\20\222\20\222\20\222\20\222\220\251\1E\16\231\347\260\261\221"
  "+*\222\223\263\61\1F\15\231\347\260\261\221+*\222\323\14\0G\20\231g!\262\221S\61#\61#"
  "\42cD\3H\24\231\347\20\221\21\221\21\221q#\42#\42#\42#\42\2I\14\231\347\260\261\222\323"
  "\225\215\11\0J\14\231\347\260\261\222\323\25\325\24\0K\22\231\347\20\221\221\230)\32\223\33\243\222\230\21"
  "\21\1L\12\231\347\220\323w\66&\0M\24\231\347\10\212\21\221q#\42#\42#\42#\42#\42\2"
  "N\26\231\347\210Q\42\63!\63!#\61#\61#\42#\42#\23\2O\23\231g!\262\21\221\21\221"
  "\21\221\21\221\21\221\61\242\1P\17\231\347(\262\21\221\21\221\261)\222S\6Q\24\231g!\262\21\221"
  "\21\221\21\221\21\221\221\230\251\232\10\1R\23\231\347(\262\21\221\21\221\261)\222\20\22\221\21\21\1S"
  "\15\231g!\262\221\253\252ScD\3T\12\231\347\260\261\222\323G\0U\25\231\347\20\221\21\221\21\221"
  "\21\221\21\221\21\221\21\221\61\242\1V\24\231\347\20\221\21\221\21\221\21\221\21\21\12\211\42\223\23\2W"
  "\24\231\347\20\221\21\221\21\221\21\221\21\221q#\42\23\24\2X\24\231\347\20\221\21\21\12\211\42\23\243"
  "\12\11\22\221\21\21\1Y\17\231\347\20\221\21\21\12\211\42\223\323\21\0Z\14\231\347\260\261\23\323\235\215"
  "\11\0[\14\231g\241\242\222\323\35\25\15\0\134\16\231\347\10\24\214\14\24\214\14\24\14\1]\12\231g"
  "\241\242\323+\66\0^\15\231\347\21\13\211\12\11\223\307\10\0_\10\231\347\307\265\11\0a\14\231\347\21"
  "#\262G`D&\4\0\0\0\4\377\377\0";

/*
  Fontname: -FreeType-FalconDED-Medium-R-Normal--17-120-100-100-P-68-ISO10646-1
  Copyright: Copyright uri_ba 2014
  Glyphs: 65/106
  BBX Build Mode: 1
*/
const uint8_t u8g2_font_falconded_hr[818] U8G2_FONT_SECTION("u8g2_font_falconded_hr") = 
  "A\1\3\3\4\4\1\1\5\10\11\0\0\11\0\11\0\1\214\3\7\3\25 \7\230\343\343\77\1!\12"
  "\226\333\11\365\16\23J\0\42\13\230cId\22\231D\216\37#\22\230c%\206b\266P\60\24\14\305"
  "l\241`(\4$\22\230\343Ii\21I(\42\245JB\21I\214*\3%\23\230c\305B\221\210,"
  "\24\16\213\223b\222\244X\10\0&\20\230\343I\211\241\250\224&R\31\25'\21\0'\10\224S\305\322"
  "\61\2(\11\227_j\325Y\5\0)\11\227_\211UuT\3*\16\231g\206#\251\303\343\64\222\34"
  "\307\2+\14\230\343\243\212\205&\243X\16\4,\12\226\333\343*\224\11E\0-\11\230\343\343j\262c"
  "\4.\10\226\333\343\233P\2/\14\230c\207\305\311\342dq\30\0\60\13\230\343Ii\42\375\33U\6"
  "\61\13\227_F\205S\35i\24\0\62\15\230c\321L\312B\32Ml\262\0\63\16\230\343\230\314R\351"
  "\34 R\262\221\0\64\17\230\343\250Id\22\231Df\262\212\225\0\65\15\230\343\230L\342\232Y\244d"
  "#\1\66\15\230c\225L\342\232I'\33\11\0\67\12\230\343\230\314\252\232\25\1\70\15\230c\321LJ"
  "\66\232H\223\215\4\71\15\230c\321L:\331\312\262\32\11\0:\13\225\327\3dr\260L\16\4;\14"
  "\225\327Cer\210L$\23\1<\12\227\337#j\25\253C\0=\13\227\337#\321\350h\354\20\0>"
  "\13\227\337#\211U\205\352\60\0\77\15\230c\321L\252b\251\34A,\3@\14\230\343Iiv\254\66"
  "\252\14\0A\13\230\343Ii&\235\234t\1B\15\230\343\324LJ\265\232H\223\251\4C\13\230c\321"
  "D:k\322F\2D\23\230\343\24%\62\211L\42\223\310$\62\211L\42*\1E\15\230\343\230L\342"
  "ZM,\66Y\0F\14\230\343\230L\342ZM\254\25\0G\16\230c\321Lb\225\221d\244d#\1"
  "H\12\230\343\210\364\344\244\277\0I\13\230\343\230\214b\35M\26\0J\13\230\343\230\214b\35\211C\0"
  "K\17\230\343\210TF\265\251xJ\224\214T\0L\11\230\343\210\365\331d\1M\13\230\343\4CJN"
  "\372/\0N\21\230\343\304DJ\23\321D$\31IFz\213\0O\13\230c\321L\372O\66\22\0P"
  "\14\230\343\324L:\231jbU\0Q\14\230c\321L\372\227Qq\22\1R\16\230\343\324L:\231j"
  "\22\231H\13\0S\14\230c\321L\342bY\311F\2T\11\230\343\230\214b}\3U\12\230\343\210\364"
  "\377d#\1V\14\230\343\210\364o\241 U,\3W\14\230\343\210\364ONJ\301\10\0X\17\230\343"
  "\210\264\205\202T)\61\24\23i\1Y\14\230\343\210\264\205\202T\261n\0Z\13\230\343\230\314R\235M"
  "\26\0[\13\230c\21\211b\235\211$\0\134\24\230\343\304\1r@\34\22\7\310\1qH\34 \7D"
  "\0]\11\230c\21\311zd\2^\14\227\337\11C\261PP\216'\0_\10\230\343\343\237,\0a\14"
  "\230\343Iiv\254\66\252\14\0\0\0\0\4\377\377\0";
```

clzr111, 2025-09-16:
Can anyone make a font with the arrows working and the inverted letters, I need it so much but I dont know how to make it myself.

olikraus, 2025-09-21:
Here is the complete font:
```
/*
  Fontname: -FontForge-FalconDED-Book-R-Normal--9-90-75-75-P-47-ISO10646-1
  Copyright: Copyright uri_ba 2014
  Glyphs: 105/108
  BBX Build Mode: 2
*/
const uint8_t u8g2_font_falconded_all[1017] U8G2_FONT_SECTION("u8g2_font_falconded_all") = 
  "i\2\3\3\3\3\1\1\4\5\7\0\377\5\0\5\0\1\42\2_\3\210 \5}}<!\10}\375"
  "`:(\16\42\11}\355H(\22\307\0#\12}\355\340(\70\212C\0$\11}\375\240H(\213\3"
  "%\12}\355H(\32\213\203\1&\11}\375XP\66\32\3'\10}\355`\34\23\0(\10}\375X"
  "\60\65\16)\11}ma\266\70\4\0*\10}\355Y\60\35\12+\11}}hl\26\207\2,\11}"
  "},\261\70\4\0-\10}}\244\71\22\0.\7}}\314q\0/\12}}H\34\22\213\203\1\60"
  "\14}\375X$\24\11Ebq\0\61\10}\375\230\60M\16\62\12}\355Q$&\12\216\1\63\11}"
  "\355a,%$\7\64\11}m\231l\26\214\3\65\12}\355Qp\24\11\311\1\66\12}\355Qp\24"
  "\11\311\1\67\11}\355a,\61\16\1\70\13}\355Q$$\213\204\344\0\71\12}\355Q$\64\14\311"
  "\1:\11}\355`\34\24\7\3;\12}}`\34\24\214C\0<\11}}h,\32\207\2=\11}"
  "}\240\34\42\207\2>\11}}`\64\30\207\2\77\12}\355Q$\26\7\305\1A\13}\375\330(\22"
  "\32E\302\0B\13}\355Q$$\213\204\344\0C\14}\355H(\22\12FBr\0D\15}\355H"
  "(\22\212\204\42!\71\0E\11}\355QP\26\34\3F\12}\355QP\26\214C\0G\12}\355Q"
  "p\24\11\311\1H\15}\355H(\22\32EB\221\60\0I\10}\355Y\60m\14J\11}\355Y\60"
  "-\16\1K\13}\355HH\26\224E\302\0L\10}\355`\306\61\0M\15}\355Hh\24\11EB"
  "\221\60\0N\15}\355H(\22\32EB\221\60\0O\14}\355Q$\24\11EBr\0P\13}\355"
  "Q$\64\12\306!\0Q\12}\355Q$\24\11-\3R\13}\355Q$\64\222E\302\0S\11}\355"
  "Qp\30\222\3T\10}\355Y\60;\0U\15}\355H(\22\212\204\42!\71\0V\14}\355H("
  "\22\212\204\204q\0W\14}\355H(\22\212\204\206a\0X\13}\355H(\32\13F\302\0Y\11}"
  "\355H(\32L\7Z\10}\355a,q\14[\10}mY\60Q\16\134\11}\355`\64\16\13\3]"
  "\10}ma\66\71\0^\10}mY\34\23\0_\7}}\254c\0a\10}\375\330|\26\7\240\5"
  "}\215\77\241\10}\215\13\313Ar\242\11}\15\23ID\232\34\243\11}\15\323\204\64\223\31\244\11}"
  "\215\13e\66\71\24\246\10}\215\13\313lr\250\11}\215\13\345 \71\24\251\11}\15\333\210\62\222\31"
  "\252\12}\215\13\345 \231\34\2\253\10}\15\323\204\323\1\254\11}\215\3\245r\230\30\256\11}\15\233"
  "\344 \241\31\257\5}\215\77\260\10}\15\13\345\360 \261\11}\15\323\350\20:\24\266\12}\215C\344"
  "\20\231\34\14\271\11}\215\13eR\71\24\272\11}\15K\205\62\71\30\273\10}\15\23\205et\277\12"
  "}\15\323$\62\71H\16\300\14}\15\323$\42\211H\42\223\3\301\11}\215\313\210B\31\35\302\12}"
  "\15\323$\62\222\320\14\303\12}\15\333!\62\211\210\16\304\10}\15\323\332\204r\305\12}\15\233\204&"
  "\211\210\16\306\12}\15\233\204&\211\210\16\307\11}\15\33e\205r\10\310\13}\15\323$\42\232DD"
  "\7\311\12}\15\323$\42\243\210\16\327\10}\215\313L\355@\367\12}\215CDR\231\34\14\0\0\0"
  "\4\377\377 \24\11}\215\7\331\341  \32\12}\215\17\221\311!\0 \33\11}\15\13\345\360  "
  "\36\12}\15KD\22\71< \42\10}\215\17\226\3 \71\11}\15\323\204\245r :\11}\215K"
  "\205er \254\13}\15\13MB\223\34\2!\42\11}\15\323\354\66\71\0";
```

[image]