olikraus/u8g2 #1504: Recommendation request: u8g2-compatible display able to dim to low brightness

DrTron, 2021-05-25:
Hello Everybody,

I'm currently using a noname SSD1309 OLED 2.42" 128x64 display, which works very well with u8g2.
Sadly, it does react to a u8g2.setContrast command, but the brightness difference (0-255) is rather limited. As that is used in a bedroom environment, even on "0" it's way too bright at night.

Installing a clear cover with car tinting foils is one option, but ultimately I'd like to control the brightness by ambient light. Besides, I suppose the OLED's will last a lot longer if they would dim themselves.

Does anyone know of a display that can be dimmed to really low levels? I found some LCD's that are supposed to be able to do that, but I'm not sure how readable those are with the contrast via backlight down. Technically, an OLED display should be able to do that, at least the OLED in my phone can. So I guess it's all about finding the right compatible display.

Any ideas?

Thanks!

olikraus, 2021-05-25:
This was discussed already for the SSD1306:
https://github.com/olikraus/u8g2/issues/98 and https://github.com/olikraus/u8g2/issues/636

I assume that the SSD1309 includes similar registers, so maybe the approch in #636 might work also for the SSD1309.

Also note that u8g2 now includes the sendF (https://github.com/olikraus/u8g2/wiki/u8g2reference#sendf) command, which would simplify the code from #636 

NOT TESTED:

```
// value from 0 to 7, higher values more brighter
void setSSD1306VcomDeselect(uint8_t v)
{	
  u8g2.sendF("ca", 0x0db, v << 4);
}

// p1: 1..15, higher values, more darker, however almost no difference with 3 or more
// p2: 1..15, higher values, more brighter
void setSSD1306PreChargePeriod(uint8_t p1, uint8_t p2)
{	
  u8g2.sendF("ca", 0x0d9, (p2 << 4) | p1 );
}
```

DrTron, 2021-05-25:
Thank you very much, Oli, I must have missed those threads during my search. I'll try this approach and see if the SSD1309 is similar in this respect to the SD1306.

It seems from the other threads that the Vcom deselect gives the most range when set to "0".
Do you have any suggestions where to start with the Precharge? The description suggests that p1 and p2 both influence the brightness, but are opposed to each other?

olikraus, 2021-05-25:
You could just start with most darkest values. If your display is unstable (flicker) increase vcom value. Then adjust your preferred brightness.

olikraus, 2021-05-25:
Additionally set the contrast value to 0

DrTron, 2021-05-25:
I'll try that, thanks!

DrTron, 2021-05-26:
Hi Oli,

that worked perfectly, thanks a lot. I had to tweak the values a bit to prevent flickering but a vcom value of 0 was stable. vcom = 1 was already noticeably brighter. It was mainly the Precharge values that needed to be a bit higher than the minimum to get rid of the flickering.

So I can confirm that this works on SSD1309 displays, too. I suppose that makes sense as I do expect most displays out there to use the same controller, regardless of the size.

olikraus, 2021-05-26:
Sounds good and thanks for the feedback.

The SSD1309 is actually the big brother of the SSD1306. They should share the same commands, but the SSD1309 can drive higher currents, so the SSD1309 is the choice for OLEDs > 1.5".

closed 2021-07-02 by olikraus

olikraus, 2022-05-26:
For SH1106 the above vcomh procedure will not work, because the SH1106 seems to turn of the display if the 0xdb command is received. The vcomh setup should look like this instead:

```
// value from 0 to 7, higher values more brighter
// send a 0xaf (display on) command, which seems to be required for the SH1106
void setSSD1306VcomDeselect(uint8_t v)
{	
  u8g2.sendF("cac", 0x0db, v << 4, 0xaf);
}
```