olikraus/u8g2 #1505: Eastern Rising Display SSD1322 problems

adkeco, 2021-05-26:
Hello,

Apologies if I've done something stupid and apologies for adding to the 93 issues already :( 

I seem to be having trouble getting my OLED display to work using 4 wire HW SPI on my Arduino Mega. To test I am using the "Hello World" program in the u8g2 library examples (full buffer) using the software SPI first.

Uncommenting this constructor:

U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 52, / data=/ 51, / cs=/ 10, / dc=/ 9, / reset=*/ 8); // Enable U8G2_16BIT in u8g2.h

The SW SPI works flawlessly, but I want to try and get HW SPI working for speed reasons.

When I swap to this constructor:

U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=/ 10, / dc=/ 9, / reset=*/ 8); // Enable U8G2_16BIT in u8g2.h

I get something displaying like the following (apologies this screen shot is not from Hello World but from another program I was testing with to demonstrate full width printing, proving 16bit enabled)

[image]

I also tried the example Arduino code that was on the page where I bought the display from:

https://www.buydisplay.com/white-3-2-inch-arduino-raspberry-pi-oled-display-module-256x64-spi

That also works well and shows full width.

Hoping you can help me.. cheers,
Adam

olikraus, 2021-05-26:
You could try to adjust the timing values:
https://github.com/olikraus/u8g2/blob/c7dc077995ca513337a7d2a1970df693e18ee308/csrc/u8x8_d_ssd1322.c#L247-L254
Try to increase the time for post_chip_enable_wait_ns
Also try different spi modes (modes are from 0 to 3)

adkeco, 2021-05-26:
Thanks for the suggestions Oliver. 

I tried 40, 20...... 80,40....... 160, 80........320,160 respectively for post_chip and pre_chip values, each with SPI Mode switching from 0 to 3..

There was slight change in behaviour but still not good... instead of the fuzzy bottom half like an old tube TV, its now gone completely white background, consistent to the top half (its still showing the top half squished with colour inversion).

Cheers,

olikraus, 2021-05-26:
Maybe you also need to decrease the transfer speed: sck_clock_hz

adkeco, 2021-05-28:
Hi Oliver, 
Unfortunately setting it to 8000000, or 9000000 via the high level command didn't change the behaviour.
Cheers,

olikraus, 2021-05-28:
Did you also test very low values, like 100000?

adkeco, 2021-05-28:
Is there perhaps something you might be able to spot examining their example code below???
I noticed a reference to SPI.setClockDivider(SPI_CLOCK_DIV128). I tried incorporating this into my code but it didn't seem to make a difference.

`/***************************************************
//Web: http://www.buydisplay.com
EastRising Technology Co.,LTD
****************************************************/
#include <SPI.h>
#include "er_oled.h"

void command(uint8_t cmd)
{
  digitalWrite(OLED_DC, LOW);
  digitalWrite(OLED_CS, LOW);
  SPI.transfer(cmd);
  digitalWrite(OLED_CS, HIGH);
}

void data(uint8_t dat)
{
  digitalWrite(OLED_DC, HIGH);
  digitalWrite(OLED_CS, LOW );
  SPI.transfer(dat);
  digitalWrite(OLED_CS, HIGH);
}

void er_oled_begin()
{
  pinMode(OLED_RST, OUTPUT);
  pinMode(OLED_DC, OUTPUT);
  pinMode(OLED_CS, OUTPUT);
  SPI.begin();
    
  SPI.setClockDivider(SPI_CLOCK_DIV128);
    
  digitalWrite(OLED_CS, LOW);
  digitalWrite(OLED_RST, HIGH);
  delay(10);
  digitalWrite(OLED_RST, LOW);
  delay(10);
  digitalWrite(OLED_RST, HIGH);
    
  
  command(0xFD); /*SET COMMAND LOCK*/ 
  data(0x12); /* UNLOCK */ 
  command(0xAE); /*DISPLAY OFF*/ 
  command(0xB3);/*DISPLAYDIVIDE CLOCKRADIO/OSCILLATAR FREQUANCY*/ 
  data(0x91); command(0xCA); /*multiplex ratio*/ 
  data(0x3F); /*duty = 1/64*/ 
  command(0xA2); /*set offset*/ 
  data(0x00);
  command(0xA1); /*start line*/ 
  data(0x00); 
  command(0xA0); /*set remap*/
  data(0x14); 
  data(0x11);
  	
  command(0xAB); /*funtion selection*/ 
  data(0x01); /* selection external vdd */ 
  command(0xB4); /* */ 
  data(0xA0);
  data(0xfd); 
  command(0xC1); /*set contrast current */ 
  data(0x80); 
  command(0xC7); /*master contrast current control*/ 
  data(0x0f); 
  	
  command(0xB1); /*SET PHASE LENGTH*/
  data(0xE2); 
  command(0xD1); /**/
  data(0x82); 
  data(0x20); 
  command(0xBB); /*SET PRE-CHANGE VOLTAGE*/ 
  data(0x1F);
  command(0xB6); /*SET SECOND PRE-CHARGE PERIOD*/
  data(0x08); 
  command(0xBE); /* SET VCOMH */ 
  data(0x07); 
  command(0xA6); /*normal display*/ 
  command(0xAF); /*display ON*/  
}

void er_oled_SetWindow(uint8_t Xstart, uint8_t Ystart, uint8_t Xend, uint8_t Yend)
{ 
  command(0x15);
  data(Xstart+0x1c);
  data(Xend+0x1c);
  command(0x75);
  data(Ystart);
  data(Yend);
  command(0x5c);//write ram command
}

void er_oled_clear()
{int i,row;
  command(0x15);
  data(0x00); //col start
  data(0x77); //col end 
  command(0x75);
  data(0x00); //row start
  data(0x7f);  //row end 
  command(0x5c); 
  for (row = 0; row < 128; row++) {              
        for(i = 0; i< 240; i++ ) {
          data(0x00);// write data       
        }        
  }
}

void er_oled_char(uint8_t x, uint8_t y, const char  *acsii, uint8_t mode)
{ uint8_t i,str;uint16_t OffSet;
  x=x/4;
  OffSet = (*acsii - 32)*16;
  er_oled_SetWindow(x, y, x+1, y+15);
  for (i=0;i<16;i++)
  {     str =pgm_read_byte(&AsciiLib[OffSet + i]);  
        if(mode) str=~str;
         Data_processing (str);             
  }
}

void er_oled_string(uint8_t x, uint8_t y, const char *pString,  uint8_t Mode)
{     
  while(1)
  {
        if (*pString == 0)
        {
            return;
        }
            er_oled_char(x, y, pString,Mode);
            x += 8;
            pString += 1;              
  }   
}

void Data_processing(uint8_t temp)  //turns 1byte B/W data to 4 bye gray data  with 8 Pixel
{uint8_t temp1,temp2;

  if(temp&0x80)temp1=0xf0;
  else temp1=0x00;
  if(temp&0x40)temp2=0x0f;
  else temp2=0x00;
  temp1=temp1|temp2;
  data(temp1); //Pixel1,Pixel2
  if(temp&0x20)temp1=0xf0;
  else temp1=0x00;
  if(temp&0x10)temp2=0x0f;
  else temp2=0x00;
  temp1=temp1|temp2;
  data(temp1);  //Pixel3,Pixel4                
  if(temp&0x08)temp1=0xf0;
  else temp1=0x00;
  if(temp&0x04)temp2=0x0f;
  else temp2=0x00;
  temp1=temp1|temp2;
  data(temp1);  //Pixel5,Pixel6               
  if(temp&0x02)temp1=0xf0;
  else temp1=0x00;
  if(temp&0x01)temp2=0x0f;
  else temp2=0x00;
  temp1=temp1|temp2;
  data(temp1);  //Pixel7,Pixel8               
}

void er_oled_bitmap_mono(const uint8_t * pBuf)
{ uint8_t row,col,dat; 
  er_oled_SetWindow(0, 0, 255/4, 63); 
  for (row = 0; row < 64; row++) {              
        for(col = 0;col<256/8; col++ ) {
        dat=(pgm_read_byte(pBuf));
        *pBuf++;  
        Data_processing(dat);               
        }
  }    	
}

void er_oled_bitmap_gray(const uint8_t * pBuf)
{   uint8_t row,col; 
  er_oled_SetWindow(0, 0, 255/4, 63);
  for (row = 0; row < 64; row++) {              
        for(col = 0;col<128; col++ ) {
        data(pgm_read_byte(pBuf));
        * pBuf++;
        }
  }    	
}

`

adkeco, 2021-05-28:
I didnt test the low values yet as the documentation stated:

For SPI try values between "u8g2.setBusClock(1000000);" and "u8g2.setBusClock(8000000);".

I will try them now.

adkeco, 2021-05-28:
Its working! :) Setting it to 100000 worked and was painfully slow, I then kept upping it 100000. I got to 1000000 and it was still working. I went to 2000000 and it stopped. So far it seems stable at 1900000.

adkeco, 2021-05-28:
Thank you so much for your help. Hopefully this is fast enough for the application I am using it for (rpm display mostly)

closed 2021-05-28 by adkeco