olikraus/u8g2 #1506: increase brighness on SH1106 128x64

loicguidoux, 2021-05-28:
Hello,
Thanks for the versatile font library control.

I am using as oled type U8G2_SH1106_128X64_VCOMH0_F_HW_I2C.

I would like to increase brightness.

I increased the contrast to max value, but still not enough to read display in sunlight.
I tried your suggestions by including in my ino, the functions:
setSSD1306VcomDeselect and setSSD1306PreChargePeriod.
the first one makes the screen black and noting displayed.
the second one has no effect

I wonder if the register setting address are right for SH1106.

Thanks for recommendations.

olikraus, 2021-05-28:
The sh1106 may indeed have different registers/ commands. However I assume those little OLED are not suitable for sunlight. Try reflective LCDs instead.

loicguidoux, 2021-05-28:
Thanks for quick answer. I will change the LCD.
Your work is very useful. Thanks.

closed 2021-05-30 by loicguidoux