olikraus/u8g2 #1512: Scroll text example non in main loop

paolovr70, 2021-06-06:
Hi I'm wondering if you can add an example for Scrolling text NON in main loop. I HAVE.ADAPTED the example from the library but after a few.hours arduino DUE stop working due to SPI OLED crash. Can you prepare an example'that handle a.text that can change DUE nd.the routine.for displaying a message? Thanks. P.

olikraus, 2021-06-06:
You could move the code into a sub procedure and call the same from main.

paolovr70, 2021-06-06:
Yes in a sub routine i fine, but the problem is that you have to scroll 1 pixel at any cycle. And probably changing text something goes wrong. Do you want.to prepare something? Thanks. P.Inviato dal mio smartphone LG

olikraus, 2021-06-06:
Did you study the Weather example?
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/Weather/Weather.ino

paolovr70, 2021-06-06:
I.did.now IA look at that example. but.it.seems that it display sequentially 4/5 messages.In my case the MAIN LOOP continues to.Do things.And only once per loop there must be the scroll of 1 pixel. than the program has.to go on and the next loop it scrolls again 1'pixel and so on.P.Inviato dal mio smartphone LG

https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/Weather/Weather.ino

olikraus, 2021-06-06:
Well, I do not know how to help here. If there is any bug in u8g2 i can help better.

paolovr70, 2021-06-06:
Do you know.if there.is.someone that has prepared a code for scrolling text 1 pixel.at a.time in a.sub routine?Inviato dal mio smartphone LG

olikraus, 2021-06-06:
You can just change the offset of the weather example to 1:

https://github.com/olikraus/u8g2/blob/a1d90dbdc6e263228d071ed531fa7271cdae257f/sys/arduino/u8g2_page_buffer/Weather/Weather.ino#L406

Then it will scroll per 1 pixel and it is in a sub procedure.

paolovr70, 2021-06-06:
The problem is not 1or 2 pixel. Looking at the code drawscrollstring function scroll the entire string and only after let the program to go on. I want to scroll 1 or 2 pixel only at any loop in order to perform many other tasks.Inviato dal mio smartphone LG

https://github.com/olikraus/u8g2/blob/a1d90dbdc6e263228d071ed531fa7271cdae257f/sys/arduino/u8g2_page_buffer/Weather/Weather.ino#L406

olikraus, 2021-06-06:
You could execute your other tasks right after the offset calculation:

https://github.com/olikraus/u8g2/blob/a1d90dbdc6e263228d071ed531fa7271cdae257f/sys/arduino/u8g2_page_buffer/Weather/Weather.ino#L406

I still do not get the problem.

paolovr70, 2021-06-10:
HI OLIKRAUS I am trying to isolate the problem with my project. At a.first look I thought it was the rolling.text. But.the problem.still exist even if I remove that code. The fact is that if I disconnect.the display the problem is solved and both the Lan (wiznet 5500) and the sd card work.(they use HW SPI). I'm using OLED ST7565 with SW SPI and your library. I am wondering if the problem.is the.use if.SW SPI together with HW SPI.DO I have to use only HW SPI?Thanks.P.Inviato dal mio smartphone LG

https://github.com/olikraus/u8g2/blob/a1d90dbdc6e263228d071ed531fa7271cdae257f/sys/arduino/u8g2_page_buffer/Weather/Weather.ino#L406

olikraus, 2021-06-10:
If the pins are shared, then you must use hw SPI for the display also.

closed 2022-03-04 by olikraus