olikraus/u8g2 #1514: HW I2C with chinese SSD1306

aut0, 2021-06-10:
Hi,

After a bit of debugging I figured out why my SSD1306 display is not working with u8g2.
I use the following program:
```C
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <Wire.h>

U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);

void setup(void) {
  u8g2.setI2CAddress(0x3c);
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
}
```
It doesn't work out of the box. I noticed this: [https://github.com/olikraus/u8g2/blob/f2d2d93662e437c71970b19cb824f8311fff0259/cppsrc/U8x8lib.cpp#L1336](url)

Why is the address shifted? When I remove the shift it works as expected.

BR,
Lukas

olikraus, 2021-06-10:
> Why is the address shifted?

Because I programmed it that way.

> When I remove the shift it works as expected.

Maybe it fits to your expectation, but it then it will not fit any more to the documentation: https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress 
The address must have the lowest bit set to 0 and aligned to the MSB.

And then, what would a user of this display say? Wouldn't this use complain if he would need to shift the address?

[image]

See also here: https://github.com/olikraus/u8g2/issues/821

aut0, 2021-06-11:
> > Why is the address shifted?
> 
> Because I programmed it that way.

I see. I just found it confusing at first that u8g2 uses a different address format than Wire.

> 
> > When I remove the shift it works as expected.
> 
> Maybe it fits to your expectation, but it then it will not fit any more to the documentation: https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress
> The address must have the lowest bit set to 0 and aligned to the MSB.

I agree: RTFM ;-).

> And then, what would a user of this display say? Wouldn't this user complain if he would need to shift the address?
> 
> [image]

Mhm, yeah this whole 7 bit address thing is a pickle. I guess I'd do what Wire does and make this user shift the address, but I see where you are coming from. I'd do this as Wire is somewhat "standard" in the Arduino eco system.
Also regarding u8g2 documentation, as stated in the manual the address parameter taken by the setI2CAdress function is not the address, but the address multiplied by two. So when just looking at the function signature, I'd assume I have to pass the address and not (at least in my case) the address multiplied by two.

Anyways, thanks. I learned a bit more about I2C.

BR,
Lukas

closed 2021-06-11 by aut0

electronicsguy, 2025-06-05:
@olikraus I just came across this. Oli, I believe you are wrong in this. The standard I2C address is 7-bit not 8-bit. The last bit is for what command you want to do for that address (R/W)

[i2c](https://www.totalphase.com/support/articles/200349176-7-bit-8-bit-and-10-bit-i2c-slave-addressing/#7bit)
So even though these cheap I2C displays say their address is "0x78", the actuall address is "0x3C". That's the only valid I2C address.

0x3c = [0011 110] (only 7-bits considered)

[ 0011 110 0] = write to 0x3c
[ 0011 110 1] = read from 0x3c

You shouldn't left or write shift the address, if set using `u8g2.setI2CAddress()`. It should use a value of 0x3C correctly. 
The values printed on the OLED PCB are garbage.

olikraus, 2025-06-11:
Agree, I2C address is only 7 bit. 
But i am afraid, I can't change this any more without breaking many .ino files out there.

electronicsguy, 2025-06-11:
How about a new method like "setI2CAddress_standard()" which will not shift the specified address? Along with a deprecation notice for the current version of the method. 

Everything else remains the same.