olikraus/u8g2 #1518: Pixels origin not in the right place

ricbrue, 2021-06-18:
I've a SH1107 display controller with a 128x128 pixels display (UG-2828TSWIG01). The communication of the driver seems to work on a custom Linux distribution (arm64 device). The only issue that I encounter is that the origin of the coordinates is not in the right place as you can see in the following picture.
[image]
I also copy here the test code:
```cpp
u8g2_t u8g2;
u8g2_Setup_sh1107_i2c_128x128_f(&u8g2, U8G2_R0, u8x8_byte_arm_linux_hw_i2c, u8x8_arm_linux_gpio_and_delay);
u8x8_SetPin(u8g2_GetU8x8(&u8g2), U8X8_PIN_RESET, GPIO_DISPLAY_RESET_PIN_NO);
u8g2_SetI2CAddress(&u8g2, (0x3C<<1));
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0);

u8g2_ClearBuffer(u8g2);
u8g2_SetFont(u8g2, u8g2_font_ncenB08_tr);
//This creates the lines
for(int i=0;i<128;i++){
  u8g2_DrawPixel(u8g2, i,i); //diagonal -> dx bottom
  u8g2_DrawPixel(u8g2, i,127-i); //diagonal -> dx top
  u8g2_DrawPixel(u8g2, i,0); //line top
  u8g2_DrawPixel(u8g2, i,7); //line top
  u8g2_DrawPixel(u8g2, i,127); //line bottom
  u8g2_DrawPixel(u8g2, i,120); //line bottom
  u8g2_DrawPixel(u8g2, 0,i); //line left
  u8g2_DrawPixel(u8g2, 7,i); //line left
  u8g2_DrawPixel(u8g2, 127,i); //line right
  u8g2_DrawPixel(u8g2, 120,i); //line right
}
u8g2_SendBuffer(u8g2);
```
What is the issue here?

PS: I am actually using the Arm porting of the library ([here](https://github.com/wuhanstudio/u8g2-arm-linux))

olikraus, 2021-06-18:
Is there any datasheet available for the display)

ricbrue, 2021-06-18:
> Is there any datasheet available for the display?

Hi, I have only a confidential datasheet, if you want it I can send the PDF to you via email.

olikraus, 2021-06-18:
My e-mail is mentioned in every c-code. What I would like to know is, whether the display hardware itself includes the shift. This would mean, that the shift has to be compensated by the software.
One more thing: Did you try the other sh1107 devices?

u8x8_d_sh1107_seeed_128x128 and u8x8_d_sh1107_pimoroni_128x128?
This should be one of:
https://github.com/olikraus/u8g2/blob/f2d2d93662e437c71970b19cb824f8311fff0259/csrc/u8g2.h#L682-L705

ricbrue, 2021-06-18:
> My e-mail is mentioned in every c-code. What I would like to know is, whether the display hardware itself includes the shift. This would mean, that the shift has to be compensated by the software.
> One more thing: Did you try the other sh1107 devices?
> 
> u8x8_d_sh1107_seeed_128x128 and u8x8_d_sh1107_pimoroni_128x128?
> This should be a i2c "f" version from here:
> https://github.com/olikraus/u8g2/blob/f2d2d93662e437c71970b19cb824f8311fff0259/csrc/u8g2.h#L682-L705

Email sent, now I'll try other sh1107 devices thank you.

ricbrue, 2021-06-18:
Finally solver using **u8g2_Setup_sh1107_pimoroni_128x128_f**. Indeed I think that the issue was the default_x_offset and the flipmode_x_offset which were 96 instead of 0. Thank you anyway for your help.

closed 2021-06-18 by ricbrue

olikraus, 2021-06-18:
Ok, great. Thanks for the feedback.