olikraus/u8g2 #1519: ST7920 192x32, Parallel interface..

galyamovm, 2021-06-28:
I want to connect a display with a ST7920 controller to a STM32 via a parallel bus.
I read the documentation, but did not understand which initialization functions should be used. Could you give an example?

olikraus, 2021-06-29:
https://github.com/olikraus/u8g2/wiki/gallery#26-nov-2016-st7920-128x64-lcd-in-8080-parallel-mode

galmarm, 2021-07-01:
Thanks for your answer, olikraus!
If I understand correctly, this example is for Arduino, but I have my own board with a STM32 microcontroller.
Which functions do I need to use?

olikraus, 2021-07-02:
Ah, ok, I see. You need to modify / implement a proper gpio & delay callback: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#template-for-the-gpio-and-delay-callback

This is a uC specific function, so I do not know how this is implemented in your case.

galmarm, 2021-07-02:
Thanks, olikraus.
I changed the code considering connecting the display to the microcontroller and it worked! But there was a problem with the U8X8_MSG_DELAY_NANO latency - it had to be increased.

But there is a problem - I cannot write on the display in Russian. My development environment supports UTF8, the font has chosen the correct one, but nothing is displayed on the display : 

`
    do
    {
      u8g2_SetFont(&u8g2, u8g2_font_unifont_t_cyrillic);
      u8g2_DrawStr(&u8g2, 0, 15, "Привет!");
    } while (u8g2_NextPage(&u8g2)); 
`

Perhaps I forgot something?

olikraus, 2021-07-02:
Did you use the initial first page command?

galmarm, 2021-07-02:
Sorry, I did not immediately find the output function in UTF8 format!
Now everything is displayed on the display!

galmarm, 2021-07-02:
My example (maybe help someone in the future):

> 
  u8g2_Setup_st7920_192x32_f(&u8g2, U8G2_R0, u8x8_byte_8bit_6800mode, u8g2_gpio_and_delay_stm32); 
  u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
  u8g2_SetPowerSave(&u8g2, 0); // wake up display
  
  u8g2_FirstPage(&u8g2);
  
    do
    {
      u8g2_SetFont(&u8g2, u8g2_font_7x13_t_cyrillic );
      u8g2_DrawUTF8(&u8g2, 0, 15, "Привет!");
    } while (u8g2_NextPage(&u8g2)); 

> 

Callback:

> 
uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
  
  
  switch(msg)
  {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8      
      HAL_Delay(1);
      break;							// can be used to setup pins
      
    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
      delay_ns(50 * arg_int);
      break;    
    case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
      break;
    case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
      for (uint16_t n = 0; n < 320; n++)
      {
        __NOP();
      }      
      break;
    case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
      HAL_Delay(arg_int);
      break;
    case U8X8_MSG_DELAY_I2C:				// arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
      break;							// arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us
    case U8X8_MSG_GPIO_D0:				// D0 or SPI clock pin: Output level in arg_int
    //case U8X8_MSG_GPIO_SPI_CLOCK:
      if (arg_int) HAL_GPIO_WritePin(DL_PORT, DATA0, GPIO_PIN_SET);
      else HAL_GPIO_WritePin(DL_PORT, DATA0, GPIO_PIN_RESET);      
      break;
    case U8X8_MSG_GPIO_D1:				// D1 or SPI data pin: Output level in arg_int
    //case U8X8_MSG_GPIO_SPI_DATA:
      if (arg_int) HAL_GPIO_WritePin(DL_PORT, DATA1, GPIO_PIN_SET);
      else HAL_GPIO_WritePin(DL_PORT, DATA1, GPIO_PIN_RESET);      
      break;
    case U8X8_MSG_GPIO_D2:				// D2 pin: Output level in arg_int
      if (arg_int) HAL_GPIO_WritePin(DL_PORT, DATA2, GPIO_PIN_SET);
      else HAL_GPIO_WritePin(DL_PORT, DATA2, GPIO_PIN_RESET);      
      break;
    case U8X8_MSG_GPIO_D3:				// D3 pin: Output level in arg_int
      if (arg_int) HAL_GPIO_WritePin(DL_PORT, DATA3, GPIO_PIN_SET);
      else HAL_GPIO_WritePin(DL_PORT, DATA3, GPIO_PIN_RESET);      
      break;
    case U8X8_MSG_GPIO_D4:				// D4 pin: Output level in arg_int
      if (arg_int) HAL_GPIO_WritePin(DH_PORT, DATA4, GPIO_PIN_SET);
      else HAL_GPIO_WritePin(DH_PORT, DATA4, GPIO_PIN_RESET);      
      break;
    case U8X8_MSG_GPIO_D5:				// D5 pin: Output level in arg_int
      if (arg_int) HAL_GPIO_WritePin(DH_PORT, DATA5, GPIO_PIN_SET);
      else HAL_GPIO_WritePin(DH_PORT, DATA5, GPIO_PIN_RESET);      
      break;
    case U8X8_MSG_GPIO_D6:				// D6 pin: Output level in arg_int
      if (arg_int) HAL_GPIO_WritePin(DH_PORT, DATA6, GPIO_PIN_SET);
      else HAL_GPIO_WritePin(DH_PORT, DATA6, GPIO_PIN_RESET);      
      break;
    case U8X8_MSG_GPIO_D7:				// D7 pin: Output level in arg_int
      if (arg_int) HAL_GPIO_WritePin(DH_PORT, DATA7, GPIO_PIN_SET);
      else HAL_GPIO_WritePin(DH_PORT, DATA7, GPIO_PIN_RESET);      
      break;
    case U8X8_MSG_GPIO_E:				// E/WR pin: Output level in arg_int
      if (arg_int) HAL_GPIO_WritePin(EN_PORT, EN_PIN, GPIO_PIN_SET);
      else HAL_GPIO_WritePin(EN_PORT, EN_PIN, GPIO_PIN_RESET);         
      break;
    case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_DC:				// DC (data/cmd, A0, register select) pin: Output level in arg_int
      if (arg_int) HAL_GPIO_WritePin(A0_PORT, A0_PIN, GPIO_PIN_SET);
      else HAL_GPIO_WritePin(A0_PORT, A0_PIN, GPIO_PIN_RESET);        
      break;
    case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_CS1:				// CS1 (chip select) pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_CS2:				// CS2 (chip select) pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_I2C_CLOCK:		// arg_int=0: Output low at I2C clock pin
      break;							// arg_int=1: Input dir with pullup high for I2C clock pin
    case U8X8_MSG_GPIO_I2C_DATA:			// arg_int=0: Output low at I2C data pin
      break;							// arg_int=1: Input dir with pullup high for I2C data pin
    case U8X8_MSG_GPIO_MENU_SELECT:
      u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_NEXT:
      u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_PREV:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_HOME:
      u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 0);
      break;
    default:
      u8x8_SetGPIOResult(u8x8, 1);			// default return value
      break;
  }
  return 1;  
}

>

shlyakpavel, 2021-08-06:
So why is the issue still open?

v1993, 2021-08-06:
Because no one bothered to close it yet.

olikraus, 2021-08-06:
well... feel free to close it. Actually my time is very much limited for u8g2, but I try to do a kind of a bulk cleanup once in a while.

closed 2021-08-06 by olikraus