olikraus/u8g2 #1521: bdfconv failing to convert unifont for large number of glyphs

antiprism, 2021-06-30:
Hi

I am trying to make a unifont file with a large number of glyphs on Ubuntu, but the bdfconv program is failing with an assertion.

I include sample runs below where the program:
* succeeds on the range $0000-$3B96
* fails on the range $0000-$3B97
* succeeds again on the range $0000-$3C97

In general, the conversion nearly always fails fails on ranges larger than this.

```
tools/font$ bdfconv/bdfconv -v -f 1 -m '$0000-$3B96' bdf/unifont.bdf -o unifont_custom.c -n unifont_custom 
Parse File bdf/unifont.bdf: 57086 glyph(s) found
Map: map_cmd_list='$0000-$3B96'
Map: exclude=0 from=0/$0 to=15254/$3b96 map=0/$0
Reduce: Start
Reduce: End
CalculateMaxBBX: x=0, y=-2, w=16, h=16
CalculateMaxBBX: Encodings x=0, y=0, w=0, h=0
bf_CalculateMinMaxDWidth: dx_min=8, dx_max=16
bf_CalculateMinMaxDWidth: x_min=0, x_max=14
bf_CalculateMaxBitFieldSize: bbx.x=5, bbx.y=5, bbx.w=5, bbx.h=5, dwidth=6
RLE Compress: best zero bits 3, one bits 2, total bit size 3054344
RLE Compress: Font code generation, selected glyphs=15255, total glyphs=57086
RLE Compress: ASCII gylphs=256, Unicode glyphs=14999
RLE Compress: Glyphs per unicode lookup table entry=100
RLE Compress: Unicode lookup table len=149, written entries=149
RLE Compress: Unicode lookup table first entry: delta=596, encoding=356
RLE Compress: Unicode lookup table last entry: delta=4686, encoding=65535
RLE Compress: 'A' pos = 1499, 'a' pos = 1911
RLE Compress: Font size 382416
bf_WriteU8G2CByFilename: Write file 'unifont_custom.c'
```

```
tools/font$ bdfconv/bdfconv -v -f 1 -m '$0000-$3B97' bdf/unifont.bdf -o unifont_custom.c -n unifont_custom 
Parse File bdf/unifont.bdf: 57086 glyph(s) found
Map: map_cmd_list='$0000-$3B97'
Map: exclude=0 from=0/$0 to=15255/$3b97 map=0/$0
Reduce: Start
Reduce: End
CalculateMaxBBX: x=0, y=-2, w=16, h=16
CalculateMaxBBX: Encodings x=0, y=0, w=0, h=0
bf_CalculateMinMaxDWidth: dx_min=8, dx_max=16
bf_CalculateMinMaxDWidth: x_min=0, x_max=14
bf_CalculateMaxBitFieldSize: bbx.x=5, bbx.y=5, bbx.w=5, bbx.h=5, dwidth=6
RLE Compress: best zero bits 3, one bits 2, total bit size 3054704
RLE Compress: Font code generation, selected glyphs=15256, total glyphs=57086
RLE Compress: ASCII gylphs=256, Unicode glyphs=15000
RLE Compress: Glyphs per unicode lookup table entry=100
RLE Compress: Unicode lookup table len=150, written entries=149
RLE Compress: Unicode lookup table first entry: delta=600, encoding=356
RLE Compress: Unicode lookup table last entry: delta=4686, encoding=65535
bdfconv: bdf_rle.c:898: bf_RLECompressAllGlyphs: Assertion `unicode_lookup_table_len == unicode_lookup_table_pos' failed.
Aborted (core dumped)
```

```
tools/font$ bdfconv/bdfconv -v -f 1 -m '$0000-$3C97' bdf/unifont.bdf -o unifont_custom.c -n unifont_custom 
Parse File bdf/unifont.bdf: 57086 glyph(s) found
Map: map_cmd_list='$0000-$3C97'
Map: exclude=0 from=0/$0 to=15511/$3c97 map=0/$0
Reduce: Start
Reduce: End
CalculateMaxBBX: x=0, y=-2, w=16, h=16
CalculateMaxBBX: Encodings x=0, y=0, w=0, h=0
bf_CalculateMinMaxDWidth: dx_min=8, dx_max=16
bf_CalculateMinMaxDWidth: x_min=0, x_max=14
bf_CalculateMaxBitFieldSize: bbx.x=5, bbx.y=5, bbx.w=5, bbx.h=5, dwidth=6
RLE Compress: best zero bits 3, one bits 2, total bit size 3152504
RLE Compress: Font code generation, selected glyphs=15512, total glyphs=57086
RLE Compress: ASCII gylphs=256, Unicode glyphs=15256
RLE Compress: Glyphs per unicode lookup table entry=100
RLE Compress: Unicode lookup table len=152, written entries=152
RLE Compress: Unicode lookup table first entry: delta=608, encoding=356
RLE Compress: Unicode lookup table last entry: delta=4695, encoding=65535
RLE Compress: 'A' pos = 1499, 'a' pos = 1911
RLE Compress: Font size 394698
bf_WriteU8G2CByFilename: Write file 'unifont_custom.c'
```

I have also mentioned this in an issue to the u8g2-unifont-helper project: https://github.com/stncrn/u8g2-unifont-helper/issues/1

Adrian.

olikraus, 2021-06-30:
Hmmm strange, for my own reference:

https://github.com/olikraus/u8g2/blob/fee74cf8b8d9a820c9a7cfeeb7df19400f1ac7d8/tools/font/bdfconv/bdf_rle.c#L898

I do not know when I will have time to look into this. 
However, if '$0000-$3B97' doesn't work, but '$0000-$3C97' works, do you have an issue at the moment?

antiprism, 2021-06-30:
My aim is to include as many glyphs as possible, in order to display tags from mp3 files, so my ideal range is $0000-$FFFF (and possibly also plane 1 glyphs). The program runs on a Raspberry Pi, so if it turns out reasonable to include all the glyphs I would do that rather than choose a selection that covers the major languages.

Adrian.

antiprism, 2021-06-30:
I meant to say, I have made a library package of U8g2 for the Raspberry Pi, and whatever large unifont "world languages" file worked out I would generate as part of the package setup: https://github.com/antiprism/libu8g2arm

Adrian.

olikraus, 2021-06-30:
Well, ok, but will you have a problem. I mean you wrote only "-m '$0000-$3B97'" does not work. But maybe "-m '$0000-$ffff'" will work. Did you test -m '$0000-$ffff'?

antiprism, 2021-06-30:
Yes, I did try this. I also tried a number of ranges larger than $0000-$3B97 and they all failed except $0000-$3C97. I added some debugging code and there is a greater difference between the assert variables as the range gets higher so it seems unlikely that the higher range values will succeed.

The result for $0000-$ffff is
```
tools/font$ bdfconv/bdfconv -v -f 1 -m '$0000-$ffff' bdf/unifont.bdf -o unifont_custom.c -n unifont_custom 
Parse File bdf/unifont.bdf: 57086 glyph(s) found
Map: map_cmd_list='$0000-$ffff'
Map: exclude=0 from=0/$0 to=65535/$ffff map=0/$0
Reduce: Start
Reduce: End
CalculateMaxBBX: x=0, y=-2, w=16, h=16
CalculateMaxBBX: Encodings x=0, y=0, w=0, h=0
bf_CalculateMinMaxDWidth: dx_min=8, dx_max=16
bf_CalculateMinMaxDWidth: x_min=0, x_max=14
bf_CalculateMaxBitFieldSize: bbx.x=5, bbx.y=5, bbx.w=5, bbx.h=5, dwidth=6
RLE Compress: best zero bits 3, one bits 2, total bit size 16498328
RLE Compress: Font code generation, selected glyphs=57086, total glyphs=57086
RLE Compress: ASCII gylphs=256, Unicode glyphs=56830
RLE Compress: Glyphs per unicode lookup table entry=100
RLE Compress: Unicode lookup table len=568, written entries=563
RLE Compress: Unicode lookup table first entry: delta=2272, encoding=356
RLE Compress: Unicode lookup table last entry: delta=1537, encoding=65535
bdfconv: bdf_rle.c:898: bf_RLECompressAllGlyphs: Assertion `unicode_lookup_table_len == unicode_lookup_table_pos' failed.
Aborted (core dumped)
```

Adrian.

olikraus, 2021-06-30:
Ah, ok, i understand. Nevertheless I currently don't have time to debug this. Maybe next week.

antiprism, 2021-06-30:
That would be great. Thanks for looking at it!

Adrian.

olikraus, 2021-07-03:
There was an off by one error in the code, which is now fixed. It will only abort for a large number of glyphs. Thanks for reporting this.

antiprism, 2021-07-04:
Thanks for looking at this so quickly! I am now able to convert $0000-$FFFF, and a quick test (image attached) shows the converted font is displaying  'BTS (방탄소년단)' correctly.

[image]

Adrian.

closed 2021-07-04 by antiprism