olikraus/u8g2 #1531: ST75256_JLX256128 not full display

uldara1, 2021-07-15:
Hello i have used this library with ST75256_JLX256128
the result of LCD is working but it's work only with area 240x128
how to fix it to work with 256x128 ?

  Serial.println(u8g2.getWidth()); // 240
  Serial.println(u8g2.getHeight()); //128

thank

olikraus, 2021-07-15:
Did you enable 16 bit mode? See FAQ.

uldara1, 2021-07-15:
Hello bro
here is my code and i don't know how to enable 16bit mode pls check 
 
`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST75256_JLX256128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  

void setup(void) {
 Serial.begin(115200);
  u8g2.begin();
  u8g2.setContrast(160);

}

void loop(void) {

    u8g2.firstPage();
do{
   u8g2.drawStr(0, 0, "Bitmap overlay 123456789012345678901234567890");
 } while( u8g2.nextPage() );

   Serial.println(u8g2.getWidth());
   Serial.println(u8g2.getHeight());
   delay(1000);
}
`

olikraus, 2021-07-15:
https://github.com/olikraus/u8g2/blob/7b2b60f4184acb8524142061e306d916e7e34ebd/doc/faq.txt#L163-L180

uldara1, 2021-07-15:
it work now thank you 👯

closed 2021-07-15 by uldara1