olikraus/u8g2 #1535: Add feature getStrWidth() to U8G2_SSD1306_128X64_NONAME_F_HW_I2C

johnsmakar, 2021-07-21:
I was trying to use getStrWidth, but get...

`no matching function for call to 'U8G2_SSD1306_128X64_NONAME_F_HW_I2C::getStrWidth(String)'`

Can you add this feature?

olikraus, 2021-07-21:
It is implemented:
https://raw.githubusercontent.com/olikraus/u8g2/master/cppsrc/U8g2lib.h

and described here:
https://github.com/olikraus/u8g2/wiki/u8g2reference#getstrwidth

johnsmakar, 2021-07-21:
How come I get a no matching function error?

Note: I am using a Espressif Pico-D4

olikraus, 2021-07-21:
Hmmm... I nither know your code nor do I know your environment. Do you use Arduino IDE?

johnsmakar, 2021-07-21:
I am using the ArduinoIDE.

Is there an incorrect implementation code of code that would result with that error? Thanks in advance

olikraus, 2021-07-21:
I am not aware of any issue here.

olikraus, 2022-03-04:
Oh I just see, that you tried to pass a "String" object. You have to pass "string.c_str()" instead.

closed 2023-02-22 by olikraus