olikraus/u8g2 #1536: SSD1316 128X32 No display constructor avaliable for an ADAFRUIT (Chinese) display

lukilukeskywalker, 2021-07-22:
It isn't an Issue, more like a request
So a few years ago I bought a screen in [Aliexpress](https://es.aliexpress.com/item/32672229793.html?spm=a2g0s.9042311.0.0.274263c0UB0eRM), that worked with the Adafruit example for 128x32 screens, And some time later I got it to work with u8g2 with the constructor `U8G2_SSD1306_128X32_UNIVISION_1_HW_I2C` . Just with one catch, some libraries (like tcmenu) didn't reload the whole screen, just the upper 5 pixels¿?. I thought, well, maybe the constructor is not the appropriate one, so I searched a new one. I didn't find any that would work properly, in some cases pixels where missing in others It didn't even send the appropriate initiation message. I also got to `U8G2_SSD1306_128X32_WINSTAR_F_HW_I2C`, where the whole image was shifted to the top, and to the right. Searching a bit I got to this issue [(Issue 724)](https://github.com/olikraus/u8g2/issues/724) where the pixels in the constructor olikraus made where shifted by 4. So I decided to try changing the winstar constructor, to see if I got the shift back to position. Changing `default_x_offset` in [u8x8_d_ssd1306_128x32](https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x32.c), line 238 to 128 solved my problem. Maybe you can create a new constructor for this screen, with /* default_x_offset = */ 128 instead of 125, for Adafruit screens (As it does not appear to exist an Constructor for adafruit 128x32 screens)

Anyway, Thanks to all u8g2 developers for creating this library

olikraus, 2021-07-24:
Did you test U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C ?

The UNIVISION and WINSTAR differ mainly in the x offset.

lukilukeskywalker, 2021-07-24:
Uh, sorry, the day I tested and searched for the different constructors I didn't get it work with `U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C`, but I just tested it again, and worked without any problems. Sorry for the inconveniences. I wonder why It didn't work last day

closed 2021-07-24 by lukilukeskywalker

olikraus, 2021-07-24:
👍