olikraus/u8g2 #1537: Suggestion for HW SPI

weypro, 2021-07-23:
I want to use the library on my esp32-s, but I face the problem when I switch the sw spi to the hw spi.
While I seach for the solution, I find "https://github.com/olikraus/u8g2/issues/934"
I have the same idea that we can pass a reference to any SPI object to the class. It may be written as the following example:
```c++
xxxclass(const u8g2_cb_t *rotation, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE, SPIClass &spi=SPI) : U8G2() {
...
}
```
It's a good idea, isn't it? In that case, it doesn't matter if the files don't meet the standard, and the library will be more flexible.
The only problem is that it will take a lot of time to modify existing classes, but I think it's worthwhile.

olikraus, 2021-07-24:
https://github.com/olikraus/u8g2/issues/1083

closed 2021-07-27 by weypro

Zipdox2, 2024-11-15:
As a workaround you can use this before calling u8g2.begin():
```c
    u8x8_t *u8x8 = u8g2.getU8x8();
    u8x8->pins[U8X8_PIN_SPI_CLOCK] = CLOCK_PIN;
    u8x8->pins[U8X8_PIN_SPI_DATA] = DATA_PIN;
 ```