olikraus/u8g2 #1544: LoadFromSD.ino is not working as expected

hicksan, 2021-08-03:
I am using a Mega with an ST7920 display and SD card reader, using pins 50-52 with pin 53 as CS for the ST7920 and pin 9 as CS for the card reader. I also get a blank screen on the LCD. I have added some Serial.Println() commands to the sketch, so can confirm it is reading and writing the file correctly from the SD card; just not displaying the output properly to the LCD.

My constructor is:
U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R0, 52, 51, 53, U8X8_PIN_NONE);

The SD card is opened with:
if (!SD.begin(9)){

The LoadFromSD.ino file is otherwise unchanged. 

When it runs it outputs to serial monitor but the ST7920 screen is lit but remains blank.

I have confirmed that the screen runs normally with other sketches, with the same constructor, and that the card reader works correctly.

I have noted in another example that the #include <SPI.h> command causes problems for the ST7920 with this constructor. I tried commenting it out and adding in #include <Wire.h> (which worked previously) but with no effect in this case.

Please can you help?

Do you have a simpler text file example, rather than a binary graphic, please? That would make things easier to diagnose.

hicksan, 2021-08-07:
This issue has now been sorted.

The diagnosis was that the SD card reader and the ST7920 were in conflict in the SPI bus. This is because some SD cards, including mine, are incorrectly wired such that the MISO pin is held permanently at low - so the reader cannot give up control of the bus to share with other devices. The fault lies with the hardware: more expensive SD readers should work fine straight off.

I tested this by opening the U8G2, and getting it to write something to the screen - before initializing the SD card reader - and a second message after (each with its own do...while (u8g2.nextpage() ) loop). This proved that the screen interface is fine until the sd.begin() command. 

The solution was to change the pins to the ST7920 LCD display so as not to use the hardware SPI bus and to modify the constructor line so all the pins there point to the new addresses. This effectively creates a second SPI bus, completely independent of the first with no common pins (software controlled so slightly slower). I simply moved everything from pin 53 to 43, and all the others similarly (down by 10).

With the SPI busses isolated it works perfectly. I hope this helps other users faced with similar problems.

I believe the SdFat.h library, used instead of the SD.h library, might give a little more flexibility in terms of redefining the SPI pins to different set for the reader rather than having to move the display set.

I still would like to see a text-based example; perhaps a file picker and file read display for text content instead of a bitmap, please, Oliver. Thanks for giving us all such a fantastic library! You are a star!

olikraus, 2021-08-07:
Thanks for the update. Nice finding.

hicksan, 2021-08-08:
You're welcome Oliver. Do please consider adding a file picker and/or text file reader example. It would be very useful as well as much easier to diagnose for people getting started. Thanks again for such a great library.

closed 2021-08-08 by hicksan