olikraus/u8g2 #1549: Send commands to SSD1306 from sketch

ryanaukes, 2021-08-06:
Hello,

How can I send commands from the sketch to my SSD1306 OLED screen?
I want to change the pre-charge period to reduce the whine of the screens, but I cant get it to work when trying to set it from the sketch.
Right now I modified the u8x8_d_ssd1306_128x64_noname_init_seq function in the u8x8_d_ssd1306_128x64_noname.c file, but I prefer to leave the library unmodified and change it in the sketch itself. 
I tried the following, but it didn't have any effect:

```
u8x8_cad_StartTransfer(display.getU8x8()); //Enable chip, delay is part of the transfer start */
u8x8_cad_SendCmd( display.getU8x8(), 0x0d9); //Set Pre-charge Period phase 1 to 1 and phase 2 to 1 
u8x8_cad_SendArg( display.getU8x8(), 0x011);
u8x8_cad_EndTransfer(display.getU8x8()); //Disable chip
display.begin();
display.clearBuffer();
display.setContrast(1);
display.drawXBMP(48, 4, 56, 56, volume);
display.sendBuffer();
```

Thanks in advance!

olikraus, 2021-08-06:
The u8x8_cad calls seem to be correct, but they should be placed after display.begin().
In general there is sendF to sent display specific commands: https://github.com/olikraus/u8g2/wiki/u8g2reference#sendf
There had been also a lot of discussions in the past.
Latest one happend here: https://github.com/olikraus/u8g2/issues/1504

ryanaukes, 2021-08-07:
> The u8x8_cad calls seem to be correct, but they should be placed after display.begin().
>
> In general there is sendF to sent display specific commands: [...]

Thanks for your quick reply. I already tried to send the commaands after the display.begin() or even after the display.sendBuffer() lines:

```
display.begin(); u8x8_cad_StartTransfer(display.getU8x8()); //Enable chip, delay is part of the transfer start */
u8x8_cad_SendCmd( display.getU8x8(), 0x0d9); //Set Pre-charge Period phase 1 to 1 and phase 2 to 1 
u8x8_cad_SendArg( display.getU8x8(), 0x011);
u8x8_cad_EndTransfer(display.getU8x8()); //Disable chip
display.clearBuffer();
display.setContrast(1);
display.drawXBMP(48, 4, 56, 56, volume);
display.sendBuffer();
```

But then the displays stayed blank and didn't work at all. What is the right locations to put the code? And is it better to use the sendF commands?

Thanks again!

olikraus, 2021-08-07:
SendF include the start / end transfer. In fact it is identical to your code. I assume some other issue here. First ensure your xbm is visible without any extra code.

ryanaukes, 2021-08-07:
When I don't use the u8x8_cad lines the image is already visible and everything works fine.

olikraus, 2021-08-07:
I don't know, it should work. Here is the implementation of sendf, as I mentioned before, it is identical to your code:
https://github.com/olikraus/u8g2/blob/fe9bd0b35f96e16d865e9ab909b3dbd23bf9cc3d/csrc/u8x8_cad.c#L123-L139

Did you try some other values instead of 0x11?
Maybe 0x11 does not work on your display:
`u8g2.sendF("ca", 0x0d9, 0x022);`

ryanaukes, 2021-08-07:
I did not try any other values, but I am using these values right now with the modified library code.

olikraus, 2021-08-07:
hmm.. ok. 
Then I also don't why this doesn't work.

closed 2022-03-04 by olikraus