olikraus/u8g2 #1551: Displays bigger than 256 pixel in the Arduino lib

4ndreas, 2021-08-07:
Hi,
I got a rather unusual display with 480x64 pixel, in a WeyTech HK2000C keyboard. The display uses the T6963C chip and 6 of the T6A39 and one T6A40 chip. 
After some fiddling I got the display working, but the drawing was somehow broken. 
After future investigation I noticed that all the x and y variables for drawing where uint8_t this obvious cause problems with displays larger than 256 pixel.

I patched all the drawing functions by hand https://github.com/4ndreas/U8g2_Arduino and now the display works fine. 

I don't know if it is worth to handle this problem but it is something to be aware of.

olikraus, 2021-08-07:
U8g2 offers 16 bit mode: This mode will use 16 bit variables instead: https://github.com/olikraus/u8g2/blob/fe9bd0b35f96e16d865e9ab909b3dbd23bf9cc3d/doc/faq.txt#L163

4ndreas, 2021-08-07:
most of the functions in the clib do not use `u8g2_uint_t ` instead are hard coded to `uint8_t`
just a few examples:
```
clib/u8g2_buffer.c
- void u8g2_UpdateDisplayArea(u8g2_t *u8g2, uint8_t  tx, uint8_t ty, uint8_t tw, uint8_t th)
+ void u8g2_UpdateDisplayArea(u8g2_t *u8g2, uint16_t  tx, uint16_t ty, uint16_t tw, uint16_t th)
```

```
clib/u8x8_8x8.c
- void u8x8_DrawGlyph(u8x8_t *u8x8, uint8_t x, uint8_t y, uint8_t encoding)
+ void u8x8_DrawGlyph(u8x8_t *u8x8, uint16_t x, uint16_t y, uint8_t encoding)

- uint8_t u8x8_DrawString(u8x8_t *u8x8, uint8_t x, uint8_t y, const char *s)
+ uint8_t u8x8_DrawString(u8x8_t *u8x8, uint16_t x, uint16_t y, const char *s)
```

```
clib/u8x8_display.c
- uint8_t u8x8_DrawTile(u8x8_t *u8x8, uint8_t x, uint8_t y, uint8_t cnt, uint8_t *tile_ptr)
+ uint8_t u8x8_DrawTile(u8x8_t *u8x8, uint16_t x, uint16_t y, uint16_t cnt, uint8_t *tile_ptr)
```

olikraus, 2021-08-07:
The u8x8 coordinates refer to tiles, which are 8x8 pixel blocks, so with 0..255 a u8x8 screen can have up to 4095x4095 pixels. 
This is also the reason why I called those arguments often as tx and ty.

4ndreas, 2021-08-08:
hi, I'm sorry for the misunderstanding, I really don't know why I though this was necessary, I removed all the changes except the parts for the new display and all works. 
Thanks for your help.

closed 2021-08-08 by 4ndreas