olikraus/u8g2 #1556: Gaps/Lines in Ellipse

johnrootk, 2021-08-10:
Hi!
I try to draw some eyes with the help of this library. The base are two ellipses. Sadly there are some gaps/lines through the ellipses. What's the reason for them? I already tried to play with the values. My current values show the least lines. Smaller or bigger vertical radius increases the effect. 
I'm working with an Arduino Mega 2560. My display is a chinese noname I2C display (very similar the one on a picture in the wiki).

Here is a minimal example code:
```#include <U8g2lib.h>
#include <SPI.h>
#include <Wire.h>

U8G2_SSD1306_128X64_NONAME_2_HW_I2C display(U8G2_R0);
byte eye_gap = 3;

void setup() {
  display.begin();
  drawEyes();
}

void loop() {
}

void drawEyes(){
  display.firstPage();
  do{
    display.setDrawColor(1);
    display.drawFilledEllipse(32 - eye_gap, 32, 32 - eye_gap, 18, U8G2_DRAW_ALL);
    
    display.setDrawColor(1);
    display.drawFilledEllipse(96 - eye_gap, 32, 32 - eye_gap, 18, U8G2_DRAW_ALL);
  }while(display.nextPage());
}```

Here is how it looks:
![gap_example](https://user-images.githubusercontent.com/2363712/128863182-bb8a8197-cfce-4094-879e-775b840e31a2.jpg)

The display isn't defect. When I draw fullscreen or other shapes like triangles, anything works find.

Thanks for your help in advance and best regards,
John

olikraus, 2021-08-10:
As mentioned in the wiki, the product of the two radius values must be lower than 512 which isn't the case in your exaple (https://github.com/olikraus/u8g2/wiki/u8g2reference#drawellipse). Maybe switching to u8g2 16 bit mode (https://github.com/olikraus/u8g2/blob/c092ff12d0b17b824b783c91dc233c0b9febd44a/doc/faq.txt#L163) will help.

johnrootk, 2021-08-10:
Hi Oli,
Thanks for the fast reply. Maybe I should have mention this. My initial value for the radius height was 16. Which leads to a radius product of: (32-3)*16=29*16=464. There are even bigger gaps (2 pixel wide). 
I gonna try the 16 bit mode, but I would have prefered not to change the lib files.

olikraus, 2021-08-10:
> Anyway, isn't it possible to activate this mode without manipulating lib-files? I tried to define the value before the include, but this leads to a bunch of errors :/.

16 Bit mode is automatically activated on ARM  and ESP (32bit) systems:
https://github.com/olikraus/u8g2/blob/c092ff12d0b17b824b783c91dc233c0b9febd44a/csrc/u8g2.h#L74-L79

Initially u8g2 was optimized a lot for AVR systems, thats why there is this limited 8 bit mode. But I guess even for DIY projects, the classic 8 Bit systems will get replaced.

closed 2022-03-04 by olikraus