olikraus/u8g2 #1557: ESP8266 LoadfromSD with ST7920

hicksan, 2021-08-12:
I am trying to run the loadFromSD demo sketch with an ESP8266 connected to as ST7920 LCD and an Sd card.

I have the display on pins D5 (CLK), D7 (MOSI) and D10 (CS) with the following constructor:
U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R0, D5, D7, D10, U8X8_PIN_NONE);

I ran your Graphics Test sketch with just the ST7920 on the ESP8266 and it ran perfectly with this constructor.

I have the SD card connected to pins D11 (MOSI) , D12 (MISO), D13 (CLK) and D9 (CS). I call it with
SD.begin(D9);
as per your demo sketch. I am assuming it will go to these pins as the SPI hardware defaults, as per an UNO.

The ST7920 and SD card are on separate SPI busses because, as I already discovered when I tried this with the Mega, the SD card reader is wired wrongly and cannot share an SPI bus with any other device. This setup 

worked fine with the Mega, but I am now trying it on an ESP8266 for other reasons.

The problem is it won't compile. I get loads of error messages relating to "multiple SD.h libraries" and stuff about SdFat that I am not even using, relating to a library in the area:
...\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\SDFS\src/SDFS.h
- a library that is not even mentioned in the sketch!

Any idea what I am doing or haven't done that might be causing this? The sketch in question is basically your demo sketch since I am just testing the hardware compatibility.

I attach the file below:
```
/*

  LoadFromSD.ino

  Demonstrate how to read a XBM file from SD card.
  This example will write a XBM image to SD card as a preparation during setup ("u8g2.bin").
  The written file content will be displayed.
  
  There is also an external conversion tool, which generates the expected binary from a PNG file:
    https://github.com/olikraus/u8g2/tree/master/tools/png2bin

  Remember to adjust SD.begin(SD_CS) argument.

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2019, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

*/

#include <Arduino.h>
#include <U8g2lib.h>
//#include <SPI.h>
[... 445 lines cut ...]

  // only to ensure that there is some example data on the SD card...
  writeFile("u8g2.bin", u8g2_logo_97x51_width, u8g2_logo_97x51_height, u8g2_logo_97x51_bits);

}

void loop() {
  

  u8g2.clearDisplay();
  u8g2.setFont(u8g2_font_helvR10_tr);
  u8g2.firstPage();
  do {
    Serial.println("Read from SD:");
    u8g2.drawStr(0,20,"Read from SD:");
    
    drawFile(0, 20, "u8g2.bin");
    
    //delay(1000);
  } while ( u8g2.nextPage() );
  delay(1000);
  Serial.println("Next loop");
}

```

hicksan, 2021-08-13:
I got this to work in the end.

Problem 1: The esp8266 has to be introduced as a different board type in the 'board manager' on the Arduino IDE. What it doesn't tell you is that this brings in a whole different set of standard libraries - like its own versions of SD.h and SdFat.h. Arduino is smart and when it encounters multiple libraries with the same name it will try to choose the most appropriate one. So if it know you are using an "esp8266" processor it will give preference to one in an "esp8266..." directory. But not if the directory is named "ESP8266SDFAT". in that instance it reverts to the Arduino standard Sd / SdFat library - and all kinds of compatability compile errors get thrown at you.

Two possible solutions:
1. Delete the old Arduino libraries. Then there is no conflict.
2. Rename the directory of the new esp8266 library from "ESP8266SDFAT" to "SdFat". Then it can figure out all by itself which is the right one.

Problem 2: My SD card still hogs the SPI bus and conflicts with the ST7920 display. I noticed that esp8266 cards are helpfully labelled with two sets of SPI pins. Sets 11-13 and 5-7. So I thought putting the Sd card reader on 11-13 and the ST7920 on 5-7 would solve this up front. It didn't; they still conflicted - whichever is active locks the other out.

I solved this by moving the ST7920 display onto other non-SPI pins. In this case I set CLK as D4 and MOSI as D3. I left CS as D10. After editing the constructor accordingly - it all worked fine.

I hope this helps somebody.

olikraus, 2021-08-14:
Thanks for reporting. :+1:

hicksan, 2021-09-04:
Unfortunately I really need to use those extra pins because the ESP8266 has so few. I have invested in a more expensive SD card reader so the ST7920 and the SD and other devices can share the SPI bus. Unfortunately this hasn't worked. The bus locking is no longer the issue but I have discovered that LoadFromSD demo program doesn't work: every time u8g2.begin() is called it stops the SD from working and every time SD.begin(D10) is called it stops the ST7920. By using lots of extra .begin statements I got it partly to work, but then I found:
(1) the drawFile() subroutine is within a u8g2 page write process which then gets interrupted
(2) any drawStr or drawPixel command causes the output to revert to random Chinese characters.
It looks to me like an array somewhere is overflowing, and I cannot get this example to work.
I am using the Neu Rah Arduino menu system, which relies on this library and uses these draw commands.

olikraus, 2021-09-04:
It you share pins between sd card and display, then you MUST use the HW_SPI constructor from U8g2. In you initial example you have used the SW_SPI constructor. Using the HW_SPI constructor is a precondition to make it work. However, even then it may not work, because display and sdcard may require different speed. It might be required to set the bus speed (u8g2.setBusClock()) manually every time you want to access your display.

hicksan, 2021-09-04:
IT WORKED PERFECTLY. Both this test sketch and my actual working sketch.
Just by changing the constructor line from SW to HW and removing software pin references.
Thank you so much. You are an amazing and wonderful person, Oliver. Thankyouthankyouthankyou.

closed 2021-09-04 by hicksan

AnHardt, 2021-09-13:
Don't use ST7920 with other devices on the same SPI bus. The ST7920s CS pin is broken by design.
See [datasheet](http://www.hpinfotech.ro/ST7920.pdf) page 26 paragraph 2.

hicksan, 2021-09-13:
It's working fine for me with an ST7920, SD card reader and MCP2515 all sharing the same SPI bus.

Your document link crashed two browsers though, lost all my open pages and triggered lots of virus warnings from my firewall.