olikraus/u8g2 #1561: Update of the GetStrWidth / GetUTF8Width functions

olikraus, 2021-08-19:
Background:
Each glyph has a x offset which can be negative, 0 or positive. In many cases this offset is 0, but especially for uppercase chars, the offset is positive. This means, that there will be an extra width of n pixel on the left where n is the x offset.
An example is the uppercase "C" in u8g2_font_helvR08_tr:
x-offset = 1
delta x = 8
gylph width = 6 (bitmap width of the glyph)
The old behavior of the GetStrWidth function would return 7 (x-offset + glyph width). However this might look odd if we draw a frame (width 9) around this glyph and place the glyph inside the frame, because then there is an empty pixel col on the left, but the "C" sticks to the right frame border.
It would be better to have a width of 8: Then there would be an empty column on both sides of the "C". 
The true size of the "C" is 6, however the outside code does not know the x-offset, so drawing a frame around the "C" would be impossible if the width is returned as 6 (the x-offset would still be there and the right frame border wrould cross the right part of the "C").

olikraus, 2021-08-19:
Here is another example with the new behavior:
[image]

For the old behavior the width would be 23 and right border of the frame would be directly placed after the "e" without gap: With the new (balanced) behavior the gap will be there on the left and the right side.

Note: The gap on the left side is caused by the gylph "C" itself. There are other fonts where we would not see such a gap.

closed 2021-08-19 by olikraus

egnor, 2024-03-10:
What if I don't want the gap at all? Is there any way to query the X offset (leftmost pixel that's set), or the X extent (rightmost pixels that's set)?