olikraus/u8g2 #1563: Frequent display updates lead to Arduino crash

AmbroAnalog, 2021-08-20:
Hello,

I use your very nice library to display serial data on a small display.
The display is updated relatively often (~ 150 ms) and for a long time (1 to 2 hours).

The program is executed on an arduino pro mini.

Unfortunately, during the execution of this loop, the arduino hangs after about 10 to 60 minutes.
The display is not updated anymore and most of the time some pixels in the last frame are "Shifted".

Do you see any problem with how I use your library in the loop, anything that could cause problems?

Here is my (shortened) code:

```
U8G2_ST7567_HEM6432_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ PIN_LCD_RESET);

void setup(void) {
  u8g2.begin();

  u8g2.setDisplayRotation(U8G2_R2);

  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_t0_22b_mr);
  u8g2.drawUTF8(ALIGN_CENTER("MIELE"), 28, "MIELE");
  u8g2.sendBuffer();
}

void loop(void) {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_crox4hb_tf);
  u8g2.setContrast(255);

  //Serial Received Loop
  while (Serial.available() > 0) {
    ...
  }

  //DISPLAY DETALS
  u8g2.setFont(u8g2_font_u8glib_4_tf);
  u8g2.setCursor(8, 7);
  u8g2.print(temperatur);
  u8g2.drawUTF8(16, 7, "°C");

  u8g2.setFont(u8g2_font_crox4hb_tf);
  u8g2.setCursor(ALIGN_CENTER("00:00"), 24);

  u8g2.drawRFrame(0, 27, 64, 5, 1);
  u8g2.drawBox(1, 28, 1+PixelProgress, 3);
  
  u8g2.sendBuffer();

  delay(150);
}
```

Thank you for any advice :)

olikraus, 2021-08-20:
> Unfortunately, during the execution of this loop, the arduino hangs after about 10 to 60 minutes.
The display is not updated anymore and most of the time some pixels in the last frame are "Shifted".

The question is: What exactly hangs? Is it your Arduino Board or does the display not work any more.
Maybe you can let the on board LED blink and check whether it continues to blink after your display is frozen.

If it turns out that only the display has stopped to work, then: Check power supply and wiring to your display.
How is the display reset designed. Is the reset of the display controlled by the Arduino Board (this would be the best).
Displays are also sensitive to power supply glitches. Maybe add an additional cap (1uF or so) close to the power supply lines#
of the display. Wires to the display (SPI) should be short and soldered. You could also try to reduce bus speed (see u8g2 https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock) to minimize errors on the display.
If the display really hangs, you could execute u8g2.begin() once again to reset the display.

AmbroAnalog, 2021-08-21:
Thank you for your quick reply.

I just ran the program again and it is indeed the whole Arduino that hangs (the LED stops flashing).

I could also observe that if I decrease the pause time of the loop (more frequent execution) the error occurs faster than if the loop pause is large.
I had therefore suspected that perhaps the memory of the board runs full.
However, it would not fit that the error always occurs after a very different time and sometimes not at all.

Do you have another idea?

olikraus, 2021-08-21:
U8g2 does not allocate memory, so there can not be a memory overflow caused by u8g2.
In fact I have never heard about any crash with u8g2, except for a I2C timeout caused by the watchdog with some ESP boards.
If you have an ESP board, then maybe you can test with a disabled watchdog timer.

AmbroAnalog, 2021-08-22:
I was able to find the location in the code where the Arduino always crashes.
`u8g2.sendBuffer();`
This line seems to be where the error occurs.

Do I have a possibility to catch errors at this point or to recognize them in advance?
Could difficulties in sending cause the error?

olikraus, 2021-08-22:
Is this an ESP board?
As mentioned before: Did you try to disable the watchdog?

closed 2022-03-04 by olikraus