olikraus/u8g2 #1565: Add ERM19264 UC1609 display support

mikelemo, 2021-08-24:
Noticed the ERM19264 UC1609 was missing not sure how easy or hard of conversion to support this display would be but I've found the datasheet for the controller I'd love to get some guidance on how the u8g2 works internally per SPI display to help add support to it.

[Aliexx link](https://www.aliexpress.com/item/4001167985085.html?spm=a2g0o.cart.0.0.2ef43c00I1rGn4&mp=1)

[UC1609 Datasheet](https://www.buydisplay.com/download/ic/UC1609_Datasheet.pdf)

mikelemo, 2021-08-24:
UC1610 edited Template:

`/*

  u8x8_d_uc1610.c
  
  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

  
  cad001
  
*/
#include "u8x8.h"

static const uint8_t u8x8_d_uc1609_erm19264_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */

  U8X8_CA(0x0f1, 0x067),		/* set COM end (display height-1) */
  U8X8_C(0x0c0),            			/* SEG & COM normal */
  U8X8_C(0x040),            			/* set scroll line lsb to zero */
  U8X8_C(0x050),            			/* set scroll line msb to zero */
  U8X8_C(0x02b),            			/* set panelloading */
  U8X8_C(0x0eb),            			/* set bias 1/2 */  
  U8X8_CA(0x081, 0x05f),            	/* set contrast */
  
  /*
    AC0:	0: stop at boundary, 1: increment by one
    AC1: 	0: first column then page, 1: first page, then column increment
    AC2:	0: increment page adr, 1: decrement page adr.
  */
  U8X8_C(0x08b),            			/* set auto increment, low bits are AC2 AC1 AC0 */
  
  /*
    LC0:	0
    MX:	Mirror X
    MY:	Mirror Y
  */  
  U8X8_C(0x0c0),            			/* low bits are MY, MX, LC0 */
  
  U8X8_C(0x0f8),            			// window mode off
  U8X8_C(0x010),		                // col high
  U8X8_C(0x000),		                // col low
[... 197 lines cut ...]
      do
      {
	c = ((u8x8_tile_t *)arg_ptr)->cnt;
	ptr = ((u8x8_tile_t *)arg_ptr)->tile_ptr;
	do
	{
	  u8x8_cad_SendData(u8x8, 16, u8x8_convert_tile_for_uc1609(ptr));
	  ptr += 8;
	  x += 8;
	  c--;
	} while( c > 0 );
	
	arg_int--;
      } while( arg_int > 0 );
      
      u8x8_cad_EndTransfer(u8x8);

    
      break;
    default:
      return 0;
  }
  return 1;
}
`

olikraus, 2021-08-28:
According to the ae link i have renamed this display to u8x8_d_uc1609_slg19264.

olikraus, 2021-08-28:
I have integrated your template and updated the repo. I have also orderted one of those LCDs but I assume shipping will take 6 weeks or so to my house.

olikraus, 2021-08-28:
> to get some guidance on how the u8g2 works internally per SPI display to help add support to it.

You just need to call the functions prefixed with `u8x8_cad`. Depending on the user interface configuration, they will take care on the details.

mikelemo, 2021-09-02:
> I have integrated your template and updated the repo. I have also orderted one of those LCDs but I assume shipping will take 6 weeks or so to my house.

Cool they are actually pretty good cheap thin and useable displays my e lab is just in a bit of a mess now and I ended up using the nokia5110 display as a template modified it a bit and managed to get the code compiled(last time it was just an error in the way I edited the code) but this doesn't seem to show any graphics.
Once the lab is in working condition I'll pop up the logic analyzer and see what singnals get put out if any and compare with a working arduino library for the display.

matthijskooijman, 2021-10-01:
I'm having an issue building current master, which seems to lead me back to the commits implementing this issue.

Here's the error I get:

    u8g2/csrc/u8g2_d_setup.c:3388: undefined reference to `u8x8_d_uc1609_slg19264'

Reverting a32aa5cb83cdbae64f17da3bf2edeb4e84c02a83 fixes the build for me.

It looks like `u8x8_d_uc1609.c` was maybe forgotten to be git added before committing?

mikelemo, 2021-10-01:
I've actually decided to order a proper logic analyzer to see how the displays likes his signals but I see you've already went a head and proceed with it.

Anyways I tryed initializing the display with 

```
u8g2_Setup_uc1609_slg19264_f(
      &u8g2,
      U8G2_R0, // Display Rotation U8G2_R0-NoRot, U8G2_R1-90Deg U8G2_R2-180Deg
      u8g2_esp32_spi_byte_cb,
      u8g2_esp32_gpio_and_delay_cb); // init u8g2 structure

```

but the copiler says it has an un defined reference to "u8x8_d_uc1609_slg19264" in the newest current version, anything I missed?

olikraus, 2021-10-03:
> but the copiler says it has an un defined reference to "u8x8_d_uc1609_slg19264" in the newest current version, anything I missed?

I think this was pointed out by @matthijskooijman and should be solved already in the repo

olikraus, 2021-10-03:
Hmm... I took over your template, but I think the UC1610 has a lot of differences. Actually the UC1610 is a graylevel controller, while the UC1609 is a monochrome controller... some fixes are required here.

olikraus, 2021-10-03:
``` 
U8G2_UC1609_SLG19264_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8) 
```
I have created beta release 2.32.4. Can you test whether this works for you?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2021-10-03:
[image]

mikelemo, 2021-10-04:
`FAILED: Distance_Roller_Meter.elf 
cmd.exe /C "cd . && C:\Coding\ESP32\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\Distance_Roller_Meter.elf.rsp  -o Distance_Roller_Meter.elf  && cd ."
c:/coding/esp32/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/u8g2/lib..a(u8g2_d_setup.c.obj):(.literal.u8g2_Setup_uc1609_slg19264_f+0x4): undefined reference to `u8x8_d_uc1609_slg19264'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

I'm building the C version with esp-idf under the setup function I posted above and it 
gives me this error when trying to build after a full clean.

**To fix it** add "csrc/u8x8_d_uc1608.c" to CMakeLists.txt does it.

Beside that it works in version 2.32.6, you're the best!

Could you kindly drop a quick description on the general process you had to go throgh to adapt that display driver to the library? maybe add that to the documentation as "How to create a new u8g2 driver for a display"

[image]

olikraus, 2021-10-04:
Thanks for the feedback.
Some documentation is here: https://github.com/olikraus/u8g2/wiki/internal#new-displays

mikelemo, 2021-10-05:
Well I meant something with more reference to an actual development with a real world example if that's simple enough which it seems to be the case with this awesome library, but what ever thanks for help quit useful hopefully I'll be able to contribute with some other unsupported display I have in my dusplays boxs, fuess this case is closes untill then.😎

olikraus, 2021-10-05:
I always happy to support new displays, but I always thought u8g2 covers most of them.

mikelemo, 2021-10-06:
Slowly but surely it will, this project is on the right track.

closed 2021-11-28 by mikelemo