olikraus/u8g2 #1572: Feature: Add a battery charging icon in battery/emoji fonts

ShivamJoker, 2021-09-01:
I looked into the icons but there are only different levels of battery only, getting a charging one would really help.
Thanks.

olikraus, 2021-09-01:
Maybe you can update and contribute a new font here. I am always happy to add additional content.

ShivamJoker, 2021-09-01:
@olikraus I can definitely, can you tell me the process or the docs on how we can modify it ?

olikraus, 2021-09-01:
I think the sources for the icons you mentioned are here: 
https://github.com/olikraus/u8g2/tree/master/tools/font/battery

I had used fony to do the editing: 
https://github.com/olikraus/u8g2/tree/master/tools/font/fony

Fony basically exports this .bdf file, so if you send me an updated .bdf file, then I will be able to update the repo.

ShivamJoker, 2021-09-01:
Actually I am using mac and I tried using the pngToBdf but it doesn't seem to work or I am missing something ?
[image]

Here is the battery charging PNG
[ [image] ]

Not sure how else I can generate the .bdf
Also why can't we use something like ttf or otf ?

olikraus, 2021-09-01:
ttf is possible, but needs to be converted to bdf first.
A ttf font render engine will not fit into your microcontroller, so using ttf or otf is not an option.

Another idea: You have a png file, so why not converting the same to .xbm and insert this xbm directly in your code?
There is no need to use a font, just draw the XBM at the desired position.

ShivamJoker, 2021-09-01:
Yesterday I figured out about the XBM, but it would be helpful if we have things already built into the library so that others can directly use it.

If you have any other way to convert to BDF let me know, can't find anything online.

olikraus, 2021-09-01:
Maybe you could just attach the PNG file then.

ShivamJoker, 2021-09-01:
its already there 

https://user-images.githubusercontent.com/23727670/131672379-27e6dfdd-786c-44c2-b5a8-8ed4c9073747.png

It will be really helpful if you would add it

olikraus, 2021-09-01:
hmmm... your png is a graylevel picture, so it is more or less useless.
I have designed a new icon:
[image]

here is the font file file:
``` C++
/*
  Fontname: battery19
  Copyright: Created with Fony 1.4.7
  Glyphs: 8/8
  BBX Build Mode: 0
*/
const uint8_t battery19[185] U8G2_FONT_SECTION("battery19") = 
  "\10\0\3\3\4\5\1\1\5\10\23\0\0\23\0\23\0\0\0\0\0\0\234\60\13\70\317\42\35\304\372\377"
  "\371 \61\17\70\317\42\35\304\372\177\241H(\342\203\0\62\23\70\317\42\35\304\372/\24\11E,\241H"
  "(\342\203\0\63\27\70\317\42\35\304\372\205\42\241\210%\24\11E,\241H(\342\203\0\64\33\70\317\42"
  "\35\304\272P$\24\261\204\42\241\210%\24\11E,\241H(\342\203\0\65\37\70\317\42\35\304\22\212\204"
  "\42\226P$\24\261\204\42\241\210%\24\11E,\241H(\342\203\0\66\27\70\317\42\35\304\332B\262\220"
  "^(\22\212H\247\230(&V>\10\67\5\0\316\0\0\0\0\4\377\377\0";
```

ShivamJoker, 2021-09-02:
Okay thanks a lot. 
So now is this included in library or will have to modify it ?

olikraus, 2021-09-02:
It will be included in the next release. Until then you can include the code above and assign battery19 to setFont

ShivamJoker, 2021-09-02:
Sure thanks a lot.

pike2k, 2021-10-05:
This explains why the charging gfx did not work with 2.28 :+1: 
I read the wiki and found the gfx, but it turns out that the manual is ahead of the releases :1st_place_medal: 

Other than adding the above code snippet to my sketch, what else do I need to do to use the charging gfx with 2.28?

olikraus, 2021-10-05:
Nothing else should be required

olikraus, 2021-10-05:
Current release is 2.31

pike2k, 2021-10-05:
I am using PlatformIO and I guess I can force a later version than 2.28 somehow...

Does the code snippet add the charging gfx to battery19 even in 2.28? If so, at which 'address'? I guessed it was 54 (+1 from Full charge), but got no visible result.

olikraus, 2021-10-05:
Yes, if you use setFont(battery19), then the charging icon will be added  (provided that the above array is defined before the setFont statement) and yes, it should be 54 (see also https://github.com/olikraus/u8g2/issues/1600)

Ritika-Das, 2021-10-05:
Just define the battery font 
```
// Battery font
const uint8_t battery19[185] U8G2_FONT_SECTION("battery19") = 
  "\10\0\3\3\4\5\1\1\5\10\23\0\0\23\0\23\0\0\0\0\0\0\234\60\13\70\317\42\35\304\372\377"
  "\371 \61\17\70\317\42\35\304\372\177\241H(\342\203\0\62\23\70\317\42\35\304\372/\24\11E,\241H"
  "(\342\203\0\63\27\70\317\42\35\304\372\205\42\241\210%\24\11E,\241H(\342\203\0\64\33\70\317\42"
  "\35\304\272P$\24\261\204\42\241\210%\24\11E,\241H(\342\203\0\65\37\70\317\42\35\304\22\212\204"
  "\42\226P$\24\261\204\42\241\210%\24\11E,\241H(\342\203\0\66\27\70\317\42\35\304\332B\262\220"
  "^(\22\212H\247\230(&V>\10\67\5\0\316\0\0\0\0\4\377\377\0";
 ```
 Set battery font using
 ```
 display.setFont(battery19);
 ```
 And use the charging icon by putting 36 address
 ```
  // Battery Charging
  display.drawGlyph(5, 32, 0x0036);
 ```

pike2k, 2021-10-06:
Thanks all for help. I found that on my small 128x32 display this glyph was a tiny bit small (details was hard to notice unless I hold it very close to my eyes), so I am making a 24x20px version using Fony. It will have some more alterations also. When I am done, I will donate it to this lovely project.

closed 2022-03-04 by olikraus