olikraus/u8g2 #1573: Feature: allow to enable/disable features wiout modifying library headers (u8g2.h, u8x8.h etc)

positron96, 2021-09-01:
It would be nice to be able to disable/enable features by providing `-D` to gcc or options in IDEs and not by modifying library headers (commenting out `#defines`). Now if I tweak the library options, I either need to clone my own copy or to put the source into app git repo. Or specify in README what needs to be changed by every user who clones my app.

To kretain compatibility, I propose to wrap every `#define WITH_X` in library headers get wrapped into 
```
#ifndef WITHOUT_X
  #define WITH_X
#endif
```

So to disable WITH_X feature you will pass -DWITHOUT_X to compiler.

If there is an interest from community, I can spend some time to provide a minimal proof-of-concept PR.

olikraus, 2021-09-01:
Nice suggestion. However I do not know whether it is really required to allow such fine controll over the library configuration.
The most often used option seems to be the 16 bit macro:
https://github.com/olikraus/u8g2/blob/6eb5b91a7da758cc661f5a3f25d8b9d10279ae94/csrc/u8g2.h#L75-L79
This already supports "-D"

positron96, 2021-09-01:
Yes, I've used U8G2_16BIT in my project, though in PlatformIO the library does not enable it per-architecture (probably old version). Also, the code you posted does __not__ allow to use 8bit on ARM, ESP8266 etc if the user desires it for some reason.

In my project I am desperately struggling to free flash, so I had to disable features. The first was to get rid of I2C that I don't even use (all the I2C stuff on stm32duino took around 8Kb)

olikraus, 2021-09-01:
> In my project I am desperately struggling to free flash, so I had to disable features. The first was to get rid of I2C that I don't even use (all the I2C stuff on stm32duino took around 8Kb)

Yes, I would say that had been the main intention to introduce those #defines 

For example if you do not need font rotation, then disabling the font direction would save a good amount of flash memory:
https://github.com/olikraus/u8g2/blob/6eb5b91a7da758cc661f5a3f25d8b9d10279ae94/csrc/u8g2.h#L124

Using 8 Bit version of U8g2 on a 32 bit system only saves RAM but not flash memory (at least according to my own investigation). You can still forcefully enable 8 bit version in .h file by modifing the code ;-)

I still feel that the .h file is a good place to configure the library. I see the disadvantage of modifing the lib itself regarding any future updates, but for me it is still difficult to find time for such an update. Also not sure how many users are really interessted in this. Still U8g2 is an Arduino Lib and Arduino Users will even never see those configuration options.

positron96, 2021-09-01:
PlatformIO is a nice upgrade from Arduino IDE. It allows to use same framework & libraries while giving a lot more control, including changing compiler flags (most probably, you already know that). In the recent years I see a lot of amateur-DIY-tinkerer-youtubers suggest viewers to transition from Arduino to VSCode+PlatformIO.

Anyway, to make my project managable, I'll introduce some #ifdefs in my u8g2 fork and maybe create a PR. I actually don't see how introducing these changes may harm the library as full compatibility will be retained.

olikraus, 2021-09-01:
PR is fine

closed 2021-09-02 by positron96

olikraus, 2021-09-02:
Thanks for the PR

positron96, 2021-09-02:
No problem) Any estimate on when it hits U8g2_Arduino?

olikraus, 2021-09-02:
> Any estimate on when it hits U8g2_Arduino?

maybe today if I finish first draft for #1575

olikraus, 2021-09-02:
Beta 2.29.10 is there new.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

positron96, 2021-09-07:
Great! Btw, in platformio one can specify a library via github link, complete with a git tag.