olikraus/u8g2 #1575: ST7571 LCD 128x96 HELP

rayzhang86, 2021-09-02:
Hi Olikraus
I want to use an LCD, the driver chip is ST7571, the resolution is 128x96. After querying the u8g2 library, I found that u8x8_d_st7571.c only supports a resolution of 128x128. How can I modify some parameters to use this driver? Or can you please update the driver library? Thank you for your generous help.
The project is based on Arduino IDE (ESP8266)
LCD information link: http://www.jlxlcd.cn/html/zh-detail-1211.html

olikraus, 2021-09-02:
We probably need to update the startup sequence. This is what I read on the PDF, which I found via your link:
```
   ST=0; 
  delay(500); 
  RST=1; 
  delay(100); 
  transfer_command(0x2c); 
  delay(200); 
  transfer_command(0x2e); 
  delay(200); 
  transfer_command(0x2f); 
  delay(10);    
  transfer_command(0xae);   //显示关 
  transfer_command(0x38);   //模式设置 
  transfer_command(0x94);   //75HZ     
  transfer_command(0x48);    //duty 
  transfer_command(0x61);    //96 
  transfer_command(0x4c);     //n-line    
  transfer_command(0x0a);     //12-line(0x0a),0-line(0x00) 
transfer_command(0xab); 
  transfer_command(0x67); 
  transfer_command(0x27);   //粗调对比度，可设置范围 0x20～0x27 
  transfer_command(0x81);   //微调对比度 
  transfer_command(0x28);   //微调对比度的值，可设置范围 0x00～0x3f 
  transfer_command(0xa0);   //列扫描顺序 
  transfer_command(0xc8);   //行扫描顺序   
  transfer_command(0x44);   //Set initial COM0 register 
  transfer_command(0x20); 
  transfer_command(0x40);   //Set initial display line register 
  transfer_command(0x00); 
  transfer_command(0x57);   //0x57  1/12 bias 
  transfer_command(0xf3); 
  transfer_command(0x04)
  transfer_command(0x93); 
  transfer_command(0x10);   
  transfer_command(0xaf);   //显示开 
```

olikraus, 2021-09-02:
I have created a new beta release 2.29.10.
Maybe you can test the new ST7571 constructor for your display. Can you post a picture of the u8g2 page buffer flip mode example?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

rayzhang86, 2021-09-03:
Thank you very much for your help, I will test and feedback as soon as possible

olikraus, 2021-09-03:
Moved to next milestone

olikraus, 2022-01-07:
Any feedback/results?

rayzhang86, 2022-01-18:
It works well! I use HW SPI structure like this:

#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
U8G2_ST7571_128X96_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/0 );

[image]

closed 2022-03-04 by olikraus