olikraus/u8g2 #1577: drawButtonUTF8

olikraus, 2021-09-03:
Add drawButtonUTF8 procedure.

```
void u8g2_DrawButtonUTF8(u8g2_t *u8g2, u8g2_uint_t x, u8g2_uint_t y, u8g2_uint_t flags, u8g2_uint_t width, u8g2_uint_t padding_h, u8g2_uint_t padding_v, const char *text);

```

narder21, 2021-09-07:
hi, im not fully sure if im just being incompetent, but im trying to use this function on an esp32 with a OLED  and for whatever reason this member is not found. Any reason this oculd be?
[image]

olikraus, 2021-09-07:
It should be there in the latest release 2.29.x. maybe check your lib manager in Arduino IDE.

narder21, 2021-09-07:
[image]
im pretty sure im on 2.29.11, any other reason for this?

olikraus, 2021-09-07:
Hmm... yes, I checked the repo. It looks like, that I missed to create the C++ member function. Nice Finding.

olikraus, 2021-09-07:
Workaround would be the use of the C variant of that function:
`u8g2_DrawButtonUTF8(u8g2.GetU8g2(), ...)`

narder21, 2021-09-07:
ah good point, thanks a lot for the help and the library, great work.

olikraus, 2021-09-07:
created bugfix release 2.30.1

closed 2021-09-07 by olikraus

olikraus, 2021-09-08:
Release 2.30.1 also contains a new example ButtonEmoticon to demonstrate the drawButtonUTF8 function