olikraus/u8g2 #1578: Incorrectly documented setup call for SSD1306 i2c in Wiki

happymacer, 2021-09-04:
Oli, many thanks for your trouble to write this stuff!  Today I got it working on a barebones HW I2C on the Atmega328p using Microchip Studio (GCC) and that brought much joy!  I have found what I thought is a documentation bug in the 8x8 Wiki C setup for the SSD1306 128x64_noname setup.  I thought it would make life easier for other newbies like me to raise it...

The setup in the "C" Wiki is: 
u8x8_Setup(u8x8_d_ssd1306_128x64_noname, u8x8_cad_001, uc specific, uC specific);

and I think it should be: 
u8x8_Setup(**u8x8_t \*u8x8**, u8x8_d_ssd1306_128x64_noname, **u8x8_cad_ssd13xx_i2c,** uc specific, uC specific);

or if using fast i2c : 
u8x8_Setup(**u8x8_t \*u8x8**, u8x8_d_ssd1306_128x64_noname, **u8x8_cad_ssd13xx_fast_i2c**, uc specific, uC specific);

When copied the setup as posted I got a "missing argument" compile error and gibberish on a flashing screen when using u8x8_cad_001.   With the fixes it works beautifully!

It seems to apply to all the SSD1306 "C" setup strings in the Wiki.

Another titbit, if you care to add it to your (otherwise working) AS7 AVR setup for HW i2c instructions is to add the "AVR_USE_HW_I2C" as a defined symbol...  I got stuck there for a long time with "error: undefined reference to "u8x8_byte_avr_hw_i2c" 

If you have no objection, Ill have a bash at producing a short Youtube video with the setup in AS7 I use with a simple example.

olikraus, 2021-09-04:
Nice finding, thanks a lot. 
Videos would be great also.

olikraus, 2021-09-04:
I added the first argument to the c setup. Looks like noone ever noticed this. It must be there since the beginning of this project :-/

> Another titbit, if you care to add it to your (otherwise working) AS7 AVR setup for HW i2c instructions is to add the "AVR_USE_HW_I2C" as a defined symbol... I got stuck there for a long time with "error: undefined reference to "u8x8_byte_avr_hw_i2c"

`u8x8_byte_avr_hw_i2c` is not part of this project. Probably it is contributed from somewhere. From where did you receive this procedure?
Maybe it would make sense to update this wiki: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#atmel-avr

happymacer, 2021-09-04:
Cool thanks. Im happy to update the https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#atmel-avr if I can get access...not sure if I can.

`u8x8_byte_avr_hw_i2c` is provided as part of the download package in the u8x8_avr.c code in the ......\u8g2-master\sys\avr\avr-libc\lib directory for hardware i2c. **u8x8_avr_delay** is also there.

olikraus, 2021-09-04:
Ah, ok well, that code was contributed by someone else.

olikraus, 2021-09-04:
Hmm... yes, I see the problem:
https://github.com/olikraus/u8g2/blob/e5b82a6eacb30025ab73323fc54a0bb1e1405a24/sys/avr/avr-libc/lib/u8x8_avr.c#L1-L8

olikraus, 2021-09-04:
I have updated the readme:
https://github.com/olikraus/u8g2/blob/master/sys/avr/avr-libc/README

happymacer, 2021-09-12:
> Videos would be great also.

Hiya, Ive produced a video in Youtube - https://youtu.be/TO7-yxT6Gvw.  Please feel free to link to it

olikraus, 2021-09-12:
Thanks for the video. Great contribution. I have placed a link on the porting page: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

closed 2021-09-12 by happymacer