olikraus/u8g2 #1579: UTF issues. Cannot make cyrillic characters to work in custom font.

deividAlfa, 2021-09-04:
I've been using u8g2 for a while now, I must say it's a great library!
Recently I tried to add russian translation using **fonts/bdf/unifont.bdf**, adding the [cyrillic block](https://en.wikipedia.org/wiki/Cyrillic_(Unicode_block)) from 0x0410 to 0x044F.
So, I made the font with bdfconv (Standard latin chars+cryrillic):
>bdfconv_2_22.exe -v -b0 -f1 -m "$20-$7E, $410-$44F" "../bdf/unifont.bdf" -o c\font_custom.c  -n u8g2_font_custom

What's the difference between **bdfconv_2_22.exe** and **bdfconv.exe**? It's confusing, I didn't found anything about.

The font was correctly generated. Skipping the extra code, this is what it's being done:
>u8g2_SetFontMode(&u8g2,1);                                  // Set font transparent
>u8g2_SetFontDirection(&u8g2, 0);                            // No rotation
>u8g2_SetFontPosTop(&u8g2);                                  // Take upper font ref. as start drawing position
>u8g2_SetFont(&u8g2,u8g2_font_custom);
>u8g2_SetDrawColor(&u8g2, WHITE);
>u8g2_DrawUTF8(&u8g2,0,13,"English_1234");
>u8g2_DrawUTF8(&u8g2,0,26,"русский_1234");

The standard chars are printed, but russian characters are being skipped (Only "_1234" is drawn).
If I use an existing cyrillic font (ex. u8g2_font_8x13_t_cyrillic) it works fine.

I use  STM32 Cube IDE (Eclipse), and the encoding is set to UTF-8.
The encoding seems ok, ex. cyrillic 'р' (0x420) is encoded as 0xD0, 0xA0 , and u8g2 decodes it back to 0x420.

Extending the range to $400-$500 to see if it made any difference caused an endless loop in **u8g2_font_get_glyph_data**.
To make it worse, everything worked when extending the range to $300-$600, just no sense, and wastes a lot of space by adding glyphs I don't need.

olikraus, 2021-09-04:
The exe might be a little bit outdated. I do not always regenerate the windows exe. Moreover there are some commandline issues with the $ sign (maybe). I created the font under my linux environment.

commandline for linux:
https://github.com/olikraus/u8g2/blob/e5b82a6eacb30025ab73323fc54a0bb1e1405a24/tools/font/bdfconv/Makefile#L28-L32

[image]

```
/*
  Fontname: -gnu-Unifont-Medium-R-Normal-Sans-16-160-75-75-c-80-iso10646-1
  Copyright: Copyright (C) 1998-2019 Roman Czyborra, Paul Hardy,  Qianqian Fang, Andrew Miller, Johnnie Weaver, David Corbett, et al.  License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>  with the GNU Font Embedding Exception.
  Glyphs: 159/57086
  BBX Build Mode: 0
*/
const uint8_t unifont_cyrillic_44f[2197] U8G2_FONT_SECTION("unifont_cyrillic_44f") = 
  "\237\0\3\2\4\4\4\5\5\10\20\0\376\12\376\13\377\1\214\3\31\4\222 \5\0\350\61!\10\241\14"
  "\61\16\211\0\42\10E\212\61\62\267\0#\17\246\11\261z\32\206$j\31\206\250'\0$\21\247\11\361"
  "\302A\211\42)\234\307H\252\14b\6%\24\247\11q\64))%\221\222\306i\42%QR\322\24\0"
  "&\21\247\11\261\266RV\11\305$\322\22\61\311\244)'\7A\214\61\206\0(\14\303\373\260\222(\211"
  "z\213\262\0)\14\303\372\60\262(\213z\211\222\10*\15w\31\361J\225\266-i\252e\0+\13w"
  "\31\361\342\332\60dq\15,\10B\353\60\224D\1-\7\24J\61\206\0.\7\42\13\61\206\0/\14"
  "\246\11q\213\325\60\15\253)\0\60\21\246\11\261\264(\11\265)Q\242ML\242L\2\61\13\245\12\261"
  "\62)\11\373\64\10\62\16\246\11q\206$\24\323L\13\253\351\60\63\20\246\11q\206$\24\323hNE"
  "\61\31\22\0\64\20\246\11\61C-\211*Y\222%\303\230V\0\65\16\246\11\61\16iu\220\323TL"
  "\206\4\66\16\246\11\261\246\60M\7%tL\206\4\67\13\246\11\61\256\305\264\230\66\1\70\17\246\11q"
  "\206$\64&C\22:&C\2\71\16\246\11q\206$\64&\203\332\30M\0:\10r\33\61\206x\10"
  ";\12\222\373\60\206XI\24\0<\10\225\12\61\263\256\35=\10V)\61\356\304a>\11\225\11\61\322"
  "n\35\1\77\16\246\11q\206$\24\323\260\232\203i\4@\22\246\11\261\246LJ\224d\211\224H\211\244"
  "%\36\2A\15\246\11\261\264\250%\24\207At\14B\16\246\11\61\6%\64\16K\350\70,\0C\15"
  "\246\11q\206$\264v\24\223!\1D\15\246\11\61\206(KB\277%C\4E\14\246\11\61\16iu"
  "P\322\326aF\14\246\11\61\16iuP\322\256\0G\16\246\11q\206$\264\226\206\320\246,\1H\13"
  "\246\11\61B\307a\20=\6I\12\245\12\61\6)\354\247AJ\15\247\11\261\6\61\356)\213\262l\3"
  "K\21\246\11\61B-\211*\231(&YTK\302\0L\11\246\11\61\322\376:\14M\15\246\11\61B"
  "q\32\242\305\243c\0N\20\246\11\61\302mS\42%\222\22)\321\216\1O\14\246\11q\206$\364\307"
  "dH\0P\14\246\11\61\6%\64\16K\332\25Q\26\267\371p\206(L\302$L\302$L\302$Q"
  "\22I\32r@R\20\246\11\61\6%\64\16KTK\262$\24\3S\15\246\11q\206$\64;\212\311"
  "\220\0T\12\247\11\61\16Y\334\337\0U\13\246\11\61B\377\230\14\11\0V\21\247\11\61Rk\222E"
  "Y\224U\302$\215\63\0W\15\246\11\61B/.\323\20\215b\0X\17\246\11\61B\61\211\332D-"
  "j\11\305\0Y\16\247\11\61R\65\311\242\254\222\306\335\0Z\12\246\11\61\256\305^\323a[\11\303\374"
  "\60\206\250\177\32\134\13\246\11\61\322\270\32\247q\65]\11\303\371\60\246\376i\10^\11\66\231\261\264("
  "\11\3_\7\27\371\60\16\1`\7\63\252\61\262\2a\16\206\11q\206$L\223a\264)K\0b\15"
  "\266\11\61\322\226E\23\35\67e\1c\14\206\11q\206$T;&C\2d\14\266\11q[\26m\364"
  "\246,\1e\16\206\11q\206$\24\207A-&C\2f\14\265\11\361\244\260\64HaO\0g\23\266"
  "\351p\223EK\262$\213\266tHB\61\31\22\0h\13\266\11\61\322\226E\23}\14i\13\265\12\261"
  "\302\34\21\373\64\10j\14\325\351\60\353\210\330G)\222\0k\20\266\11\61\322\266$\252db\222E\265"
  "$\14l\11\265\12q\304\376i\20m\21\207\11\61\26%\212\244H\212\244H\212\244H*n\12\206\11"
  "\61\222E\23}\14o\14\206\11q\206$\364\61\31\22\0p\16\246\351\60\222E\23\35\67eIS\0"
  "q\14\246\351p\26m\364\246,i\1r\13\206\11\61\222E\23\325\256\0s\14\206\11q\206$\224\35"
  "\223!\1t\13\245\11\261\302\322 \205]\5u\12\206\11\61B\277)K\0v\14\206\11\61Bc\22"
  "\365&J\0w\21\207\11\61R)\222\42)\222\42)\222*\26\0x\16\206\11\61B\61\211\62Q\213"
  "\222P\14y\15\246\351\60B\217IdI+C\2z\12\206\11\61\256a\257\303\0{\16\324\352\260\224"
  ",\314\242b-\312B\1|\7\341\354\60>\10}\16\324\352\60\304,\312\302R-\314\22\11~\12\67"
  "\211q\64)\322\24\0\0\0\0\4\377\377\4\20\16\246\11\261\264\250%\24\207At\14\4\21\17\246\11"
  "\61\6%\255\16J\350\70,\0\4\22\17\246\11\61\6%\64\16K\350\70,\0\4\23\12\246\11\61\16"
  "i\177\5\4\24\23\310\350\60\307(\214\302(\353S\30\205\311p\226\3\4\25\15\246\11\61\16iuP"
  "\322\326a\4\26\23\247\11\61\242H\252\64%\265qKJIS$\25\4\27\21\246\11q\206$L+"
  "C\32\247b\62$\0\4\30\21\246\11\61B\355\224H\211\22)\321\266\211\1\4\31\24\326\11q\242L"
  "GB\355\224H\211\22)\321\266\211\1\4\32\21\246\11\61\262\251K\226db\222E\265$\14\4\33\16"
  "\246\11\261\206\250\377K\226db\0\4\34\16\246\11\61Bq\32\242\305\243c\0\4\35\14\246\11\61B"
  "\307a\20=\6\4\36\15\246\11q\206$\364\307dH\0\4\37\12\246\11\61\16\241\177\14\4 \15\246"
  "\11\61\6%\64\16K\332\25\4!\16\246\11q\206$\264v\24\223!\1\4\42\13\247\11\61\16Y\334"
  "\337\0\4#\22\247\11\61R\65\311\242\254\22&i\234\246\42\0\4$\24\267\11\361\302A\211\42)\222"
  "\42)\222*\203\30\327\0\4%\20\246\11\61B\61\211\332D-j\11\305\0\4&\27\307\351\60\302$"
  "L\302$L\302$L\302$L\302d\30\342\2\4'\13\246\11\61B\337\224%m\4(\26\247\11\61"
  "\242H\212\244H\212\244H\212\244H\212\244h\30\4\4)\32\310\350\60\242JT\211*Q%\252D\225"
  "\250\22U\242\312\60\350@\16\4\4*\21\247\11\61\346\346!+eQ\26e\321\220\0\4+\22\246\11"
  "\61B\307!\222\22)\221\22)\31\242\0\4,\15\246\11\61\322\326A\11\35\207\5\4-\17\246\11q"
[... cut ...]
  "\306\11\321D\355p\350MY\322\2\4\366\14\306\351P\34\322~U\213\0\4\367\14\246\351P\34\322^"
  "\325\42\0\4\370\24\346\11\321D\355p\350\70DR\42%R\42%C\24\4\371\23\306\11\321D\355p"
  "h\34\42)\221\22)\31\242\0\4\372\20\307\351P\15R\334\70\210qYN\65\0\4\373\17\247\351P"
  "\15R\34\16b\134\226S\15\4\374\22\306\351P\204b\22\265\211Z\324\22\212i\230\0\4\375\22\246\351"
  "P\204b\22e\242\26%\241\230\206\11\0\4\376\21\246\11Q\204b\22\265I\303\22\265\204b\0\4\377"
  "\20\206\11Q\204b\22e\322\260DI(\6\0";
```

deividAlfa, 2021-09-04:
I also tested **-m "32-127, 1024-1279", with the same result, I don't think  it's the issue here?
I'll try again with linux version.

deividAlfa, 2021-09-04:
Got the problem. The issue is caused by **bdconv_2_22**::

**bdfconv.exe** (Works)
>const uint8_t u8g2_font_custom[2295] U8G2_FONT_SECTION("u8g2_font_custom")

**bdfconv_2_22.exe** (Not working)
>const uint8_t u8g2_font_custom[2291] U8G2_FONT_SECTION("u8g2_font_custom")

deividAlfa, 2021-09-04:
The issue is caused by **bdconv_2_22**, the font code is different:

**bdfconv.exe** (Works)
>const uint8_t u8g2_font_custom[2295] U8G2_FONT_SECTION("u8g2_font_custom")

**bdfconv_2_22.exe** (Not working)
>const uint8_t u8g2_font_custom[2291] U8G2_FONT_SECTION("u8g2_font_custom")

olikraus, 2021-09-04:
I forgot to mention this. Yes: bdfconv_2_22.exe is for all versions before and equal to 2.22. Later versions of u8g2 require bdfconv.exe. If I remember correctly I changed the font format during that version change to 2.23.

deividAlfa, 2021-09-04:
I know documenting a big project like this isn't not easy, but perhaps adding a simple readme in bdfconv folder to avoid confusion?

A bit unrelated: Do you know any software to edit bdf that supports unicode? Fony doesn't seem compatible...

Thanks!

olikraus, 2021-09-04:
> I know documenting a big project like this isn't not easy, but perhaps adding a simple readme in bdfconv folder to avoid confusion?

There are some remarks in the U8g2 FAQ

> A bit unrelated: Do you know any software to edit bdf that supports unicode? Fony doesn't seem compatible...

Once created with fony you could change the encoding number in the generated bdf file to move the glyph to a different position.

deividAlfa, 2021-09-04:
Will close the issue and try that. Thanks a lot for yoru help!

closed 2021-09-04 by deividAlfa