olikraus/u8g2 #1581: LC7981 No 240X128

Duet356, 2021-09-08:
LCD EW24DA3FLY 240X128, Display Controller LC7981. The display only shows U8G2_LC7981_160X80. 240X128 The display does not work.

olikraus, 2021-09-09:
Can you provide the datasheet of your display?

Duet356, 2021-09-09:
Datasheet: https://drive.google.com/file/d/1_y_qH78Hpf_gV9B5D64WRw_-mQrcR74H/view?usp=sharing
 [https://drive.google.com/file/d/10qvQ2fG8EtS4JafzXYjgJ99gAa1PCQ8N/view?usp=sharing](url)

olikraus, 2021-09-09:
I tried to access but needed a request for confirmation, which I didn't receive. So, I have no access to the datasheet.

Duet356, 2021-09-09:
I beg your pardon.

[image]
The display is connected correctly, it works at a resolution of 160x80. 

[Datasheet](https://www.datasheetarchive.com/pdf/download.php?id=6e104258f946ed34c4d0a32494260318567e7a&type=M&term=ew24)

olikraus, 2021-09-09:
What happens if you try the 240x128 constructor from here:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#lc7981-240x128

Is there anything visible?

Duet356, 2021-09-09:
Used all buffer options, 1, 2, F, no image.

Duet356, 2021-09-09:
This is the only way there is an image. Arduino DUE.
[image]
`#include <Arduino.h>
#include <U8g2lib.h>
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

//U8G2_LC7981_240X128_F_6800 u8g2(U8G2_R0, 52, 50, 48, 46, 44, 42, 40, 38, /*enable=*/ 36, /*cs=*/ 34, /*dc=*/ 26, /*reset=*/ 30); // Connect RW with GND
U8G2_LC7981_160X80_F_6800 u8g2(U8G2_R0, 52, 50, 48, 46, 44, 42, 40, 38, /*enable=*/ 36, /*cs=*/ 34, /*dc=*/ 26, /*reset=*/ 30); // Connect RW with GND

void setup(void) {
  u8g2.begin();
}
void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World! Hello World! Hello World! Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}`

olikraus, 2021-09-10:
Pleas try the following, will this work?

```
#include <Arduino.h>
#include <U8g2lib.h>
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_LC7981_240X128_F_6800 u8g2(U8G2_R0, 52, 50, 48, 46, 44, 42, 40, 38, /enable=/ 36, /cs=/ 34, /dc=/ 26, /reset=/ 30); // Connect RW with GND

void setup(void) {
u8g2.begin();
u8g2.sendF("ca", 2, 240/8-1);
u8g2.sendF("ca", 3, 128);
}
void loop(void) {
u8g2.clearBuffer(); // clear the internal memory
u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
u8g2.drawStr(0,10,"Hello World! Hello World! Hello World! Hello World!"); // write something to the internal memory
u8g2.sendBuffer(); // transfer internal memory to the display
delay(1000);
}
```

If this doesn't work, can you change the line with `u8g2.sendF("ca", 3, 128);` to another value like `u8g2.sendF("ca", 3, 80);`. Are there any values other than 128 or 80 which will work?

Duet356, 2021-09-10:
u8g2.sendF("ca", 3, 80);
[image]

u8g2.sendF("ca", 3, 100);
[image]

u8g2.sendF("ca", 3, 110);
[image]

u8g2.sendF("ca", 3, 128); Nothing is displayed. Each time the u8g2.sendF parameter ("ca", 3, 110) is increased, the brightness of the display decreases.

Duet356, 2021-09-10:
Is there not enough power supply?

olikraus, 2021-09-10:
If contrast decreases (which is normal for higher values) then you need to adjust your var pot at V0. In other words higher negative voltage values at V0 are required.

olikraus, 2021-09-10:
Small update from my end. You need an external connection like described in chapter 11 (page 9, pdf page 13): This includes a variabpe potentiomenter between -17 volt  and GND. Your display probably requires -16 volt at V0 (see table at chapter 4).
I assume that something is visible, if you apply the correct -16 volt. 

> u8g2.sendF("ca", 3, 128); Nothing is displayed. 

This is because V0 has the wrong voltage. A more negative voltage is required. If you increase the voltage (adjust the var pot) then something should be visible.

However there is an issue in the code i assume. The question is:
Is 
u8g2.sendF("ca", 3, 128);
or
u8g2.sendF("ca", 3, 127);

correct?  To figure out the correct value, please load the u8g2 FlipMode example. It will display a full frame around the border.
Will the full frame visible with "u8g2.sendF("ca", 3, 127);" or with "u8g2.sendF("ca", 3, 128);" or with both?

Duet356, 2021-09-11:
Connected a potentiometer. Everything works well. The problem was not in the code. u8g2.sendF ("ca", 3, 128); is displayed perfectly. Sorry, the problem had nothing to do with the program code. Thank you so much for your help!
[image]

I did not connect V0, due to the fact that in the device where the display was installed: 
[image]
The V0 pin was not used in the device. 
[image]

olikraus, 2021-09-11:
Thanks for the feedback. Nice pictures. Nevertheless, V0 must be connected somehow. Maybe this is a multilayer PCB and the connection is invisible. I can not imagine how this could work without V0 being connected.

Duet356, 2021-09-11:
All perfectly. Thanks a lot! 
[image]

closed 2022-03-04 by olikraus