olikraus/u8g2 #1582: u8g2.userInterfaceInputValue

swb2020, 2021-09-09:
Does anyone have or can recommend a complete and working example for this function? As I understand it the, function returns an int to a location via the pointer. I may not have my pointer set right. I can I can load an int to the pointer and it will show up in the pointed location. The Select Voltage does display, but nothing changes in the location the pointer points to.
Sorry to dredge this up again. 
Thank you for any advice. 

#include <Arduino.h>
#include <U8g2lib.h>
#include "stdint.h"

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

int x = 0;
int *px;

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /*clock=*/ 23 /* A4 */ , /* data=*/ 17 /* A2 */, /* CS=*/ 16 /* A3 */, /* reset=*/ U8X8_PIN_NONE);

void setup() 
{
Serial.begin(115200);
Serial.println("Test Input Value");
u8g2.begin();
u8g2.setFont(u8g2_font_6x12_tr);
px = &x;
*px = 5;
Serial.println(x);
}
void loop() 
{
Serial.println(x * 2);   
u8g2.userInterfaceInputValue("Select Voltage", "DAC = ", *px, 1, 150, 3, " Volts");

}

olikraus, 2021-09-09:
I think 
`u8g2.userInterfaceInputValue("Select Voltage", "DAC = ", px, 1, 150, 3, " Volts");`
should be correct (px without *). Moreover the `px` is not required. You can just pass &x instead of px:
`u8g2.userInterfaceInputValue("Select Voltage", "DAC = ", &x, 1, 150, 3, " Volts");`

swb2020, 2021-09-09:
Thank you,
I tried both. The only way I could get it to compile was to declare both x and px as uint8_t. Here is the sketch that will compile display correctly. However the sketch hangs at:
u8g2.userInterfaceInputValue("Select Voltage", "DAC = ", &x, 1, 150, 3, " Volts");

&x has a red squiggly line under it. When I hover the cursor over it I get:

Cannot initialize a parameter of type 'uint8_t *' (aka 'unsigned char *') with an rvalue of type 'int *'

C:\Users\Warren\Documents\Arduino\libraries\U8g2\src\U8g2lib.h:366:82:
note: passing argument to parameter 'value' hereclang(init_conversion_failed)

I don't get this if I move:
u8g2.userInterfaceInputValue("Select Voltage", "DAC = ", &x, 1, 150, 3, " Volts");
to setup(). Is this a compiler thing?

#include <Arduino.h>
#include <U8g2lib.h>
#include "stdint.h"

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

uint8_t x = 0;
uint8_t *px;

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /*clock=*/ 23 /* A4 */ , /* data=*/ 17 /* A2 */, /* CS=*/ 16 /* A3 */, /* reset=*/ U8X8_PIN_NONE);

void setup() {
Serial.begin(115200);
Serial.println("Test Input Value");
u8g2.begin();
u8g2.setFont(u8g2_font_6x12_tr);
px = &x;
*px = 23;
Serial.println(x);
}

void loop() 
{   
u8g2.userInterfaceInputValue("Select Voltage", "DAC = ", &x, 1, 150, 3, " Volts");
// >>>>>>>>>>>>>>>>>   program does not enter this if statement.....
if (x == 1)
{
Serial.print("Input value: ");
Serial.println(x);
}
delay(4000);

}
Thanks again for your help. 
At least it compiles and displays. It may not do what I need anyway. I thought its function was to pass data to &x but it looks like it is the other way round.

olikraus, 2021-09-09:
The purpose of "userInterfaceInputValue" is to let the user enter a value. For this, the user needs to press some buttons and obvioulsy "userInterfaceInputValue" will wait for the user to press a button. So the code probably does not hang, but instead waits for some user input.

On the other side the button signals are not configured in "u8g2.begin();". So, yes, the code waits for digital inputs, which are not configured.

swb2020, 2021-09-09:
Awesome.... Yes, I was thinking the same thing - it is not going to let go till I accept the data. I made the mods in my main sketch which does use the encoder for input and it preforms as expected. For me the solution was the declaration type for the variables used.
Thanks for all your hard work developing this library. It allowed me to use my old 3d printer parts.
Going to close this as a success.......

closed 2021-09-09 by swb2020