olikraus/u8g2 #1583: Migration form u8g to u8g2 on JN51XX chip

pynting, 2021-09-09:
Hi,
Several years ago I made a working implementation of the u8g library for a JN5168 chip I was working on. Now I have taken up the development again, and in trying to speed up the device I thought it would be good to try and switch to u8g2, but I am having difficulties getting anything but noise or a blank screen...

This was my u8g i2c implementation (there were some delay functions as well, but I found out that it worked without as well, so removed them):
```
PUBLIC uint8_t u8g_com_hw_i2c_fn(u8g_t *u8g, uint8_t msg, uint8_t arg_val, void *arg_ptr)
{
  switch(msg)
  {
    case U8G_COM_MSG_STOP:
      //STOP THE DEVICE
      break;

    case U8G_COM_MSG_INIT:
      //INIT HARDWARE INTERFACES, TIMERS, GPIOS...
        vAHI_SiMasterConfigure( TRUE,  FALSE,  1); //bPulseSuppressionEnable, bInterruptEnable, 16M/((scale+1)*5) = 3.2M
      break;

    case U8G_COM_MSG_ADDRESS:
      //SWITCH FROM DATA TO COMMAND MODE (arg_val == 0 for command mode)
        if (arg_val == 0) 	control = 0;
        else 				control = 0x40;
     break;

    case U8G_COM_MSG_RESET:
      //TOGGLE THE RESET PIN ON THE DISPLAY BY THE VALUE IN arg_val
    	//no reset pin available
        //u8g_10MicroDelay();
      break;

    case U8G_COM_MSG_WRITE_BYTE:
      //WRITE BYTE TO DEVICE
 	   vAHI_SiMasterWriteData8(dispI2Caddr);
 	   bAHI_SiMasterSetCmdReg(TRUE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
 	   while(bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess

 	   vAHI_SiMasterWriteData8(control);
 	   bAHI_SiMasterSetCmdReg(FALSE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
 	   while(bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess

 	   vAHI_SiMasterWriteData8(arg_val);
 	   bAHI_SiMasterSetCmdReg(FALSE, TRUE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
 	   while(bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess
       //u8g_MicroDelay();
      break;

    case U8G_COM_MSG_WRITE_SEQ:
    case U8G_COM_MSG_WRITE_SEQ_P:
    {
      //WRITE A SEQUENCE OF BYTES TO THE DEVICE
    	register uint8 *ptr = arg_ptr;

 	   vAHI_SiMasterWriteData8(dispI2Caddr);
 	   bAHI_SiMasterSetCmdReg(TRUE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
 	   while(bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess

 	   vAHI_SiMasterWriteData8(control);
 	   bAHI_SiMasterSetCmdReg(FALSE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
 	   while(bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess

		while( arg_val > 0 )
		{
		  vAHI_SiMasterWriteData8(*ptr++);
		  bAHI_SiMasterSetCmdReg(FALSE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
		  arg_val--;
		  while(bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess
		}
		bAHI_SiMasterSetCmdReg(FALSE, TRUE, FALSE, FALSE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
    }
      break;
  }
  return 1;
}
```

but trying to make it work with u8g2 was not as easy as I had  hoped!

This is one of my many attempts at an implementation that yelds a blank screen:

```
PUBLIC uint8_t u8x8_JN51XX_gpio_and_delay(U8X8_UNUSED u8x8_t *u8x8,
										  U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int,
										  U8X8_UNUSED void *arg_ptr)
{
	switch (msg)
	{
	case U8X8_MSG_GPIO_AND_DELAY_INIT:
		delay_ms(2);
		break;
	case U8X8_MSG_DELAY_MILLI:
		delay_ms(arg_int);
		break;
	case U8X8_MSG_GPIO_DC:
		break;
	case U8X8_MSG_GPIO_RESET:
		break;
	}
	return 1;
}

PUBLIC uint8_t u8x8_byte_com_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
		register uint8 *ptr = arg_ptr;
	switch (msg)
	{
	case U8X8_MSG_BYTE_SEND:
		while (arg_int > 0)
		{
			vAHI_SiMasterWriteData8(*ptr++);
			bAHI_SiMasterSetCmdReg(FALSE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
			arg_int--;
			while (bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess
		}
		break;
	case U8X8_MSG_BYTE_INIT:
		vAHI_SiMasterConfigure(TRUE, FALSE, 2); //bPulseSuppressionEnable, bInterruptEnable, 16M/((scale+1)*5) = 3.2M
		break;
	case U8X8_MSG_BYTE_SET_DC:
		break;
	case U8X8_MSG_BYTE_START_TRANSFER:
		vAHI_SiMasterWriteData8(dispI2Caddr);
		bAHI_SiMasterSetCmdReg(TRUE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
		while (bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess
		break;
	case U8X8_MSG_BYTE_END_TRANSFER:
		bAHI_SiMasterSetCmdReg(FALSE, TRUE, FALSE, FALSE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
		break;
	default:
		return 0;
	}
	return 1;
}
```
This is my initiation and test code:
```
	u8g2_Setup_ssd1306_i2c_128x64_noname_1(&u8g2, U8G2_R0, u8x8_byte_com_hw_i2c, u8x8_JN51XX_gpio_and_delay);
	u8g2_SetI2CAddress(&u8g2, dispI2Caddr);
	u8g2_InitDisplay(&u8g2); //Send initialization code to the display
	u8g2_SetPowerSave(&u8g2, 0);
	u8g2_SetFontDirection(&u8g2, 0);
	u8g2_SetFont(&u8g2, u8g_font_6x13r);
	u8g2_DrawStr(&u8g2, 0, 22, "U");
	delay_ms(3000);
	u8g2_DrawBox(&u8g2, 8, 8, 10, 10);
	delay_ms(3000);
```

I also tried to make it work with approach 2 from here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication
but that created noise on screen, maybe better but could not get it to work either.

as the Jennic (now NXP) chip might be unfamiliar, I have included the truth table for bAHI_SiMasterSetCmdReg from the API documentation:
```
Start     Stop     Read    Write
Resulting Instruction to SI Bus
0             0             0              0              No active command (idle)
1              0             0              1              Start followed by Write
1              1              0              1              Start followed by Write followed by Stop
0              1              1              0              Read followed by Stop
0              1              0              1              Write followed by Stop
0              0              0              1              Write only
0              0              1              0              Read only
0              1              0              0              Stop only
```

for further info page 323 and onward: https://www.nxp.com/docs/en/user-guide/JN-UG-3087.pdf

Many thanks for any pointers on how to make this work!
PS: I have been out of mcu programming for many years, and even then I was just a hobbyist, so an high concept explanation might go over my head!

olikraus, 2021-09-09:
I think seeing noise is a good sign. It means, that at least some data was transfered correctly. I personally would suggest to use the second example of the referenced code tempate:
``` C++
uint8_t u8x8_byte_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  static uint8_t buffer[32];		/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
  static uint8_t buf_idx;
  uint8_t *data;
 
  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;      
      while( arg_int > 0 )
      {
	buffer[buf_idx++] = *data;
	data++;
	arg_int--;
      }      
      break;
    case U8X8_MSG_BYTE_INIT:
      /* add your custom code to init i2c subsystem */
      break;
    case U8X8_MSG_BYTE_SET_DC:
      /* ignored for i2c */
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      buf_idx = 0;
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      i2c_transfer(u8x8_GetI2CAddress(u8x8) >> 1, buf_idx, buffer);
      break;
    default:
      return 0;
  }
  return 1;
}
```

So basically you sould transfer the buffer content only once, where the i2c_transfer call is done. In fact it might be a good idea to develop a procedure which works similar to the i2c_transfer procedure: I2C address and multiple bytes as input, which are simply written to the the target device (display).

With a little help from google, the expected function i2c_transfer seems to be implemented here (`suli_i2c_write`):
https://github.com/Seeed-Studio/Mesh_Bee/blob/7b5f3287c5100af9a54e83cad29060bee06fdd5d/suli/suli.c#L350-L377

So you could just take the template from above and call the i2c transfer function (`suli_i2c_write` from the above link) and it should work. Maybe some init code is required for your uC which needs to be placed after `U8X8_MSG_BYTE_INIT`.

olikraus, 2021-09-09:
The complete code could look like this (suli_i2c_write seems to be available under 2-clause BSD license):

``` C++
// suli_i2c_write from https://github.com/Seeed-Studio/Mesh_Bee/blob/master/suli/suli.c (2-clause-BSD)
uint8 suli_i2c_write(void * i2c_device, uint8 dev_addr, uint8 *data, uint8 len)
{
    vAHI_SiMasterWriteSlaveAddr(dev_addr, FALSE);
    // bSetSTA,  bSetSTO,  bSetRD,  bSetWR,  bSetAckCtrl,  bSetIACK);
    bAHI_SiMasterSetCmdReg(TRUE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
    while(bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess

    int i;
    uint8 *old = data;
    for(i = 0; i < len; i++)
    {
        vAHI_SiMasterWriteData8(*data++);
        if(i == (len - 1))  //should send stop
        {
            bAHI_SiMasterSetCmdReg(FALSE, TRUE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
        } else
        {
            bAHI_SiMasterSetCmdReg(FALSE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
        }
        while(bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess
        if(bAHI_SiMasterCheckRxNack())
        {
            bAHI_SiMasterSetCmdReg(FALSE, TRUE, FALSE, FALSE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
            break;
        }
    }
    return data - old;
}

uint8_t u8x8_byte_i2c_jn51xx(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  static uint8_t buffer[32];		/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
  static uint8_t buf_idx;
  uint8_t *data;
 
  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;      
      while( arg_int > 0 )
      {
	buffer[buf_idx++] = *data;
	data++;
	arg_int--;
      }      
      break;
    case U8X8_MSG_BYTE_INIT:
      vAHI_SiMasterConfigure(TRUE, FALSE, 2); //bPulseSuppressionEnable, bInterruptEnable, 16M/((scale+1)*5) = 3.2M
      break;
    case U8X8_MSG_BYTE_SET_DC:
      /* ignored for i2c */
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      buf_idx = 0;
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      suli_i2c_write(NULL, u8x8_GetI2CAddress(u8x8) >> 1, buffer, buf_idx);
      break;
    default:
      return 0;
  }
  return 1;
}
```

pynting, 2021-09-09:
oh, man you are quick to reply, almost to fast as I just solved it (after two days of experimentation) by going back to the other solution, looking at it with new eyes. I discovered one possible mistake I made, and that was that I had used someone else's implementation, and they had changed the buffer to 34, not 32. Also, I had not properly understood what the u8g2 lib did and did not do, like switching between sending to display and setup registers on the display driver... basically, that I do not need to think about that part!
The last problem I suspect could have made things worse was the fact that one can basically crash the display by sending unexpected messages to it, needing to cycle power to restore function. I might have got it working more than once not knowing it!

Here is my working implementation (but I will perhaps switch over to your suggestion after looking at it some more!):
```
uint8_t u8x8_byte_com_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	static uint8_t buffer[32]; /* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
	static uint8_t buf_idx;
	uint8_t *data;
	uint8_t i = 0;

	switch (msg)
	{
	case U8X8_MSG_BYTE_SEND:
		data = (uint8_t *)arg_ptr;
		while (arg_int > 0)
		{
			buffer[buf_idx++] = *data;
			data++;
			arg_int--;
		}
		break;
	case U8X8_MSG_BYTE_INIT:
		/* add your custom code to init i2c subsystem */
		vAHI_SiMasterConfigure(TRUE, FALSE, 1); //bPulseSuppressionEnable, bInterruptEnable, 16M/((scale+1)*5) = 3.2M
		break;
	case U8X8_MSG_BYTE_START_TRANSFER:
		buf_idx = 0;
		break;
	case U8X8_MSG_BYTE_END_TRANSFER:

		vAHI_SiMasterWriteData8(dispI2Caddr);
		bAHI_SiMasterSetCmdReg(TRUE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
		while (bAHI_SiMasterPollTransferInProgress()); //Waitforanindicationofsuccess

		while (buf_idx > 0)
		{
			vAHI_SiMasterWriteData8(buffer[i++]);
			bAHI_SiMasterSetCmdReg(FALSE, FALSE, FALSE, TRUE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
			buf_idx--;
			while (bAHI_SiMasterPollTransferInProgress())
				; //Waitforanindicationofsuccess
		}
		bAHI_SiMasterSetCmdReg(FALSE, TRUE, FALSE, FALSE, E_AHI_SI_SEND_ACK, E_AHI_SI_NO_IRQ_ACK);
		break;
	default:
		return 0;
	}
	return 1;
}
```
It is already a lot faster than the u8g lib.
Now I am going to try and see if I can get around the `while (bAHI_SiMasterPollTransferInProgress())`... Not Ideal when working with a transmitter!

Thank you Oli, for the library and support.

closed 2021-09-09 by pynting

olikraus, 2021-09-09:
> Here is my working implementation (but I will perhaps switch over to your suggestion after looking at it some more!):

I think your solution is almost identical to my suggestion :-)