olikraus/u8g2 #1590: Missing monospace font heights: 32, 64pxl

mrx23dot, 2021-09-14:
Common font heights (32 and 64 pixels) are missing from:
https://github.com/olikraus/u8g2/wiki/fntlistmono

e.g. on a 128x32 screen I want to use full height, not to waste pixels.

olikraus, 2021-09-14:
For which font do you request the target size?
Which characters would be required?

mrx23dot, 2021-09-14:
In fntlistmono we have 30 and 33 pixel, and 63 pixel, but not 32 and 32.

https://github.com/olikraus/u8g2/wiki/fntlistmono

olikraus, 2021-09-15:
```
/*
  Fontname: -FreeType-Inconsolata LGC-Medium-R-Normal--86-860-72-72-P-387-ISO10646-1
  Copyright: Original Roman version created by Raph Levien using his own tools and FontForge. Copyright 2006 Raph Levien. Hellenisation of the Roman font, by Dimosthenis Kaponis, using FontForge. Hellenic glyphs Copyright 2010-2012 Dimosthenis Kaponis. Released under the SIL Open Font License, http://scripts.sil.org/OFL.    Cyrillic glyphs added by MihailJP, using FontForge. Cyrillic glyphs Copyright 2012 MihailJP. Released under the SIL Open Font License, http://scripts.sil.org/OFL.    Some glyphs modified by Greg Omelaenko, using FontForge.
  Glyphs: 16/658
  BBX Build Mode: 2
*/
const uint8_t u8g2_font_inr64_m[1730] U8G2_FONT_SECTION("u8g2_font_inr64_m") = 
  "\20\2\6\5\6\7\1\6\7\64T\5\357@\356C\357\0\0\0\0\6\245*q\64\365C\377\301\377\271"
  "\10\213b\217\366p\315\363\202\307\360\10\306`\206\307PLM\202\24\11\322,Fq,\213C\24E\263"
  "@\3\61<\316\23PS\247]!t*+\262*)s\32\201a\36\202A\36\204\61\32\305\21\36\306"
  "\361\35\310\321\35\311\301!\312\241!\13\202!\15r!\16R!OB\35\321a\31R\201\21\23\241\11"
  "\225\360\203\377\377_\5\0+*\64\365C\377\301\377\275\206k\376\377\270\17P\226\17P\226\17P\226\17"
  "P\226\17P\226\17P:\303\65\377\377\375\340\377\377K\1,\60\64\365C\377\301\377\377\377\377\377/+"
  "\334\203M\324%eP\266tQ\225\65a\221\366h\15\327\324\32\256\321\32n\341\32\255\231k\274D,"
  "x\0-$\64\365C\377\301\377\377\377\256\17(\240\17(\240\17(\240\17(\240\17(\240\17(~\360"
  "\377\377\377\377\42\0.!\64\365C\377\301\377\377\377\377\377W\22.\262*\252\222\62([\252\250\312\212"
  "\270\304\17\376\237\11\0/u\64\365C\67\203y\200\15\367h\17\366h\17\366h\17\366h\17\26a\17"
  "\26a\17\26a\217\366`\217\366`\217\366`\21\366`\21\366`\21\366`\21\366h\17\366h\17\366h"
  "\17\26a\17\26a\17\26a\217\366`\217\366`\217\366`\217\366`\21\366`\21\366`\21\366h\17\366"
  "h\17\366h\17\26a\17\26a\17\26a\17\26a\217\366h\15\230\220\203\37\374\317\2\0\60\332\64\365"
  "C\377\301TceLh\224B\352\263:eX\66$I\62dI\260\243A\256\343A,#:\252#"
  "B(c:&\203B$cJ\242cR cZ Cb\240Bj\36Cr\34#B\2\303\61\240"
  "$\60\34\343I\4\243\61 d\60\32\343A\210\203\61\34\304\60\30\243I\14\203\61\32\344\60\30\203A"
  "\20\203\61\26$\61\30CA\24\203\61\24D\61\30#A\26\203\61\20\204\61\35Hc\60\206\201\70\6"
  "c\30\307c\60\7\201<\6s\14\10d\64\206\200D\205c\10Gd\70FpH\206\343H\206\323L"
  "\306\263P\306\243P\6\224TF\204PGtT\206tT\306tL\6\205DGuD\306u<\307\205"
  "\64G\226$\211\246\24\12wy\25\70\211\220\331\244\310\17\376\337\10\0\61*\64\365C\377\301X\203E"
  "T%]\316\246|\306(\234\276\352[\206\3<\214\3$\222\243=\377\377\377\377\377\377\373\301\377w\1"
  "\62d\64\365C\377\301\320\4}\306\351\273\270L\333\260\305`\254dY\250\304I\250#J&cB\250"
  "\202B\250\241B\254\300BX\264\366\214E\261\7\213Z\21\26Y\221\65Y\321\326DM\227\252S\221\25"
  "\335\232\254h\354\211E\330S{\260\307\26D\37@@\37P@\37P@\37P@\37P@\37P\374"
  "\340\377S\0\63n\64\365C\377\301\316\4}\306\351\263\272L\333\260\305`.\204I\60\303I\260!B"
  "\262`B\332\243E\332\363\261\10{\264\7\213\254\250\65I\25\263\11\243\361)!R\22\247\224Y\25\66"
  "a\221\26i\17\367hO\356\321\236\217E\330\303\12*\244\22*\204\42&\244*\242\204:\236\204R\226"
  "\205j\212\306\352\260m\303\272J\204N\345\7\377\217\4\0\64\204\64\365C\377\301f\243=\261\310\232R"
  "U\351\202\262\316\306\64\202\303\64\202\243\64\204\203\64\206\203\64\206c\64\210C\64\212C\64\212#\64\214"
  "\3<\214\3\64\216\343\63\220\343\63\220\303\63\222\243\63\224\243\63\224\203\63\226c\63\230c\63\230C\63"
  "\232#;\232#\63\234\3\63\236\343:\236\343\62\240\303\62\242\243:\242\243\62\244\203\372\200\204\371\200\204"
  "\371\200\204\371\200\204\371\200\204\371\200\244<\332\363\377\367\203\377\17\2\0\65k\64\365C\377\301\274\17\10"
  "\250\17\10\250\17\10\250\17\10\250\17\10\250\17\10(\303\65k\317\377#\26\357\10\242M\313\272\313\263\276"
  "\252\71\30Ja\226\11\201\222\351\220\222\251\240\220J\250\22\26\325\42,\252=\177,\332$X\310\64X"
  "HTT\11tP\10\224HI\244\70\312\264,J\305\30\215\325a\233\206yU\30\235\311\17\376\337\10"
  "\0\66\245\64\365C\377A\66SF\42\345]\234\226m\330\202 \230\302\24X\362\14\31\22\5\31\302\42"
  "+\302\36,\302\36\355\301\242\330\243=\261\10\202p\210\321d\10\21]\310\60Y\307PU\207pQG"
  "\220\24\313\344(\311\324\60\211\304\70I\264@H\244HG\224L\10\204T\7\204T\7tX\310s\134"
  "\307s\134\307s\134\307s\134\307\203X\307\203X\7tX\7tX\7tX\7\204T\7\204P\207t"
  "P\207\204L\207\204\232\220\350\240\222\7\251\20\7\261\224%\271\224$\311\226B\331\62\356\362(\61\42\235"
  "\24\371\301\377w\1\67g\64\365C\377\301\270\17 \242\17 \242\17 \242\17 \242\17 \242\17\30X"
  "\204=\332\203=\332\203E\330\203E\330\243=X\204=X\204=\332\203E\330\203E\330\243=X\204="
  "\332\203E\330\203E\330\243=X\204=\332\203E\330\203E\261\7\213b\17\26\305\36,\212=X\24{"
  "\260(\366`Q\354\301\242~\360\377\243\0\70\252\64\365C\377A\63SF\42\345]\134\246mY\202,"
  "W\262$\327\341$\25\362 \325\1!\24\22!\323!!\323!!\323y\22\62\35\22\62!\321Q\35"
  "\321Q!\17R%\316q!\15r)\12\222)\10\262)\6\322-\3\362Q\241D:%\64N\237\325"
  "%\3\263%\207\222!\212r!MR!OB\35Q\42\241\244\203B\240\243B\240\303:\36\304B\34"
  "\304B\234\343B\234\3;\234\3;\34\304B\34\304B\34\304B\36\244: \204B\240DJ\242\4J"
  "$\305Q&\205Q*\306`..\323\66\254\253\302\310\134~\360\377]\0\71\243\64\365C\377A\362r"
  "B\342\4Z\35\266i\231\202$Y\242$\27\342 \25\362 \25\2!\324!\35\23\42!\322A\35\322"
  "A\35\21R\35\20R\35\320a\35\320a\35\320a\35\320a\35\320q\35\17b\35\17b\35\17b\35"
[... cut ...]
  "\201a\311%w\134\312\222K,\271\344\216KYr)K.\271\343Rw\134\312\222KYr\311\207\337"
  "\3\70L;\15\373\207\204\231\352\65GJa\344\20\66\22Q$\21E\22Q$\221E\20Y\344\220F"
  "\314pF\224\247\346\231\352\21R\32\61e\221D\324P$\221E\20Y\4\221E\20Y\4\221EPI"
  "D\225SV)\245\265\267\344\371\360S\0\71D;\15\373\207\204\231\12\62WHa\345\220E\20QT"
  "\42\212$\262F\42k$\262F\42\213 \262H\42\251$\202\214*\4\255\25\10K\202\270B\310\35x"
  "\134\232%\227\60\201\10#\245\264\366\226<\37~\13\0\0\0\0\4\377\377\0";
```

```

mrx23dot, 2021-09-15:
Thank you I will give it a go, I would recommend checking them in for others.
Cheers!

closed 2021-09-15 by mrx23dot