olikraus/u8g2 #1592: #include of mui.h must be in " " instead of <> to look for the file in current directory

jp112sdl, 2021-09-20:
I think it has to be `#include "mui.h"` or am I wrong?

https://github.com/olikraus/u8g2/blob/d16b09fe20d3b4f35dc95039de260a142503ecb1/csrc/mui_u8g2.h#L41

With the current version I'm getting a compiler error:
```
.../U8g2_Arduino/src/clib/mui_u8g2.h:41:10: fatal error: mui.h: No such file or directory
 #include <mui.h>
          ^~~~~~~
```

olikraus, 2021-09-20:
hmmm... Where does this error appear? Is this within Arduino IDE?

jp112sdl, 2021-09-20:
> Where does this error appear? Is this within Arduino IDE?

It is Arduino CLI used by Github Actions.

[image]

[image]

olikraus, 2021-09-20:
Ah, ok, thanks. It's fixed in the repo now.

closed 2021-09-20 by olikraus

jp112sdl, 2021-09-20:
Thanks for the quick reply!

olikraus, 2021-09-20:
Thanks for the hint.

jp112sdl, 2021-09-20:
Please change it in the `U8g2_Arduino`, too:
https://github.com/olikraus/U8g2_Arduino/blob/c232065db7035f4a0419a82d0e938d41be829ab0/src/clib/mui_u8g2.h#L41

olikraus, 2021-09-20:
sure, U8g2_Arduino is just a mirror

olikraus, 2021-09-20:
I tested everything with Arduino 1.8.10 and also created a new release for U8g2: 2.31.2

aschwtzr, 2021-11-19:
I'm still getting this error in 2.31.2

```
Warning: Board Intel:i686:izmir_ec doesn't define a 'build.board' preference. Auto-set to: I686_IZMIR_EC
In file included from C:\Users\aaa\Documents\Arduino\libraries\U8g2\src\clib\mui_u8g2.c:97:0:
C:\Users\aaa\Documents\Arduino\libraries\U8g2\src\clib\mui_u8g2.h:41:10: fatal error: mui.h: No such file or directory
 #include <mui.h>
          ^~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino MKR WiFi 1010.
```