olikraus/u8g2 #1593: Add st7920 256x32 display support

olikraus, 2021-09-20:
https://github.com/olikraus/u8g2/pull/1545

olikraus, 2021-09-27:
@shlyakpavel I have created beta release 2.32.2.

Maybe you can check the 256x32 constructor once...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

shlyakpavel, 2021-09-29:
> You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Fine. I'll do it ASAP (most likely today)
Thanks a lot for your work!

shlyakpavel, 2021-09-29:
Test done, everything works as expected
Thanks a lot!

olikraus, 2021-09-29:
Thanks for the feedback

olikraus, 2021-10-04:
pr closed, so i will also close this issue.

closed 2021-10-04 by olikraus