olikraus/u8g2 #1595: RA8835 320x240 porting to PIC24F

anshuman-ag, 2021-09-23:
Hi,
I am trying to port U8g2 to PIC24FJ256GB206 MCU, I have my board ready, I hope I am able to intilialize the LCD well but my code is not giving desired draw results.

My init code is as below, I have kept 16bit mode enabled and dynamic buffer enabled

unsigned char disp_buff[9600];
u8g2_Setup_ra8835_320x240_f(&u8g2, U8G2_R0, u8x8_byte_8bit_8080mode,u8x8_gpio_and_delay_template);
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2,0);
u8g2_SetContrast(&u8g2,200);
u8g2_SetBufferPtr(&u8g2, disp_buff);

drawing code as below

  u8g2_ClearBuffer(&u8g2);
    u8g2_SetFont(&u8g2,u8g2_font_9x18_tf);
    u8g2_SetFontRefHeightText(&u8g2);
    u8g2_SetFontPosTop(&u8g2);
    u8g2_DrawStr(&u8g2,0,0,"Hello From Anshuman!");
     u8g2_DrawLine(&u8g2,10,10,100,100);
    u8g2_SendBuffer(&u8g2);

display looks as below

please help
[image]

olikraus, 2021-09-23:
Looks like DC line is not working.

anshuman-ag, 2021-09-24:
Hi,

I do not feel that it is a issue with the Dc/A0 line, as i have tried to run this LCD with vendor provided init code and results are satisfactory.

I am hereby attaching gpio_delay_template code, along with schematic snapshot, and LCD vendor code 

uint8_t u8x8_gpio_and_delay_template(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    int i;
  switch(msg)
  {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
        LCD_RD = 1;
        __delay_ms(500);
      break;							// can be used to setup pins
    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
        for(i=0;i<arg_int;i++)Nop();
      break;    
    case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
        
        for(i=0;i<arg_int;i++)
        { 
               
        Nop();Nop();Nop();Nop();Nop();
        Nop();Nop();Nop();Nop();Nop();
        }
       
      break;
    case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
        for(i=0;i<arg_int;i++){__delay_us(10);}
      break;
    case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
        for(i=0;i<arg_int;i++){__delay_ms(1);}
      break;
    case U8X8_MSG_DELAY_I2C:				// arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
      break;							// arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us
    case U8X8_MSG_GPIO_D0:				// D0 or SPI clock pin: Output level in arg_int
    //case U8X8_MSG_GPIO_SPI_CLOCK:
        if(arg_int){LCD_D0 = 1;}
        else {LCD_D0 = 0;}
      break;
    case U8X8_MSG_GPIO_D1:				// D1 or SPI data pin: Output level in arg_int
    //case U8X8_MSG_GPIO_SPI_DATA:
        if(arg_int){LCD_D1 = 1;}
        else {LCD_D1 = 0;}
      break;
    case U8X8_MSG_GPIO_D2:				// D2 pin: Output level in arg_int
        if(arg_int){LCD_D2 = 1;}
        else {LCD_D2 = 0;}
      break;
    case U8X8_MSG_GPIO_D3:				// D3 pin: Output level in arg_int
        if(arg_int){LCD_D3 = 1;}
        else {LCD_D3 = 0;}
      break;
    case U8X8_MSG_GPIO_D4:				// D4 pin: Output level in arg_int
        if(arg_int){LCD_D4 = 1;}
        else {LCD_D4 = 0;}
      break;
    case U8X8_MSG_GPIO_D5:				// D5 pin: Output level in arg_int
        if(arg_int){LCD_D5 = 1;}
        else {LCD_D5 = 0;}
      break;
    case U8X8_MSG_GPIO_D6:				// D6 pin: Output level in arg_int
       if(arg_int){LCD_D6 = 1;}
        else {LCD_D6 = 0;}
      break;
    case U8X8_MSG_GPIO_D7:				// D7 pin: Output level in arg_int
        if(arg_int){LCD_D7 = 1;}
        else {LCD_D7 = 0;}
      break;
[... 633 lines cut ...]
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00 , 0x00,
};

olikraus, 2021-09-24:
It is long time back, that I implemented RA8835 support.
See here: https://github.com/olikraus/u8g2/wiki/gallery#24-sep-2016-ra8835-240x128-nhd-display

According to this, WR needs to be connected to GND permanently and U8g2 E should be connected to RD.

anshuman-ag, 2021-09-24:
Hi, I just checked the data-sheet RD is useful as enable when using 6800 mode while WR works as EN in 8080 mode. Moreover in the example code we could clearly see WR used as enable in active inverting input, I mean En is active while low, Attached is the data-sheet for LCD.
[WG320240C0-TMIVZ#220(GLCD 320x240 RA8835).pdf](https://github.com/olikraus/u8g2/files/7225310/WG320240C0-TMIVZ.220.GLCD.320x240.RA8835.pdf)

olikraus, 2021-09-24:
Hmm yes, you are correct. Maybe I have used 6800 mode for my own tests.

closed 2022-03-04 by olikraus