olikraus/u8g2 #1598: add SH1107 128x80 screen support.

Eddddddddy, 2021-09-26:
I bought a screen with 128x80 pixels. I try to use 128x64 code to drive it. But it can't fill the whole screen. Can I add a new screen configuration? and how.

olikraus, 2021-09-26:
Any datasheet/link of the display?
Can you try one of the 128x128 constructors (https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1107-128x128)? 
 * U8G2_SH1107_128X128_1_4W_HW_SPI
 * U8G2_SH1107_PIMORONI_128X128_1_4W_HW_SPI
 * U8G2_SH1107_SEEED_128X128_1_4W_HW_SPI
Which constructor will look best? Can you sent pictures?

Eddddddddy, 2021-09-26:
I started to try it immediately.

olikraus, 2021-09-26:
Is this the display: https://www.buydisplay.com/white-0-78-inch-oled-display-panel-128x80-4-wire-spi-interface-sh1107 ?

Eddddddddy, 2021-09-26:
Yes. By the way, I prefer to use iic.

Eddddddddy, 2021-09-26:
`U8G2_SH1107_128X128_1_HW_I2C u8g2(U8G2_R3, /* reset=*/ U8X8_PIN_NONE);`
```
u8g2.drawPixel(0, 24);
u8g2.drawPixel(127, 24);
u8g2.drawPixel(0, 103);
u8g2.drawPixel(127, 103);
```

This one suits me. I found the border of the screen.
Wonderful!

Eddddddddy, 2021-09-26:
[image]

olikraus, 2021-09-26:
Oh, nice. If I would create a `U8G2_SH1107_128X80_1_HW_I2C` constructor, would you be able to test the same?

Eddddddddy, 2021-09-26:
Of course.

olikraus, 2021-09-26:
I have create a beta release 2.31.3

Can you test the FlipMode example? It should always show a solid border around the display.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Eddddddddy, 2021-09-26:
[image]
[image]
[image]
[image]
It is not correct. Swap width and height?

olikraus, 2021-09-26:
> It is not correct. Swap width and height? 

In your first "Hello World" picture: Did you apply any rotation? I mean, what had been the exact constructor line? U8G2_R0?

olikraus, 2021-09-26:
I am a little bit list now. It looks more complicated.
I have another request: How would the hello word look with the SEEED constructor?
U8G2_SH1107_SEEED_128X128_1_HW_I2C

Eddddddddy, 2021-09-26:
> > It is not correct. Swap width and height?
> 
> In your first "Hello World" picture: Did you apply any rotation? I mean, what had been the exact constructor line? U8G2_R0?

U8G2_SH1107_128X128_1_HW_I2C u8g2(U8G2_R3, /* reset=*/ U8X8_PIN_NONE);

U8G2_R3

This line.

Eddddddddy, 2021-09-26:
wait. 

when I use the test library, 'hello world' code got a different result.

I am doing more test.

Eddddddddy, 2021-09-26:
```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SH1107_SEEED_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(30, 50, "Hello World!");
    u8g2.drawPixel(24, 0);       //Upper left
    u8g2.drawPixel(103, 0);     //Upper right
    u8g2.drawPixel(24, 127);   //Lower left
    u8g2.drawPixel(103, 127); //Lower right
  } while ( u8g2.nextPage() );
}
```
[image]

This is all the code.
I'm using the beta release 2.31.3.

This might help.

olikraus, 2021-09-26:
Ah, great, thanks for all the feedback. This helps a lot.
I will create another beta tomorrow.

Eddddddddy, 2021-09-26:
Waiting for your good news~

olikraus, 2021-09-27:
ok, I have created beta 2.32.1.
Can you check the hello world example with U8G2_R0 again? The x offset should be zero now (hopefully)
Thanks

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Eddddddddy, 2021-09-27:
```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SH1107_128X80_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2.begin();
  u8g2.setFont(u8g2_font_6x10_tr);
}

void draw(const char *s)
{
  u8g2.firstPage();
  do {
    u8g2.drawStr(2,15,"Hello World!");    
    u8g2.drawStr(2,30,s);    
    u8g2.drawFrame(0,0,u8g2.getDisplayWidth(),u8g2.getDisplayHeight() );

    u8g2.drawStr(2, 45, u8x8_u16toa(u8g2.getDisplayWidth(), 5));
    u8g2.drawStr(2, 60, u8x8_u16toa(u8g2.getDisplayHeight(), 5));
    
  } while ( u8g2.nextPage() );
  delay(2000);
}

void loop(void) {

  u8g2.clearDisplay();
  u8g2.setDisplayRotation(U8G2_R0);
  u8g2.setFlipMode(0);
  draw("R0,F0");
  u8g2.clearDisplay();
  u8g2.setFlipMode(1);
  draw("R0,F1");

  u8g2.clearDisplay();
  u8g2.setDisplayRotation(U8G2_R1);
  u8g2.setFlipMode(0);
  draw("R1,F0");
  u8g2.clearDisplay();
  u8g2.setFlipMode(1);
  draw("R1,F1");

  u8g2.clearDisplay();
  u8g2.setDisplayRotation(U8G2_R2);
  u8g2.setFlipMode(0);
  draw("R2,F0");
  u8g2.clearDisplay();
  u8g2.setFlipMode(1);
  draw("R2,F1");

  u8g2.clearDisplay();
  u8g2.setDisplayRotation(U8G2_R3);
  u8g2.setFlipMode(0);
  draw("R3,F0");
  u8g2.clearDisplay();
  u8g2.setFlipMode(1);
  draw("R3,F1");

}
```
[image]
[image]
[image]
[image]

Fantastic!

olikraus, 2021-09-27:
:-) looks good
Thanks for feedback and testing

Eddddddddy, 2021-09-27:
My pleasure. Close it for success

closed 2021-09-27 by Eddddddddy