olikraus/u8g2 #1599: stm32 Blue Pill -st7920

madhavanatmel, 2021-09-26:
bellow constructor 
U8G2_ST7920_128X64_F_8080 u8g2(U8G2_R0, PA0, PA1, PA2, PA3, PA4, PA5, PA6, PA7, /*enable=*/ PB0 /* A4 */, /*cs=*/ U8X8_PIN_NONE, /*dc/rs=*/ PB4 /* A3 */, /*reset=*/ PB1 /* A1 */);

Board : stm32 Blue Pill / Arduino Platform-Stm32 Duino Core .

Display not perfect 

[image]

olikraus, 2021-09-26:
maybe some wiring problems. Wires too long or instable. Did you solder the wires?

madhavanatmel, 2021-09-27:
it's soldered and it's working properly with Arduino UNO in Parallel Mode,SW SPI & HW SPI.

in Blue Pill display properly working with SW SPI & HW SPI but it's not properly working with 8 bit parallel mode .
do you think due to speed display behaving like this .

Actually i need blue pill SPI port is other purpose that what i try to use parallel mode .

olikraus, 2021-09-27:
Speed could be one topic (although u8g2 will slow down speed accordingly). Another issue might be the signal level. The ST7920 expects 5V logic levels while the STM32 only provides 3.3V logic levels. This might be too less for the display.

shlyakpavel, 2021-11-15:
>  ST7920 expects 5V logic levels while the STM32 only provides 3.3V logic levels. This might be too less for the display.
[image]

olikraus, 2021-11-15:
so your ST7920 board operates with 3.3V?

madhavanatmel, 2021-11-18:
Yes, you are correct . In SPI mode I have tested the display with 3.3v
operating voltage but the display contract was very low  even though I
couldn't reconigend anything on display and I have adjusted the contrast
potionmeter  nothing appears . I have tested display operating voltage as
5v Display VCC and stm32 is operated on 3.3v logic voltage output but this
combination is working with SPI Mode . The same combination is not good in
parallel mode display connection . I think the display data terminal
(d0-d7) needs to be exactly the same as vcc at the same time SPI & I2C
(more or less they shift register concept ) . They possibly work with 3.3v
logic inputs .

olikraus, 2021-11-18:
Hmm not sure how I can support here...

madhavanatmel, 2021-11-21:
no issue.... Now  i have tested SW SPI Contructor is working and Hardware
SPI port using for SD card  also working

madhavanatmel, 2021-11-21:
but I am
 not sure  this combination is correct . i have tested with load from sd
card examples file

closed 2022-03-04 by olikraus