olikraus/u8g2 #1600: I don't understand font preview images

Aezarien, 2021-09-28:
If I may ask a question (Is it okay to ask a question here or is there a better place?):
To the left of each row of characters there is a number that looks something like 48/0030 (for the u8g2_font_battery19_tn font). I can tell it has something to do with how the characters are mapped, but I have exhausted every keyword I can think of trying to figure out how it works. 0030 seems to correlate with zero on the keyboard, per windows character map. What does the 48 stand for? Is there somewhere that explains this I just missed? Are there keywords I can google to get a clearer picture of what is going on here? I'm a tad bit confused. Thank you!

olikraus, 2021-09-28:
Thanks for the question (any place within this project is fine for questions). 

Before discussing the fonts, please recall that each character has a number. In ancient times there had been ASCII codes, now we usually talk about Unicode (ASCII now is a subset of Unicode).
In ASCII the number of the blank (space) character is 32 in decimal notion (which is 0x20 in hexadecimal notation).
Another example is the uppercase "A", which has the decimal code 65 (0x41 hex). With this knowledge, let us look at one of the font preview pictures:
[image]

In the first column of the first row, you just see the character code in decimal and hexadecimal notion of the leftmost character (which is the blank char in this case). In each row you will find exactly 16 chars (0x10 hex) in increasing order. So the "!" is the next char after the blank and consequently has the ASCII / Unicode number 33 (or 0x21 hex).

To make the calculation easier, each row is prefixed with the code of the first character of that row. This means the "0" has the code 48 (decimal).

Let's look at the uppercase "A" it has the code 65. So it is in the second position after the "@" (which has the code 64).

In most cases these code numbers are of little interest. If you want an uppercase "A", then you can just type an uppercase "A". Of course, things will become more complicated of there are some random pictures in those fonts, so let us look at the battery symbols.

[image]
The leftmost icon (empty battery) is next to 48/0030, so the empty battery has the code 48, which is the character "0".
As a consequence, if you type "0" in U8g2 will draw the empty battery symbol instead. 
Of course you probably want to draw only a single char, so in this case use [u8g2.drawGylph](https://github.com/olikraus/u8g2/wiki/u8g2reference#drawglyph) with 48 as an argument. This will draw the empty battery symbol at the desired location.

If you need the number of the charging batter icon, just count from left until you reach the target icon. Start counting with zero, then the charging icon is at position  6.
Now add 6 to the left decimal number (which is 48) to get the number 54 for the charging icon. Again this can be passed to u8g2.drawGlyph.

The hex number, which is also provided, can be used if you still want to use the string drawing functions.
As we discussed earlier, you could use "0" to draw the empty battery icon. But writing "0" in your code is a little bit obscure if you want to draw the battery icon, because you need to know that "0" has the code 48, which is used for the empty battery icon.
Unfortunately we can not write the decimal 48 into the string to get the icon at position 48, however this can be done with the hex code: The string "\x30" means: Use the character with hex code 30 (https://en.wikipedia.org/wiki/Escape_sequences_in_C#Table_of_escape_sequences). This again would draw empty battery symbol.

Conclusion:
 * d/xxxx in the font pictures provide the decimal and hexadecimal unicode number of the leftmost glyph in the same row.
 * The numbers of the other glyphs in the same row can be derived by counting from left to right
 * Use the decimal number together with u8g2.drawGlyph()
 * Use the hexadecimal number inside strings (\x escape code)

Test:
For the following font:
[image]

Which number do you need to pass to u8g2.drawGlyph() if you want to draw the "pull a rabbit out of a hat" icon?

Aezarien, 2021-09-29:
Oh Gosh. Thank you for such an extensive answer. I managed to get it to work, with my limited understanding, but this is sooooo much easier and transparent than what I was doing. 

I assume if the first icon is 64 and the rabbit  is the third symbol , I would use 66 with drawGlyph();? And "\x43" in drawStr();?

olikraus, 2021-09-29:
Yes, 66 for drawGlyph, (and \x42 for the string ;-)

Aezarien, 2021-09-29:
Oh, right. And you probably know exactly what I did - skipped that zero. 

Thank you, again, for such a detailed answer. It connected all the right dots and answered some other questions I had.

closed 2022-03-04 by olikraus