olikraus/u8g2 #1602: simultaneously  R2 and MIRROR

efstathios82, 2021-10-01:
Hi,
I use the oled screen -> SparkFun Transparent Graphical OLED Breakout .LCD-15173.
and i used well the library ->
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R2, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ U8X8_PIN_NONE);

can i keep R2 (180 degree clockwise rotation) and do  mirroring in R2 ;

when changed to MIRROR lost the R2 rotation!

thank you for your time!!

olikraus, 2021-10-01:
Yes, correct.
Maybe additional using https://github.com/olikraus/u8g2/wiki/u8g2reference#setflipmode might help here.
setFlipMode(1) is basically the same as R2 but can be applied independently.

efstathios82, 2021-10-01:
I used 
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_MIRROR, /* cs=/ 10, / dc=/ 9, / reset=*/ U8X8_PIN_NONE);
and
u8g2.setFlipMode(1); 

working fine!!!

thank you!!!

stathis

closed 2021-10-01 by efstathios82