olikraus/u8g2 #1603: Font with Greek language

efstathios82, 2021-10-01:
Hi, 
 Exist font with Greek language like world ξ, λ, θ, Τ, σ, φ. ..  Where do I find it?

Thank you!!!

olikraus, 2021-10-01:
You need to search manually in the font groups: https://github.com/olikraus/u8g2/wiki/fntgrp

Greek is included here:
 * https://github.com/olikraus/u8g2/wiki/fntgrpnbp
 * https://github.com/olikraus/u8g2/wiki/fntgrpcu12
 * https://github.com/olikraus/u8g2/wiki/fntgrpunifont

efstathios82, 2021-10-01:
hi,

i find some font with Greek characters but when input into the loop code for example
->
u8g2.setFont(u8g2_font_cu12_t_greek)

only sow the characters with english word and the characters with Greek type does not show them.

i will try with the other font with Greek such as :
 u8g2.setFont(u8g2_font_unifont_t_greek);
or
 u8g2.setFont(u8g2_font_nine_by_five_nbp_t_all);
or
 u8g2.setFont(u8g2_font_mercutio_basic_nbp_t_all);

but again does not show the greek word and shows some strange symbols.
Αm i doing something wrong?
I simply change the command line in the code with the corresponding font!

stathis

olikraus, 2021-10-01:
Did you use https://github.com/olikraus/u8g2/wiki/u8g2reference#enableutf8print ?
Or https://github.com/olikraus/u8g2/wiki/u8g2reference#drawutf8 ?

drawStr will not work as it does not support UTF8.

efstathios82, 2021-10-01:
Νο! I was not using UTF8!
I used drawStr !

Νow displays the Greek characters!!!

Thank you very much!!!!

Stathis

closed 2021-10-04 by efstathios82