olikraus/u8g2 #1604: GU128x64-800B Support

Bankst, 2021-10-02:
Hello, there is an [existing Adafruit_GFX based implementation](https://github.com/rhalkyard/Noritake_GU800_GFX) to drive this display, but I would like to use u8g2. 

I began digging through u8g2 to see if I could port it myself but unfortunately quickly got lost. 
Some support on porting this display would be greatly appreciated.
[Here is the datasheet](https://media.digikey.com/pdf/Data%20Sheets/Noritake%20PDFs/GU128X64-800B.pdf).

Bankst, 2021-10-02:
What I was able to get started was the display info, based on this section of the datasheet.
[image]

```cpp
static const u8x8_display_info_t u8x8_d_gu800_128x64_display_info =
{
  /* chip_enable_level = */ 0,      /* GU800: Not used */
  /* chip_disable_level = */ 1,     /* GU800: Not used */
  
  /* post_chip_enable_wait_ns = */ 150,
  /* pre_chip_disable_wait_ns = */ 40,
  /* reset_pulse_width_ms = */ 1, 
  /* post_reset_wait_ms = */ 2,
  /* sda_setup_time_ns = */ 12,     /* GU800: Not used */
  /* sck_pulse_width_ns = */ 80,    /* GU800: Min 80 per datasheet */
  /* sck_clock_hz = */ 4000000UL,
  /* spi_mode = */ 0,               /* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,   /* GU800: Not used */
  /* data_setup_time_ns = */ 200,
  /* write_pulse_width_ns = */ 200, /* GU800: Min 150 per datasheet */
  /* tile_width = */ 16,            /* width of 16*8=128 pixel */
  /* tile_hight = */ 8,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 128,
  /* pixel_height = */ 64
};

```

olikraus, 2021-10-03:
Nice display, but seems to be very expensive. Indeed there is no support for such device in u8g2 (I came across such a VFD device before but it is too expensive for a open source project like u8g2), but I you are willing to do some testing we can try to implement support for your display.

After review of the datasheet: We need to implement this as a 64x128 device. The memory structure seems to imply this. This means, that we need to drive the display with U8G2_R1 (https://github.com/olikraus/u8g2/wiki/u8g2reference#setdisplayrotation) to get landscape mode.

I will go ahead with the implementation...

olikraus, 2021-10-03:
Technical question: The GU128x64 seems to consume 700mA. This is more than the Arduino can provide, right? So an external power supply is require, correct?

olikraus, 2021-10-03:
```
static const u8x8_display_info_t u8x8_d_gu800_128x64_display_info =
{
  /* chip_enable_level = */ 0,      /* GU800: CSS signal */
  /* chip_disable_level = */ 1,     /* GU800: CSS signal */
  
  /* post_chip_enable_wait_ns = */ 40,
  /* pre_chip_disable_wait_ns = */ 150,
  /* reset_pulse_width_ms = */ 2,   /* GU800: Unspecified in datasheet */
  /* post_reset_wait_ms = */ 2,     /* GU800: Min 1.5ms per datasheet */
  /* sda_setup_time_ns = */ 40,     /* GU800: 40ns according to the timing diagram */
  /* sck_pulse_width_ns = */ 80,    /* GU800: Min 80ns per datasheet */
  /* sck_clock_hz = */ 4000000UL,
  /* spi_mode = */ 0,               /* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,   /* GU800: Not used */
  /* data_setup_time_ns = */ 40,    /* GU800: Min 40ns per datasheet */
  /* write_pulse_width_ns = */ 150, /* GU800: Min 150ns per datasheet */
  /* tile_width = */ 16,            /* width of 16*8=128 pixel */
  /* tile_hight = */ 8,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 128,
  /* pixel_height = */ 64
};
```

olikraus, 2021-10-03:
I have created alle the constructors, for example:
``` 
U8G2_GU800_128X64_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_GU800_128X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```

olikraus, 2021-10-03:
I have created beta release 2.32.3.
Can you test the same?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Bankst, 2021-10-03:
> 
> 
> Technical question: The GU128x64 seems to consume 700mA. This is more than the Arduino can provide, right? So an external power supply is require, correct?

Yes, that is correct. I use a large 5V powerbank for my testing.

> I have created beta release 2.32.3. Can you test the same?
>
> You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
>
>     1. Remove the [...]

I'll give this a go now and report back. Thank you.

Bankst, 2021-10-03:
I've tested both the full buffer and page buffer u8g2 logo examples, and get nothing on the screen. I tested my code using the library I mentioned in the original post, and confirm it still works.

olikraus, 2021-10-03:
Which constructor did you use? Did you apply the correct pin values? Maybe you can share your test code.

Bankst, 2021-10-03:
I'm using this constructor.
```cpp
U8G2_GU800_128X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```

Here is my code repo that I have driven the display with successfully. It uses the exact same pins for CS, DC, and Reset.
https://github.com/Bankst/GU800_VFD_GL/blob/master/src/GU800/GU800_GFX.cpp

olikraus, 2021-10-03:
Are you able to see any kind of activity on CS or DC line? The reset should be driven high by u8g2.
Can you see any other kind of activity on the screen?

Bankst, 2021-10-03:
There is no screen activity. I will connect my logic analyzer and see what it shows.

olikraus, 2021-10-03:
> I will connect my logic analyzer and see what it shows.

logic analyzer, nice. 

If you see activity, we probably need to check the init sequence for the display. It is here:
https://github.com/olikraus/u8g2/blob/438906ed94227f113314080d86cfb3f03e286fd5/csrc/u8x8_d_gu800.c#L143-L172

Bankst, 2021-10-03:
Here is a capture with the u8g2 logo example ino, in the `loop()`.
[image]

Here is a capture from my code, looping printing text to the screen.
[image]

olikraus, 2021-10-03:
Hmm.. but should the reset signal be high? I mean it is low in both cases, but according to the datasheet, the reset signal must be high.
Another point is the sample frequency: U8g2 is configured to 4 MHz SPI. Not sure what 4096 samples at 8kHz means, but if the sampling frequency of your analyzer is 8KB, then the individual SPI bytes can not be detected.

Bankst, 2021-10-03:
for quick reference here are the `writeSPI` methods from the adafruit-gfx based library.
```cpp
size_t GU800::writeSPI(uint8_t data, bool isCommand) {
    *csport &= ~cspinmask;              // Assert /CS
    
    if (isCommand) {
        *dcport |= dcpinmask;           // Write command: DC high
    } else {
        *dcport &= ~dcpinmask;          // Write data: DC low
    }

    delayMicroseconds(1);               // Minimum 40nS between asserting /CS and starting transaction

    SPI.beginTransaction(this->spi);
    SPI.transfer(data);
    SPI.endTransaction();

    delayMicroseconds(2);               // Minimum 1.5uS between last clock and deasserting /CS
    *csport |= cspinmask;
    delayMicroseconds(1);               // Minimum 80nS between deasserting /CS and next transaction

    return 1;
}

size_t GU800::writeSPI(uint8_t * buf, size_t count, bool isCommand) {
    // GU800 requires that we deassert and reassert /CS after every byte, so no
    // gain in doing batch transfers :(

    for (int i = 0; i < count; i++) {
        this->writeSPI(buf[i], isCommand);
    }

    return count;
}
```

Bankst, 2021-10-03:
> Hmm.. but should the reset signal be high? I mean it is low in both cases, but according to the datasheet, the reset signal must be high. Another point is the sample frequency: U8g2 is configured to [...]

The reset line was labelled incorrectly, my mistake. I'll relabel that now. I also increased the sample rate to 50kHz and things look more sensible. Though the Select looks more like a Clock than the clock does...
[image]

olikraus, 2021-10-03:
It might look like some signals are swapped. However there is still a sampling issue:
With 4MHz SPI you need at least 8MHz sampling frequency (twice frequency for sampling: https://en.wikipedia.org/wiki/Nyquist%E2%80%93Shannon_sampling_theorem). So you should either sample with 8MHz or reduce the SPI speed with u8g2 by using setBusClock(2000000) to assign 2MHz SPI clock.

Bankst, 2021-10-03:
Ok, with a 12.5MHz sample rate this makes more sense. Only thing I'm unsure of is the SPI mode. 
[image]
[image]

olikraus, 2021-10-03:
U8g2 is configured for spi mode 0 (https://www.arduino.cc/en/reference/SPI)
But I think this doesn't matter much. How does the capture of the first view bytes look like with 12.5MHz?

olikraus, 2021-10-03:
I am not sure which SPI frequency is used by the Adafruit lib, however maybe the 4MHz suggested by your code is too high.
It might be worth to try setBusClock(1000000).

Bankst, 2021-10-03:
I tried 1MHz and got no display from u8g2 again. Let me capture the initial bytes again on 12.5MHz

Bankst, 2021-10-03:
Ah. The display is expecting CS to be asserted on every byte, but u8g2 is only asserting it at the end of a transaction.
[image]

Here is my code.

olikraus, 2021-10-03:
The pin assinment seems to be correct. In fact DC and select (ccs) seem to be ok (u8g2 does not unselect between each byte, but this should be ok). 
I see two differences:
 * Adafruit code seems to be slower, maybe only 1MHz instead of 4MHz
 * The data line behaves differently, which seems to indicate a different SPI mode.

> also of note, u8g2 sends the first 0x62 very early. it is off to the far left of the screenshot.

Please review the code from above: I first sent 0x5f, then wait for 1ms (as described in the datasheet) and then send the 0x62. It should be the same as for your code. 

Let's try different SPI modes.

Please locate the following line 187 in your local u8g2 code (file u8x8_d_gu800.c):
https://github.com/olikraus/u8g2/blob/61feaec9952a6dc47beb350d93b8f29b9e86ccfc/csrc/u8x8_d_gu800.c#L186-L187
There are four modes, from 0 to 3
Maybe also reduce SPI to 1000000 in line 186

Bankst, 2021-10-03:
I will give that a try, but I will note when I was first trying my own implementation to drive this display, asserting the select line between each bite was something that I had to do.

olikraus, 2021-10-03:
Here are some more ideas for testing (at least this is what I would do if I would have this display in my lab):
 * Use the U8g2 SW SPI constructor: It is slower and does not require the SPI mode thing. It will work with devices in any SPI mode.
 * I just checked the X / Y addressing: This seems to be ok: https://github.com/olikraus/u8g2/blob/61feaec9952a6dc47beb350d93b8f29b9e86ccfc/csrc/u8x8_d_gu800.c#L115-L118
 * Another issue might be the auto increment, however I think it should be ok (although we could test with 0x82 or 0x86): https://github.com/olikraus/u8g2/blob/61feaec9952a6dc47beb350d93b8f29b9e86ccfc/csrc/u8x8_d_gu800.c#L166 
 * There there is this layer option during display enable. Maybe for testing we should enable both layers, just to avoid that u8g2 writes to the wrong layer. Instead of 0x24 use 0x2c: https://github.com/olikraus/u8g2/blob/61feaec9952a6dc47beb350d93b8f29b9e86ccfc/csrc/u8x8_d_gu800.c#L49
 * Another idea could be to put the display into reverse mode, just to see whether the display accepts u8g2 commands in general: Replace the 0x40 with 0x50 for reverse bit operation: https://github.com/olikraus/u8g2/blob/61feaec9952a6dc47beb350d93b8f29b9e86ccfc/csrc/u8x8_d_gu800.c#L49

As you can see, adding a complete new display can become a difficult task...

olikraus, 2021-10-03:
> I will give that a try, but I will note when I was first trying my own implementation to drive this display, asserting the select line between each byte was something that I had to do.

Really? This would be the first device where this is required.

olikraus, 2021-10-03:
hmmm... I would love to buy this display for further testing but it's far too expensive :-(

Bankst, 2021-10-03:
I'll run through everything you've suggested and get back to you. And yeah they are pricey. Cheapest I see one right now is ~$70 USD on eBay. A few at that price, but still not a cheap display.

Bankst, 2021-10-03:
I've tried what you've suggested.
- SW SPI mode
- Writing to both layers
- Reverse bit mode
And have seen nothing yet on the display.

Bankst, 2021-10-03:
> 
> 
> > I will give that a try, but I will note when I was first trying my own implementation to drive this display, asserting the select line between each byte was something that I had to do.
> 
> Really? This would be the first device where this is required.

From the datasheet, section 10.3 Serial Interface
> To use the serial interface of this 
module, (RXD,TXD and SCK) will 
be activated by CSS= "L". 
The internal shift registers and 
counters will be reset by CSS= "H"

olikraus, 2021-10-03:
> From the datasheet, section 10.3 Serial Interface
> 
> > To use the serial interface of this
> > module, (RXD,TXD and SCK) will
> > be activated by CSS= "L".
> > The internal shift registers and
> > counters will be reset by CSS= "H"

Sure, this kind of text is there in all data sheets. It still does not mean, that this is needed for every byte.

We could change this, but I still do not think that this is the issue.

Bankst, 2021-10-03:
I can confirm removing the assertion of CSS on each byte stops my code from working.

olikraus, 2021-10-03:
oh....

olikraus, 2021-10-03:
I have added some more cs changes into new beta 2.32.5

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Bankst, 2021-10-03:
It doesn't seem the CSS behavior has changed at all with that update.
[image]
It is still only between chunks of bytes

olikraus, 2021-10-03:
> It is still only between chunks of bytes.

True, I just added the end/start transfer commands between the commands.
https://github.com/olikraus/u8g2/blob/3b4c6f8c9e1418061bbb77751bff308d0c7f235d/csrc/u8x8_d_gu800.c#L159-L165

> From what I can gleam I think this would need something to change in the u8x8_cad_SendSequence method that asserts it between each byte it sends in a sequence.

Actually we need to change 
https://github.com/olikraus/u8g2/blob/3b4c6f8c9e1418061bbb77751bff308d0c7f235d/csrc/u8x8_cad.c#L224

Each byte which is sent, has to be enclosed into a byte start and end transfer call:
```
uint8_t u8x8_byte_StartTransfer(u8x8_t *u8x8)
uint8_t u8x8_byte_EndTransfer(u8x8_t *u8x8)
```

So the first case will probably look like this:
```
    case U8X8_MSG_CAD_SEND_CMD:
      u8x8_byte_SetDC(u8x8, 1);
      u8x8_byte_StartTransfer(u8x8);
      u8x8_byte_SendByte(u8x8, arg_int);
      u8x8_byte_EndTransfer(u8x8);
      break;
```

`U8X8_MSG_CAD_SEND_ARG` has to be handled in the same way. `U8X8_MSG_CAD_SEND_DATA` is more complicated, because we have to send each byte individually. `U8X8_MSG_CAD_START_TRANSFER` and U8X8_MSG_CAD_END_TRANSFER just need to return 1.

All in all I am not sure when I will have time to work on this. Weekend is over...

Bankst, 2021-10-03:
That may be enough guidance for me to take it from here. Thank you greatly for your help so far. I will report any findings here.

olikraus, 2021-10-03:
Ok, here is the replacement for the cad_110 procedure. You could just replace your local copy of the cad_110 procedure:
``` C++
uint8_t u8x8_cad_110(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t *data;
  switch(msg)
  {
    case U8X8_MSG_CAD_SEND_CMD:
      u8x8_byte_SetDC(u8x8, 1);
      u8x8_byte_StartTransfer(u8x8);
      u8x8_byte_SendByte(u8x8, arg_int);
      u8x8_byte_EndTransfer(u8x8);
      break;
    case U8X8_MSG_CAD_SEND_ARG:
      u8x8_byte_SetDC(u8x8, 1);
      u8x8_byte_StartTransfer(u8x8);
      u8x8_byte_SendByte(u8x8, arg_int);
      u8x8_byte_EndTransfer(u8x8);
      break;
    case U8X8_MSG_CAD_SEND_DATA:
      u8x8_byte_SetDC(u8x8, 0);
      data = (uint8_t *)arg_ptr;
      whle( arg_int > 0 )
      {
        u8x8_byte_StartTransfer(u8x8);
        u8x8_byte_SendByte(u8x8, *data);
        u8x8_byte_EndTransfer(u8x8);
        data++;
        arg_int--;
      }
      break;
    case U8X8_MSG_CAD_INIT:
      u8x8->byte_cb(u8x8, msg, arg_int, arg_ptr);
      break;
    case U8X8_MSG_CAD_START_TRANSFER:
    case U8X8_MSG_CAD_END_TRANSFER:
      break;
    default:
      return 0;
  }
  return 1;
}
```

Not tested, not even compiled...

Bankst, 2021-10-03:
We have a display! I needed SPI Mode 2 for this, and 4MHz worked fine. Some optimization of the sequences so there are not multiple transfer start/ends between bytes may be good, but it works!
[image]

Bankst, 2021-10-03:
The screen is somewhat dim, I'm not sure the brightness is being set to max. I'll see if I can figure it out.

olikraus, 2021-10-03:
wow, very nice. 
So the updated cad_110 is required, right?
Could you test the following:
Currently the cad code includes this:
``` C++
    case U8X8_MSG_CAD_SEND_DATA:
      u8x8_byte_SetDC(u8x8, 0);
      data = (uint8_t *)arg_ptr;
      while( arg_int > 0 )
      {
        u8x8_byte_StartTransfer(u8x8);
        u8x8_byte_SendByte(u8x8, *data);
        u8x8_byte_EndTransfer(u8x8);
        data++;
        arg_int--;
      }
      break;
```

Will this also work? It would be a great performance improvement...
``` C++
    case U8X8_MSG_CAD_SEND_DATA:
      u8x8_byte_SetDC(u8x8, 0);
      data = (uint8_t *)arg_ptr;
      u8x8_byte_StartTransfer(u8x8);
      while( arg_int > 0 )
      {
        u8x8_byte_SendByte(u8x8, *data);
        data++;
        arg_int--;
      }
      u8x8_byte_EndTransfer(u8x8);
      break;
```

Bankst, 2021-10-03:
Unfortunately that does not work. :( Is there a good test I could run to evaluate performance?

olikraus, 2021-10-03:
> The screen is somewhat dim, I'm not sure the brightness is being set to max. I'll see if I can figure it out.

You can use the set contrast command of u8g2, however remember to use the value 255 for max brightness. You display has brightness levels from 0 to 15, however u8g2 uses 0..255: The value in u8g2 will be divided by 16 to get the GU800 value.

olikraus, 2021-10-03:
There is the FPS.ino example, which also includes some reference values of other displays.

Bankst, 2021-10-03:
My device is an Arduino Mega 2560
Page mode results:
- draw clip test, 12.3
- clear screen, 16.0
- draw @, 3.7
- draw pixel,  5.7

Full buffer results:
- draw clip test, 14.8
- clear screen, 16.5
- draw @, 6.1
- draw pixel, 7.9

olikraus, 2021-10-03:
well... better than expected :-)

olikraus, 2021-10-03:
I have created a new cad_110 procedure. Also reverted some of the recent changes.

olikraus, 2021-10-03:
Created beta 2.32.6
Does this still work?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2021-10-03:
Did you solve the "dim" brightness issue?

olikraus, 2021-10-03:
Thanks for testing and working with me on this topic. I am happy to have VFD support inside u8g2 :-)

Bankst, 2021-10-03:
> 
> 
> Did you solve the "dim" brightness issue?

I'll be looking in to that soon. I'll give the 2.32.6 beta a test now.

Bankst, 2021-10-03:
2.32.6 results in all pixels being On. I cannot make out whether there is anything actually happening on the screen.

Bankst, 2021-10-03:
Ah, simple fix. the SPI mode must be 2, not 0.
https://github.com/olikraus/U8g2_Arduino/blob/f014fc898afb13460323243588a17a73043aecc2/src/clib/u8x8_d_gu800.c#L192

Bankst, 2021-10-03:
I just PR'd the SPI mode fix and some other cleanups. A bunch of unnecessary start/end transfers were still there. I have confirmed that the display works fine with these changes. All that is left is brightness control.

olikraus, 2021-10-04:
> After changing this to 2, all is well again.

puh, guess it was too late for me to remember all the changes :-)
Thanks for the PR

olikraus, 2022-01-07:
ok, i assme this can be closed...

closed 2022-01-07 by olikraus

WuSiYu, 2024-10-25:
@olikraus 
This lib works fine on a Arduino UNO with this display. But when I switched to a esp32s3 (with a 3.3v to 5v I/O converter chip), it's not work anymore (no display), until I changed the SPI mode from 2 to 3, it works again.

(FYI: On esp32s3 I'm using `U8G2_GU800_128X64_F_4W_HW_SPI` with `u8x8_SetPin(u8g2.getU8x8(), U8X8_PIN_SPI_XXX, xx);` to set a custom SPI pin)

WuSiYu, 2024-10-26:
I checked the screen's datasheet, according to the fig I think spi mode 3 is the correct one:

[image]

Later I will test spi mode 3 on an Arduino UNO as well.

olikraus, 2024-10-29:
Hmm.. there had been similar issues in history, where the SPI modes had not been compatible. See for example here: https://github.com/olikraus/u8g2/issues/53 
Maybe ESP32 has a similar problem.

[4 more replies]