olikraus/u8g2 #1608: Fysetc mini12864 with u8g2

Syldan, 2021-10-04:
Hallo,
ich bräuchte Hilfe beim folgenden Problem.

Ich arbeite an einem Projekt, bei dem wir einen eigenen 3D-Drucker bauen. Dieser läuft OHNE die Software Marlin die normalerweise für 3D-Drucker verwendet wird. Die Infos für das verwendete Modul fysetc mini12864 v2.1 mit einem eingebauten LCD Display (anscheinend ST7567) sind hier auf der Seite beschrieben:

https://github.com/FYSETC/FYSETC-Mini-12864-Panel

Ich habe schon die Beispiele der u8g2 Bibliothek für ST7567 anzuwenden jedoch ist wahrscheinlich die Pin Anordnung nicht korrekt.
Ich würde gerne für den Anfang dass ich das Display auf dem Modul ansteuern und Texte anzeigen lassen kann.

die Anschlusspins am Arduino welche für das Modul verwendet sind hier definiert.

/*LED pin for controlling the colour of the backlight*/
#define ledp 47

/* define display and sd */
    #define MISO    33
    #define MOSI    28
    #define SS      30
    #define SCK     32
    #define RST     26
    #define CD      27
    #define LEDRST  46
    #define CS      44
    #define A0      45
    #define beep    42

Ich würde mich sehr über Hilfe freuen.

olikraus, 2021-10-04:
Ich fürchte hier sind viel zu wenige information. Aber folgende Vorgehensweise:
Denkbar wäre folgender U8g2 Constructor (wenn es ein ST7567 controller ist):

U8G2_ST7567_JLX12864_1_4W_SW_SPI(rotation, clock, data, cs, dc [, reset])
rotation is eine u8g2 constante, meistens U8G2_R0, dann die fünf signale

| LCD |
|---|
| clock |
| data |
| cs |
| dc |
| reset |

Diese signale können auch andere namen haben. Ich hab hier ein paar alternative Namen zusammengetragen: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#wiring

| U8g2 | Alt |
|---|---|
| clock | SCK | 
| data |MOSI |
| cs | SS |
| dc | CD / A0 |
| reset | |

Aber: Oben stehen 10 signale während das display nur 5 benötigt. Welches sind also die richtigen?
Das also musst Du wohl irgendwie herausfinden. Man könnte sich den EXP1 Stecker anschauen und versuchen das auf die Tabelle zu matchen:

| U8g2 | Alt | EXP1 Mini | EXP2 Mini  |
|---|---|---|---|
| clock | SCK |  | SCK |
| data |MOSI |  | MOSI |
| cs | SS | LCD CS | |
| dc | CD / A0 | LCD A0 | |
| reset | | LCD RST | |

Aber wo und wie sind die beiden EXP1/2 Stecker verbunden? Wie auch immer, ab hier solltest Du anfangen die Tabelle zu erweitern und herausfinden wohin die Leitungen gehen.

Hier noch ein Video, von einem Typen, der wohl das gleiche Problem hatte:
https://www.youtube.com/watch?v=4CD8ERaylmY

olikraus, 2021-10-11:
Super, Danke für die Rückmeldung.