olikraus/u8g2 #1612: Custom TTF to U8g2

xavieraijon, 2021-10-05:
Hi,

How I convert ttf font to bdf, to u8g2? I'm using Windows or MacOs too.
I need in various pt sizes: 18, 20, 22, 24 & 26

[Hornet UFC.zip](https://github.com/olikraus/u8g2/files/7289238/Hornet.UFC.zip)

Thanks!

xavieraijon, 2021-10-05:
> I personally use otf2bdf: http://sofia.nmsu.edu/~mleisher/Software/otf2bdf/

Thanks! Can I use .bdf directly with u8g2? Or need convert to other format?

olikraus, 2021-10-05:
Yes, .bdf needs to be converted into a byte array with bdfconv.exe

https://github.com/olikraus/u8g2/blob/91fd7ad476d5b2ee88e096123b8c9d640c0467e3/doc/faq.txt#L245-L272

xavieraijon, 2021-10-06:
Hi,
 Is normal that only have 2 lines of byte array?
```
const uint8_t u8g2_font_hornet[35] U8G2_FONT_SECTION("u8g2_font_hornet") = 
  "\1\0\2\2\0\0\1\1\6\0\0\0\0\30\1\30\1\0\0\0\0\0\6\0\4\373\0\0\0\0\4\377"
  "\377\0";
```

olikraus, 2021-10-06:
Probably not. You may need to use -m or - M to specify the encodings.

xavieraijon, 2021-10-06:
> Probably not. You may need to use -m or - M to specify the encodings.

I use this sentence:
`bdfconv -v -f 1 -m '32-65535' hornetufc.bdf -o u8g2_font_hornet.c  -n u8g2_font_hornet`

xavieraijon, 2021-10-06:
I tried without -m or -M and compiled aparently correct, but the characters are rendered in random heights...
[image]

xavieraijon, 2021-10-06:
Ouh, thanks so much! Where can I find the ftview tool?

olikraus, 2021-10-06:
I checked the font by myself. The only useful -p value will be 26. 

The only proper result with otf2bdf will require this arguments:
```
otf2bdf -a -r 72 -p 26 -n Hornet\ UFC.ttf -o hornet.bdfhornet
```

[image]

Unfortunately something is wrong with some numbers. 6 is too high and 7 too low. Same for P, R and T and others. Especially K looks funny.

Doing a further investigation of the bdf file which was generated from otf2bdf, then it looks like that the BBX is calculated incorrectly:
Lets take the "7" as example:
```
STARTCHAR 0037
ENCODING 55
SWIDTH 615 0
DWIDTH 16 0
BBX 10 20 4 2
```
The last number of the BBX entry is 2, but this is wrong. It should be 4 instead:
```
STARTCHAR 0037
ENCODING 55
SWIDTH 615 0
DWIDTH 16 0
BBX 10 20 4 4
```

> I need in various pt sizes: 18, 20, 22, 24 & 26

Only pt 26 will be possible (26pt in ttf equals to 21 pixel height, see my screenshot).
Other values will generate ugly chars. But even with 26pt a lot of effort in fixing the .bdf will be required.

closed 2022-03-04 by olikraus

GameLord2011, 2024-12-11:
I want to put Caskaydia Mono Nerd Font on my flipper zero as a main or secondary font. I already put u8g2_font_lucasarts_scumm_subtitle_o_tn as a big numbers font. is there a way to convert it straight from ttf to u8g2 font section?