olikraus/u8g2 #1630: CM12864-10C T6963C CE is active-low

prenticedavid, 2021-10-13:
Which particular T6963C 128x64 display are you using?
The u8g2_Setup_t6963_128x64_1() expects CE to be active-high

My CM12864-10C display has regular active-low CE.   Just like the 160x80, 240x64, 240x128, 256x64, 256x128 u8g2_Setup_t6963() functions,

I can invert the logic in my custom u8x8_avr_gpio_and_delay_8080() callback function.  e.g.
```
        case U8X8_MSG_GPIO_CS:        // CS (chip select) pin: Output level in arg_int
            if (arg_int == 0)             // >>>>> .kbv <<<<<<
                DISPLAY_CS_PORT |= (1 << DISPLAY_CS_PIN);
            else
                DISPLAY_CS_PORT &= ~(1 << DISPLAY_CS_PIN);
            break;

```
I note that vergo77's WG12864D had regular active-low CE but inverted DC levels.   He could do a similar edit for 
```
         case U8X8_MSG_GPIO_DC:
```

This is simple but a little untidy.   I initially tried:
```
	u8g2_GetU8x8(&u8g2)->display_info->chip_enable_level=0;
```
but this does not work because `display_info` is read-only.

David.

olikraus, 2021-10-13:
One more aspect. One plane is to move u8x8_display_info_t into Flash ROM. In such a case
```
 *((uint8_t*)&u8g2_GetU8x8(&u8g2)->display_info->chip_enable_level)=0;
```
will not work any more. In fact on ARM systems this is already in FlashROM (due to the const keyword) and the attempt of modifing  chip_enable_level will not have any effect.

olikraus, 2021-10-13:
Based on https://github.com/olikraus/u8g2/blob/cd911cc8316c7e8d2ba50ba3229f5a1671955bbf/csrc/u8x8_d_t6963.c#L443:
 * Create new device u8x8_d_t6963_128x64_cs, which is a copy of the u8x8_d_t6963_128x64 device
 * Create new device u8x8_d_t6963_128x64_ncs, which is a copy of the u8x8_d_t6963_128x64 with negative chip select.

olikraus, 2021-10-14:
> Is it "unusual" or do many T6963C displays have active-high CE ?

I stopped wondering about unusual displays :-D There always seems to be an exception of the "normal" behaviour ;-)

closed 2022-01-07 by olikraus