olikraus/u8g2 #1633: U8g2 on platformio is out of date

alliefitter, 2021-10-15:
Not sure if this belongs here or on platform's site, but the version that's available in platform's library manager is 2.28.8. I'm trying to use streamline icons, and they weren't available in that build. After banging my head against the wall for an hour or so, I was able to manually install 2.32.6 using `pio`. I don't know if libraries are added to platform by maintainers or by platform themselves. The way I was able to install the newest version was to download the zip and then install it with `$ pio lib -g install /path/to/U8g2_Arduino-master.zip`.

olikraus, 2021-10-17:
Hmmm not sure how u8g2 project can help here

alliefitter, 2021-10-17:
So then, should I go to PIO and ask them to update it? Sorry if I wasn't clear, but that's what I'm asking. Who is it that uploads new versions? Y'all or PIO?

olikraus, 2021-10-17:
U8g2 project is not related to platformio and there is also no trigger from my end to upload something somewhere.
U8g2 is a pure Arduino library.

alliefitter, 2021-10-17:
Cool, I'll get with them. I'm new to Arduino, so I wasn't sure if PIO was like Python's PyPI where library maintainers upload new versions or like Linux distros. Thanks.

chriscrown, 2021-11-25:
@alliefitter Did you get any response from PIO? I read the Lib-crawler would run every 24 hours, so the update process seems to be broken. Would love to see the current version of u8g2 in PIO lib! :-)

alliefitter, 2021-11-25:
I got distracted and never reached out. :/