olikraus/u8g2 #1642: Support for ST75160 display

olikraus, 2021-10-18:
### Discussed in https://github.com/olikraus/u8g2/discussions/1641

<div type='discussions-op-text'>

<sup>Originally posted by **johngineer** October 18, 2021</sup>
Hello! I am working with a 160x96 monochrome display that uses an ST75160 controller. It would be great if you could add support for this chip to u8g2.

To that end, I am willing to provide a sample display + breakout PCB for test & eval. I just need to know where to send it.

Thanks!</div>

Is the datasheet of the controller available?

olikraus, 2021-10-18:
Actually both would be perfect: The datasheet of the controller and the datasheet of the display.

olikraus, 2021-10-19:
The ST75160 seems to have almost the same commands as the ST75256. I assume we could just update the ST75256 driver.
Can you already test any code with the display?

johngineer, 2021-11-11:
Just a quick update on this. There was a customs delay in the shipping of the LCDs and breakout PCB, but I should receive them by the end of next week. Once I have them assembled, I will test the code and follow up with my feedback.

olikraus, 2022-08-13:
no further feedback... closing for now...

closed 2022-08-13 by olikraus