olikraus/u8g2 #1651: Platformio folder structure

spassignat, 2021-11-01:
Hello,

I'm suddenly have a build error of this library. Platformio tries to buid example folder (at least) and fails for several reasons like multiple definition of loop function (sure mine + all the ones from examples).

Reading the platformio documentation, it seems to expect the header files in subfolder include and sources in src, with library.properties or library.json. Otherwise it will build everything which it does.

Are you aware of this ?

thanks

olikraus, 2021-11-01:
U8g2 has been designed to support Arduino. I do not know anything about Platformio.

Berkays, 2021-11-02:
Place csrc folder into lib folder then it will compile assuming you don't have any other problem.

spassignat, 2021-11-03:
Never mind, my build options were wrong I was using --target Z_DUMMY_TARGET. Using Production solved my errors.

closed 2021-11-03 by spassignat