olikraus/u8g2 #1654: I2C with ESP32 CAM

FGgithubprocessing, 2021-11-01:
Hi,
I have an issue trying to connect an OLED display to my ESP32-CAM.
I'm using fine this device with SD card but without camera.
I have only few pins available to connect I2C OLED; these pins are U0T and U0R (IO1 and IO3 respectively) and these are used
also for serial comunication to upload the sketch.

What I want to do is upload a sketch wit I2C comunication and using U0T as SDA and U0R as SCL.
Obviously, after uploading, I have to disconnect them from serial port and attach them to I2C device.
I tryed with the simple sketch seen on github (here it is...)

[I2C_OLED.txt](https://github.com/olikraus/u8g2/files/7454689/I2C_OLED.txt)
 
In that sketch I did these changes:
#define SDA 1 // instead of 23
#define SCL 3 // instead of 19
.
.
.
U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);
// instead of SSD1306... since my display is SH1106
and then upload to my ESP32-CAM and connect pins correctly.

I tryed also with ...F_HW_I2C... but without success. Nothin happens on OLED display.
The display is working perfectly with avr processor but not with ESP32-CAM!

Is there something wrong on that?
Can somebody help me?

I'll appreciate some answer from you.
Thanks and regards.
Fiorenzo Daniele Giust

olikraus, 2021-11-01:
So, SW_I2C works?

FGgithubprocessing, 2021-11-02:
Hi,
thr display OLED works but only with library Adafruit_SH1106 and on avr processor (Arduino Nano, Arduino Uno...) but not
with ESP32-CAM (I tryed both Adafruit_SH1106 and U8g2lib libraries with ESP).
With Adafruit it seems that is working only with pins 4 and 5 (default?).
I would like to use Adafruit assigning pis to 1 and 3 for ESP32-CAM. Is it possible?

What is the difference between SW_I2C and HW_I2C ?

It is very hard to work with only 2 IO's in sharing (I2C and serial)!!
I don't want to loose the SD card since it's important for me.
The pins used for SD are GPIO14, GPIO15, GPIO2, GPIO4, GPIO12, GPIO13.
The only available are those for serial (GPIO1 and GPIO3).

Regards

Fiorenzo

olikraus, 2021-11-02:
With SW_I2C you can use any available gpio pin. The HW_I2C constructor will try to activate the I2C subsystem if your controller. 
Try to use SW_I2C first.

closed 2022-03-04 by olikraus