olikraus/u8g2 #1655: T6963C 320x128 display

ankas2, 2021-11-01:
Hello, I try to  make new device with 320x128 pix, but it work as 240x12
[U8g2_T6963C_320x128.zip](https://github.com/olikraus/u8g2/files/7456355/U8g2_T6963C_320x128.zip)
8.

olikraus, 2021-11-01:
Any datasheet available for your display?

ankas2, 2021-11-03:
[image]
I don't have datasheet.

olikraus, 2021-11-03:
So your mod works as expected?

ankas2, 2021-11-04:
[image]
[image]
[image]

ankas2, 2021-11-04:
Something I miss, and don't find what.

ankas2, 2021-11-06:
At last found what I miss.
In a few days send here all my info about display.

olikraus, 2021-11-07:
Nice pictures. I still do not know how I could support. Shell I create an empty constructor for this?

ankas2, 2021-11-07:
[EDT EW50404FLW_PINOUT.txt](https://github.com/olikraus/u8g2/files/7492953/EDT.EW50404FLW_PINOUT.txt)
[media.zip](https://github.com/olikraus/u8g2/files/7492954/media.zip)
[U8g2_my added-T6963C_320x128.zip](https://github.com/olikraus/u8g2/files/7492956/U8g2_my.added-T6963C_320x128.zip)

All my info about this display, library is working, my mistake was 16bit mode not enabled.

closed 2021-11-07 by ankas2