olikraus/u8g2 #1665: UC1601 - "Constructor does not name a type"

letshin, 2021-11-06:
When including

``U8X8_UC1601_128X64_8080 u8x8(25, 26, 27, 14, 12, 13, 4, 2, 32, 23, 18, 19);

into the start the following error returns

``HelloWorld:219:1: error: 'U8X8_UC1601_128X64_8080' does not name a type
``U8X8_UC1601_128X64_8080 u8x8(25, 26, 27, 14, 12, 13, 4, 2, 32, 23, 18, 19);
`` ^
Not sure what is going on - any advice please?

olikraus, 2021-11-07:
Sometimes the error is generated due to a missing object name, but in your constructor line it seems to be available.

The requested constructor is here: https://github.com/olikraus/u8g2/blob/2e7b5be78cf0067043a929ad7dbf6677f2bfc1b3/cppsrc/U8x8lib.h#L5133

There up to date reference should be https://github.com/olikraus/u8g2/wiki/u8x8setupcpp
This also lists your constructor: U8X8_UC1601_128X64_8080 https://github.com/olikraus/u8g2/wiki/u8x8setupcpp#uc1601-128x64

The latest version of u8g2 is 2.31 (https://www.arduinolibraries.info/libraries/u8g2)

letshin, 2021-11-07:
Sorry & thanks. I think this was because I had installed u8x8 earlier and the script must have been loading the older .h file instead of the one in the u8g2 library. Deleting the u8x8 folder and restarting the IDE meant that I have no issues uploading the code now!

closed 2021-11-07 by letshin